google/android-riscv64 #1: system/core/init/: more ASLR bits when we have Kconfig support for more

enh-google, 2023-01-30:
init can ask for more aslr bits when we have newer kernels: https://android-review.git.corp.google.com/c/platform/system/core/+/2251618/1..2/init/security.cpp#b120

TODO: find out what kernel version improves this; ToT still only has 24 in arch/riscv/Kconfig --- https://github.com/torvalds/linux/blob/master/arch/riscv/Kconfig#L150 ... it looks like maybe VA_BITS has been increased already (https://github.com/torvalds/linux/blob/master/arch/riscv/include/asm/pgtable.h#L60) but no-one came back to fix the Kconfig, so there are still upstream fixes needed before we can use this in init?

enh-google, 2023-09-19:
(Q3 update: kernel still only supports 24 bits. https://android-review.googlesource.com/c/platform/system/core/+/2755787 at least updates the comment, and makes the "minimum maximum" 24 too.)

samitolvanen, 2023-09-29:
Kernel patches to increase the maximum to 33 with Sv48/57:

https://lore.kernel.org/linux-riscv/20230929211155.3910949-4-samitolvanen@google.com/

enh-google, 2023-11-13:
is https://github.com/torvalds/linux/commit/add2cc6b6515f78d3a150f1fbbaf12c28c4bb20a all we need upstream?

samitolvanen, 2023-12-01:
> is [torvalds/linux@add2cc6](https://github.com/torvalds/linux/commit/add2cc6b6515f78d3a150f1fbbaf12c28c4bb20a) all we need upstream?

No, that patch reduces the size of mmap address space to avoid breaking userspace applications (and should fix the CFI shadow issue I worked around earlier by switching Cuttlefish to use Sv48 instead of Sv57). If we want to increase mmap randomness, we still need the patches I posted earlier.

samitolvanen, 2024-01-26:
The kernel patches were queued up for v6.9:
https://lore.kernel.org/linux-riscv/170621822929.6239.356777907601716533.git-patchwork-notify@kernel.org/

samitolvanen, 2024-05-21:
android-mainline(-riscv64) has the patches now, so here's the matching init change to allow Sv48/57 to use 33 bits: https://r.android.com/3095551

closed 2024-05-22 by samitolvanen