google/android-riscv64 #2: external/scudo/: optimize CRC32 when we have useful instructions

enh-google, 2023-01-30:
it looks like Zbr didn't make it? is Zbc's clmulr useful?

enh-google, 2023-05-26:
@negge

nick-knight, 2023-05-27:
SiFive has a prototype CRC32 for Zlib-ng using Zvbc vclmul/h. We are just putting some finishing touches on it. Happy to redirect it at Scudo after we land it it Zlib-ng (and Chromium fork).

nick-knight, 2023-05-27:
By the way, you can avoid the overheads of bit-reversal with some algebra tricks. We don't need any of the Zvbb stuff. (And Zvbc doesn't have a reversed carryless multiply, as opposed to Zbc.)

enh-google, 2023-08-07:
i note that the scudo checksum code (https://github.com/llvm/llvm-project/blob/0ba9aec38faa8f567741fdb4eac1149c3fe94ceb/compiler-rt/lib/scudo/standalone/checksum.h etc) is probably quite suboptimal for vectors because it breaks the data up into word-sized chunks and checksums them separately, rather than there being a single call to checksum a whole range? that's something we might want to look at, if only for riscv64. @cferris1000 @chiahungduan

(also, it seems like the zlib-ng code -- https://github.com/zlib-ng/zlib-ng/tree/develop/arch/riscv -- is adler32 rather than crc32.)

nick-knight, 2023-08-07:
> (also, it seems like the zlib-ng code -- https://github.com/zlib-ng/zlib-ng/tree/develop/arch/riscv -- is adler32 rather than crc32.)

We (SiFive) haven't upstreamed our RVV (+Zvbb+Zvbc) CRC32 implementation yet. It's just sitting in our downstream fork waiting for me to finish my edits of our PDF writeup to accompany the code. May also be blocked on Zvbb and Zvbc support in LLVM and QEMU.

lukel97, 2023-08-08:
> May also be blocked on Zvbb and Zvbc support in LLVM and QEMU.

Just commenting on the status here, [the LLVM IR intrinsics for Zvbb + Zvbc landed last week](https://reviews.llvm.org/D138809), but [the patch for C intrinsics is still in review](https://reviews.llvm.org/D138810).
Unfortunately the former didn't make it for the LLVM 17 release, but it should be available from ToT. It's marked as experimental so I think you'll need to specify the `-menable-experimental-extensions` flag from Clang.

enh-google, 2023-08-08:
> SiFive has a prototype CRC32 for Zlib-ng using Zvbc vclmul/h. We are just putting some finishing touches on it. Happy to redirect it at Scudo after we land it in Zlib-ng (and Chromium fork).

thanks! (fwiw, i don't think scudo cares much about the _specific_ crc algorithm --- aiui "whatever's cheap + easy" is their goal.)

enh-google, 2024-01-18:
some more details here, as we've actually started to look at this problem...

1. scalar carryless multiply isn't in RVA23, so we probably won't have it.
2. vector carryless multiply is unlikely to amortize for scudo's use case, which is always exactly 16 bytes. (the adler32 rv64 code currently in zlib-ng explicitly uses 16 bytes as its threshold for "don't bother with vector".)
3. the scudo folks report that "The checksum doesn't surface up in the performance evaluation [for x86-64] we have done so far. I took a quick look at the few [x86-64] profiles, the checksum calculation takes about 0.15% time in both allocation/deallocation paths."

so right now it's not obvious that there's anything useful we can do here. (but also not obvious that it's going to be a huge problem.)

nick-knight, 2024-01-18:
Just a quick clarification.

Recall that there are two families of carryless multiply instructions:

* "scalar", defined in the Zbc extension
* "vector", defined in the Zvbc extension

Both Zbc and Zvbc are optional in the (not yet ratified) RVA23 profile. Originally, Zvbc was supposed to be mandatory in RVA23, but the profiles committee couldn't agree in time, so it is optional for RVA23 and intended to become mandatory in RVA24+. On the other hand, Zbc is expected to remain optional in RVA24+. I expect this to motivate hardware vendors to implement Zvbc but not necessarily Zbc. Further motivation may arise from the fact that Zbc can be emulated by invoking Zvbc with VL = 1 (note*). 

* Small CRCs will not benefit from carryless multiplication (scalar or vector).
* Large CRCs will benefit from carryless multiplication, vector more so than scalar.
* Scudo's (and CoreMark's ;) CRCs fall in between these two cases: they will likely benefit from scalar carryless multiplication, but there is likely insufficient data parallelism to benefit from vectorization. However, on a machine with Zvbc but not Zbc, the vector instructions can be invoked with VL = 1 to emulate the scalar ones. (The design space is richer than the Zbc-only case --- albeit in an uninteresting way --- since it is probably worthwhile doing other parts of the computation with VL = 1 vector instructions to avoid data movement between the scalar and vector register files.) This has come up a few times now so I should probably write it down rather than talking about it.

(note*) As I mentioned earlier in this thread, algebraic tricks can be played in the context of CRC to avoid the need for `clmulr` or its hypothetical vector analogue.