google/android-riscv64 #3: bionic/: implement TLSDESC (not yet in psabi)

enh-google, 2023-01-30:
risc-v is the first Android ABI to ship after we added ELF TLS support. emutls doesn't even seem to work (with LLVM at least) for risc-v, but we can't go straight to using TLSDESC relocations because they haven't been standardized for risc-v yet. blocked on https://github.com/riscv-non-isa/riscv-elf-psabi-doc/issues/94 getting some traction.

appujee, 2023-09-14:
Reference to sourceware implementation: https://inbox.sourceware.org/binutils/DS7PR12MB576522B309E27EBA26708A85CB1BA@DS7PR12MB5765.namprd12.prod.outlook.com/T/

appujee, 2024-05-08:
Support for RISC-V TLSDESC Relocations has landed in llvm: https://github.com/llvm/llvm-project/pull/66916 Thanks to @ilovepi and @MaskRay

enh-google, 2024-11-12:
@appujee --- is there anything left to do here? do we have a prebuilt clang where this is on by default?

appujee, 2024-11-12:
Yes it is enabled in the current clang-r530567

appujee, 2024-11-12:
And enabled by default for Android targets in https://github.com/llvm/llvm-project/pull/81198

ilovepi, 2024-11-12:
Yeah, TLSDESC should be on by default in Clang for all Android RISC-V targets. I don't think there's anything left to do, since we've resolved all the psABI issues, updated the compiler and linker, and bionic.

enh-google, 2024-11-13:
yeah, and the only CTS test i can think of for arm64 already includes riscv64, so i don't think there's any extra tests to be written either? thanks all!

closed 2024-11-13 by enh-google