google/android-riscv64 #12: llvm: should sqrt/sqrtf and the lrint/lrintf/llrint/llrintf family be builtins?

enh-google, 2023-01-31:
both riscv64 (https://cs.android.com/android/platform/superproject/+/master:bionic/libm/riscv64/) and arm64 (https://cs.android.com/android/platform/superproject/+/master:bionic/libm/arm64/) just have two .S files in libm, and they're all single-instruction functions for both architectures. this seems like something that llvm should just know about, so it can turn calls to sqrt etc into single instructions?

godbolt on a trivial function calling sqrt() shows that this isn't happening currently for arm64 or riscv64 (tested at -O2).

appujee, 2023-02-01:
both sqrt and lrint functions are getting generated as native instructions in RISC-V as well as Aarch64

https://godbolt.org/z/Wj9eonzzb

enh-google, 2023-02-01:
Android's libm doesn't set errno, and clang knows this, so it doesn't do the overflow test and call to the real function so it can set errno.

cleaned up for both arm64 and riscv64 here: https://android-review.googlesource.com/c/platform/bionic/+/2414394

closed 2023-02-01 by enh-google