google/android-riscv64 #15: hardware cfi ("landing pads") support

enh-google, 2023-01-31:
if https://github.com/riscv/riscv-cfi lands early enough, we'll want that as part of the ABI. even if it doesn't, we'll want to support it and enable it by default for the kernel and OS.

samitolvanen, 2023-02-13:
Kernel patches for U-mode Zisslpcfi support: https://lore.kernel.org/lkml/20230213045351.3945824-1-debug@rivosinc.com/

Note that for fine-grained forward-edge CFI we'll still need the compiler to emit landing pad tags.

appujee, 2023-04-11:
Currently there are two blockers to CFI proposals
- depends on Zimop https://wiki.riscv.org/display/HOME/Specification+Status
- cfi extension for landing pads

deepak0414, 2023-04-11:
I would edit above comment to 

Support for Zisslpcfi based CFI in android-riscv64 would be dependent on 

Spec ratification dependencies

- Ratification of zimops spec
- Ratification of Zisslpcfi spec

Software dependencies
- Support for U mode shadow stack and landing pad in toolchain (gcc and llvm)
- Support for U mode shadow stack and landing pad in linux kernel
- Support for shadow stack and landing pad in libc (glibc. If android uses different libc then mention that)

samitolvanen, 2023-08-17:
A lot has changed here since the previous comments. Landing pads were split into the Zicfilp extension, which no longer uses Zimops, and the `lpad` instruction is now in HINT space with the goal of making Zicfilp enabled programs compatible with non-Zicfilp hardware.

Unfortunately, it doesn't look like the latest spec has an option to disable Zicfilp per application (or even better, per page), it can only be disabled for the entire privilege mode, which creates a different set of problems. Specifically, if we want to enable Zicfilp in Android:

1. We must have landing pads in all userspace binaries from the beginning. Otherwise, it's going to be challenging to enable the feature later without breaking compatibility with existing apps that don't have landing pads.
2. We're going to be stuck with ARM BTI style coarse-grained CFI, because ensuring all userspace software works with fine-grained CFI is not feasible. For components where we actually want to use fine-grained CFI, we'll presumably need a separate set of libraries that also use fine-grained CFI.

On the compiler side, I see LLVM recently added MC layer support for Zicfilp, which simply adds support for the `lpad` instruction: https://reviews.llvm.org/rG818e76d6f2dd90362e71e84f492c167958ef3257
So far it doesn't look like anyone has submitted patches to actually emit these instructions as part of a CFI scheme.

samitolvanen, 2023-08-18:
> We must have landing pads in all userspace binaries from the beginning. Otherwise, it's going to be challenging to enable the feature later without breaking compatibility with existing apps that don't have landing pads.

Ved kindly explained to me that the idea is for the kernel to disable Zicfilp when context switching to a process that hasn't declared support for the extension. This also disables the extension for any libraries loaded into the process, so ideally we would still enable landing pads from the beginning, but at least it should be possible to enable the extension in other processes.

samitolvanen, 2024-11-12:
We still need kernel support before we can test this in userspace. The kernel patches are now at v8, which was just sent out yesterday:

https://lore.kernel.org/linux-riscv/20241111-v5_user_cfi_series-v8-0-dce14aa30207@rivosinc.com/

Presumably more compiler patches are needed to actually emit landing pads too.

enh-google, 2024-11-12:
ah, i sorry... i just assumed -- based on the long gap since the last time the kernel support was mentioned -- that that would be in already. which makes me realize i've only _assumed_ that the compiler side is done too. i'll check up on that...

deepak0414, 2024-11-12:
I imagine for android llvm support will be needed.
We can check with Ming-Yi as well
https://github.com/mylai-mtk/llvm-project/tree/zicfilp-func-sig-intrinsic
For some reason I can't tag Ming-Yi here (Ming-yi.Lai@mediatek.com)

And I think it does support function-signature based landing pads
(Note: only for static and with musl. `glibc` needs gcc)

I imagine if we can make llvm patches upstream and add support for
shadow stack. We should be able to build bionic libc and unblock AOSP

deepak0414, 2024-11-12:
On topic of emitting landing pad.
I am of the opinion that we follow one scheme for labeling landing pads.
Primary reason is RVA23 is clean break from existing software and hardware and thus gives the opportunity
to implement finer grained control flow integrity which doesn't require any backward compat.

I've written down the detailed arguments in this document here
https://docs.google.com/document/d/1AqUadimUsaBFv7_Ff7VkPF8yeJPR6n33W8QlFPz9Br0/edit?tab=t.0#heading=h.bdqz6b6k9yev

Above arguments in document apply to Android as well.