google/android-riscv64 #17: external/libvpx: missing optimized assembler

enh-google, 2023-01-31:
external/libvpx [https://android-review.googlesource.com/c/platform/external/libvpx/+/2168948] is missing optimized assembler (present for arm/arm64/x86/x86-64), using C fallback for now.

sunmin89, 2023-06-13:
@rbosetti 
Hi, could you share any insights about the ways and progress around this optimization work.

I'd like to help optimize this library too.

So far, It is challenging to just prepare a environment to test these V related functions.

Does the  toolchain clang or simulator cuttlefish  support the V extension already?

Thanks !

enh-google, 2023-06-13:
> Does the toolchain clang or simulator cuttlefish support the V extension already?

the AOSP clang does, yes, but the AOSP kernel does not yet (we need the _next_ AOSP clang to be able to compile kernels that have V support, and we need to cherrypick the V support from 6.5 back to 6.4 which is what we're currently using).

rbosetti, 2023-06-15:
Adding some more color here. We're currently using custom kernels for testing, and AOSP clang for builds.

For folks not plugged directly into the AOSP environment, [there are some instructions](https://wiki.riseproject.dev/display/HOME/Basic+RISCV64+userspace+environment) on the RISE Wiki on producing a test environment

enh-google, 2023-06-15:
> For folks not plugged directly into the AOSP environment, [there are some instructions](https://wiki.riseproject.dev/display/HOME/Basic+RISCV64+userspace+environment) on the RISE Wiki on producing a test environment

seems like you need to create an account and sign in to see that? worth having something on this wiki for easier access?