google/android-riscv64 #18: llvm: missing libunwind support

enh-google, 2023-01-31:
it looks like no-one's tried to upstream https://github.com/riscv-android-src/toolchain-llvm-project/commit/839f6eecc8c9272b8095cb38555ee7954c5bb4f3 yet (and worse, it looks like upstream has rewritten that code quite a bit, so there will be merge conflicts to deal with: https://github.com/llvm/llvm-project/blob/main/libunwind/src/UnwindCursor.hpp).

enh-google, 2023-01-31:
@rprichard - i'm assuming this is actually needed for exception handling unwinds?

rprichard, 2023-01-31:
It looks like that patch ports arm64's special sigreturn handling to riscv. Other architectures (e.g. x86 and x86_64) are able to express sigreturn handling using DWARF .eh_frame instead, so they don't need the special handling.

> @rprichard - i'm assuming this is actually needed for exception handling unwinds?

I don't think C++ exceptions can be thrown beyond a signal handler frame?

enh-google, 2023-02-02:
hmm... @cferris1000 since i thought we had unit tests for sigreturn unwinding? or are they in debuggered (whose tests i haven't yet run) rather than bionic (whose tests i have already run)?

cferris1000, 2023-02-02:
The bionic unit tests do unwinds through signal handlers. Specifically, stack_unwinding.unwind_through_signal_frame. That is using the llvm unwinder, and I think should be using the same code paths in libunwind/src. However, that test is really basic, but I haven't seen a case where it passes and the unwinding is wrong.

Lifang-XIA, 2023-02-03:
We have supported in clang-12, which used in Android-12.
https://github.com/riscv-android-src/toolchain-llvm-project/commit/839f6eecc8c9272b8095cb38555ee7954c5bb4f3

enh-google, 2023-02-03:
yeah, i linked to that in the first comment ... but it looks like it wasn't upstreamed? the current copy of that file  (https://github.com/llvm/llvm-project/blob/main/libunwind/src/UnwindCursor.hpp) doesn't seem to  contain your change?

enh-google, 2023-09-12:
@MaoHan001 --- did this get upstreamed yet?

appujee, 2023-09-13:
doesn't look like it has been upstreamed.

closed 2023-09-13 by appujee