google/android-riscv64 #20: lld: investigate state of linker relaxation

enh-google, 2023-01-31:
alibaba had questions (what, exactly?) about the state of linker relaxation. 

@maskrey in https://maskray.me/blog/2022-07-10-riscv-linker-relaxation-in-lld had at least one suggestion for further work, but aiui he's saying we should _not_ have gp relaxation (which isn't useful for realistic modern systems) and free up a register: https://github.com/riscv-non-isa/riscv-elf-psabi-doc/issues/298

topperc, 2023-02-14:
First patch for GP relaxation is here https://reviews.llvm.org/D143673 It's controlled by a command line option. On Linux GP is already part of the ABI so I don't know if we could remove it. For Android, maybe we have flexibility?

MaskRay, 2023-02-14:
> First patch for GP relaxation is here https://reviews.llvm.org/D143673 It's controlled by a command line option. On Linux GP is already part of the ABI so I don't know if we could remove it. For Android, maybe we have flexibility?

It's not in the ABI. https://reviews.llvm.org/D143673#4116403 GNU ld performs this relaxation without an option, but it does not mean that lld should default to this behavior.

enh-google, 2023-02-14:
(and, yes, _until_ we start shipping an NDK and a riscv64 CTS, we do have the flexibility to have differences from other OS' ABIs. but once we ship, we're as stuck with ours as they are with theirs.)

enh-google, 2023-02-14:
> It's not in the ABI. https://reviews.llvm.org/D143673#4116403 GNU ld performs this relaxation without an option, but it does not mean that lld should default to this behavior.

maskray: you mean "in general", right? or is there a reason why we shouldn't make this the default _for Android_?

topperc, 2023-02-14:
@MaskRay does ld still do the relaxation if the R_RISCV_RELAX isn't paired with the relocation? Or is lack of R_RISCV_RELAX considered ill-formed?

MaskRay, 2023-02-14:
I am trying to push others to benchmark (a) GP relaxation vs (b) no GP relaxation vs (c) no GP relaxation with GP as an allocatable register... For what I can see, some are exploring the idea using GP for other purposes. If lld defaults to GP relaxation, it will make these users inconvenient. Also, we have very little information about how GP relaxation decreases the code size and why.

topperc, 2023-02-14:
> I am trying to push others to benchmark (a) GP relaxation vs (b) no GP relaxation vs (c) no GP relaxation with GP as an allocatable register... For what I can see, some are exploring the idea using [...]

allocatable as callee saved or caller saved?

enh-google, 2023-06-28:
we've reused gp as the shadow stack pointer on Android for now, so i don't think there's anything left to do here? (reopen with a comment if i'm missing something!)

closed 2023-06-28 by enh-google