google/android-riscv64 #9: clang: check that the global clang riscv64 defaults make sense for Android

enh-google, 2023-01-31:
we've typically had bits of Android-specific configuration for the llvm targets, and we'll obviously want that to specify our base ABI, but we should run `clang -v` or whatever and manually check that _all_ the defaults make sense.

appujee, 2023-02-08:
For C++:

```text
/prebuilts/clang/host/linux-x86/clang-r475365/bin/clang++ -v b.cpp -target riscv64-linux-android -c
Android (9342159, based on r475365) clang version 16.0.1 (https://android.googlesource.com/toolchain/llvm-project e1a2893d16330144736de1f4f0d06a844a03b9b0)
Target: riscv64-unknown-linux-android
Thread model: posix
InstalledDir: /prebuilts/clang/host/linux-x86/clang-r475365/bin
 (in-process)
 "/prebuilts/clang/host/linux-x86/clang-r475365/bin/clang-16" -cc1 -triple riscv64-unknown-linux-android -emit-obj -mrelax-all --mrelax-relocations -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name b.cpp -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=all -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature +relax -target-feature -save-restore -target-abi lp64d -msmall-data-limit 8 -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -fcoverage-compilation-dir=/aosp-master-with-phones -resource-dir /aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r475365/lib/clang/16.0.1 -internal-isystem /prebuilts/clang/host/linux-x86/clang-r475365/lib/clang/16.0.1/include -internal-isystem /usr/local/include -internal-externc-isystem /include -internal-externc-isystem /usr/include -fdeprecated-macro -fdebug-compilation-dir=/aosp-master-with-phones -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fcolor-diagnostics -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o b.o -x c++ b.cpp
```

appujee, 2023-02-08:
For C  programs:

> "/prebuilts/clang/host/linux-x86/clang-r475365/bin/clang-16" -cc1 -triple riscv64-unknown-linux-android -emit-obj -mrelax-all --mrelax-relocations -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name b.c -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=all -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature +relax -target-feature -save-restore -target-abi lp64d -msmall-data-limit 8 -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -fcoverage-compilation-dir=/aosp-master-with-phones -resource-dir /prebuilts/clang/host/linux-x86/clang-r475365/lib/clang/16.0.1 -internal-isystem /prebuilts/clang/host/linux-x86/clang-r475365/lib/clang/16.0.1/include -internal-isystem /usr/local/include -internal-externc-isystem /include -internal-externc-isystem /usr/include -fdebug-compilation-dir=/aosp-master-with-phones -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcolor-diagnostics -o b.o -x c b.c
clang -cc1 version 16.0.1 based upon LLVM 16.0.1 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /prebuilts/clang/host/linux-x86/clang-r475365/lib/clang/16.0.1/include
 /usr/local/include
 /usr/include

MaoHan001, 2023-03-22:
> we've typically had bits of Android-specific configuration for the llvm targets, and we'll obviously want that to specify our base ABI, but we should run `clang -v` or whatever and manually check that _all_ the defaults make sense.
> 
> * [ ]  check defaults for C programs at -O0
> * [ ]  check defaults for C++ programs at -O0
> * [ ]  check defaults for C programs at -O3
> * [ ]  check defaults for C++ programs at -O0
> * [ ]  Create separate tasks for changes to defaults as needed.

Duplicate “check defaults for C++ programs at -O0”

enh-google, 2023-03-22:
> Duplicate “check defaults for C++ programs at -O0”

fixed, thanks. i assume the -O3 (which i don't think we use at all) was meant to be -O2 (which we use by default), so i've updated that too.

appujee, 2023-05-03:
With the latest build that is probably close to the release here are the flags

$ clang-10059077/bin/clang -O0 -target riscv64-linux-android10000 test.c -c -v

```
Android (10059077, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362)
Target: riscv64-unknown-linux-android10000
Thread model: posix
InstalledDir: /usr/local/google/home/appujee/g/binaries/clang-10059077/bin
 (in-process)
 "/usr/local/google/home/appujee/g/binaries/clang-10059077/bin/clang-17" -cc1 -triple riscv64-unknown-linux-android10000 -emit-obj -mrelax-all -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name test.c -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=all -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-cpu generic-rv64 -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature -e -target-feature -h -target-feature -zihintpause -target-feature -zfhmin -target-feature -zfh -target-feature -zfinx -target-feature -zdinx -target-feature -zhinxmin -target-feature -zhinx -target-feature -zba -target-feature -zbb -target-feature -zbc -target-feature -zbs -target-feature -zbkb -target-feature -zbkc -target-feature -zbkx -target-feature -zknd -target-feature -zkne -target-feature -zknh -target-feature -zksed -target-feature -zksh -target-feature -zkr -target-feature -zkn -target-feature -zks -target-feature -zkt -target-feature -zk -target-feature -zmmul -target-feature -v -target-feature -zvl32b -target-feature -zvl64b -target-feature -zvl128b -target-feature -zvl256b -target-feature -zvl512b -target-feature -zvl1024b -target-feature -zvl2048b -target-feature -zvl4096b -target-feature -zvl8192b -target-feature -zvl16384b -target-feature -zvl32768b -target-feature -zvl65536b -target-feature -zve32x -target-feature -zve32f -target-feature -zve64x -target-feature -zve64f -target-feature -zve64d -target-feature -zicbom -target-feature -zicboz -target-feature -zicbop -target-feature -svnapot -target-feature -svpbmt -target-feature -svinval -target-feature -xventanacondops -target-feature -xtheadvdot -target-feature -experimental-zihintntl -target-feature -experimental-zca -target-feature -experimental-zcd -target-feature -experimental-zcf -target-feature -experimental-zvfh -target-feature -experimental-zawrs -target-feature -experimental-ztso -target-feature +relax -target-feature -save-restore -target-abi lp64d -msmall-data-limit 8 -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -fcoverage-compilation-dir=/usr/local/google/home/appujee/g/binaries -resource-dir /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17 -internal-isystem /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17/include -internal-isystem /usr/local/include -internal-externc-isystem /include -internal-externc-isystem /usr/include -O0 -fdebug-compilation-dir=/usr/local/google/home/appujee/g/binaries -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcolor-diagnostics -o test.o -x c test.c
clang -cc1 version 17.0.2 based upon LLVM 17.0.2 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17/include
 /usr/local/include
 /usr/include
```

$ ./clang-10059077/bin/clang -O2 -target riscv64-linux-android10000 test.c -c -v 
```
Android (10059077, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362)
Target: riscv64-unknown-linux-android10000
Thread model: posix
InstalledDir: /usr/local/google/home/appujee/g/binaries/clang-10059077/bin
 (in-process)
 "/usr/local/google/home/appujee/g/binaries/clang-10059077/bin/clang-17" -cc1 -triple riscv64-unknown-linux-android10000 -emit-obj -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name test.c -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=none -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-cpu generic-rv64 -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature -e -target-feature -h -target-feature -zihintpause -target-feature -zfhmin -target-feature -zfh -target-feature -zfinx -target-feature -zdinx -target-feature -zhinxmin -target-feature -zhinx -target-feature -zba -target-feature -zbb -target-feature -zbc -target-feature -zbs -target-feature -zbkb -target-feature -zbkc -target-feature -zbkx -target-feature -zknd -target-feature -zkne -target-feature -zknh -target-feature -zksed -target-feature -zksh -target-feature -zkr -target-feature -zkn -target-feature -zks -target-feature -zkt -target-feature -zk -target-feature -zmmul -target-feature -v -target-feature -zvl32b -target-feature -zvl64b -target-feature -zvl128b -target-feature -zvl256b -target-feature -zvl512b -target-feature -zvl1024b -target-feature -zvl2048b -target-feature -zvl4096b -target-feature -zvl8192b -target-feature -zvl16384b -target-feature -zvl32768b -target-feature -zvl65536b -target-feature -zve32x -target-feature -zve32f -target-feature -zve64x -target-feature -zve64f -target-feature -zve64d -target-feature -zicbom -target-feature -zicboz -t [... cut ...] al/google/home/appujee/g/binaries -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcolor-diagnostics -vectorize-loops -vectorize-slp -o test.o -x c test.c
clang -cc1 version 17.0.2 based upon LLVM 17.0.2 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17/include
 /usr/local/include
 /usr/include
End of search list.
```

appujee, 2023-05-03:
$ clang-10059077/bin/clang++ -target riscv64-linux-android10000 test.cpp -c -v 

> Android (10059077, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362)
Target: riscv64-unknown-linux-android10000
Thread model: posix
InstalledDir: /usr/local/google/home/appujee/g/binaries/clang-10059077/bin
 (in-process)
 "/usr/local/google/home/appujee/g/binaries/clang-10059077/bin/clang-17" -cc1 -triple riscv64-unknown-linux-android10000 -emit-obj -mrelax-all -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name test.cpp -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=all -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-cpu generic-rv64 -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature -e -target-feature -h -target-feature -zihintpause -target-feature -zfhmin -target-feature -zfh -target-feature -zfinx -target-feature -zdinx -target-feature -zhinxmin -target-feature -zhinx -target-feature -zba -target-feature -zbb -target-feature -zbc -target-feature -zbs -target-feature -zbkb -target-feature -zbkc -target-feature -zbkx -target-feature -zknd -target-feature -zkne -target-feature -zknh -target-feature -zksed -target-feature -zksh -target-feature -zkr -target-feature -zkn -target-feature -zks -target-feature -zkt -target-feature -zk -target-feature -zmmul -target-feature -v -target-feature -zvl32b -target-feature -zvl64b -target-feature -zvl128b -target-feature -zvl256b -target-feature -zvl512b -target-feature -zvl1024b -target-feature -zvl2048b -target-feature -zvl4096b -target-feature -zvl8192b -target-feature -zvl16384b -target-feature -zvl32768b -target-feature -zvl65536b -target-feature -zve32x -target-feature -zve32f -target-feature -zve64x -target-feature -zve64f -target-feature -zve64d -target-feature -zicbom -target-feature -zicboz -target-feature -zicbop -target-feature -svnapot -target-feature -svpbmt -target-feature -svinval -target-feature -xventanacondops -target-feature -xtheadvdot -target-feature -experimental-zihintntl -target-feature -experimental-zca -target-feature -experimental-zcd -target-feature -experimental-zcf -target-feature -experimental-zvfh -target-feature -experimental-zawrs -target-feature -experimental-ztso -target-feature +relax -target-feature -save-restore -target-abi lp64d -msmall-data-limit 8 -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -fcoverage-compilation-dir=/usr/local/google/home/appujee/g/binaries -resource-dir /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17 -internal-isystem /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17/include -internal-isystem /usr/local/include -internal-externc-isystem /include -internal-externc-isystem /usr/include -fdeprecated-macro -fdebug-compilation-dir=/usr/local/google/home/appujee/g/binaries -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fcolor-diagnostics -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o test.o -x c++ test.cpp
clang -cc1 version 17.0.2 based upon LLVM 17.0.2 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17/include
 /usr/local/include
 /usr/include
End of search list.

$ ./clang-10059077/bin/clang++ -target riscv64-linux-android10000 test.cpp -c -v  -O2

> Android (10059077, +pgo, +bolt, +lto, -mlgo, based on r487747c) clang version 17.0.2 (https://android.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362)
Target: riscv64-unknown-linux-android10000
Thread model: posix
InstalledDir: /usr/local/google/home/appujee/g/binaries/clang-10059077/bin
 (in-process)
 "/usr/local/google/home/appujee/g/binaries/clang-10059077/bin/clang-17" -cc1 -triple riscv64-unknown-linux-android10000 -emit-obj -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name test.cpp -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=none -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-cpu generic-rv64 -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature -e -target-feature -h -target-feature -zihintpause -target-feature -zfhmin -target-feature -zfh -target-feature -zfinx -target-feature -zdinx -target-feature -zhinxmin -target-feature -zhinx -target-feature -zba -target-feature -zbb -target-feature -zbc -target-feature -zbs -target-feature -zbkb -target-feature -zbkc -target-feature -zbkx -target-feature -zknd -target-feature -zkne -target-feature -zknh -target-feature -zksed -target-feature -zksh -target-feature -zkr -target-feature -zkn -target-feature -zks -target-feature -zkt -target-feature -zk -target-feature -zmmul -target-feature -v -target-feature -zvl32b -target-feature -zvl64b -target-feature -zvl128b -target-feature -zvl256b -target-feature -zvl512b -target-feature -zvl1024b -target-feature -zvl2048b -target-feature -zvl4096b -target-feature -zvl8192b -target-feature -zvl16384b -target-feature -zvl32768b -target-feature -zvl65536b -target-feature -zve32x -target-feature -zve32f -target-feature -zve64x -target-feature -zve64f -target-feature -zve64d -target-feature -zicbom -targ [... cut ...] ed-char -fgnuc-version=4.2.1 -fcxx-exceptions -fexceptions -fcolor-diagnostics -vectorize-loops -vectorize-slp -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o test.o -x c++ test.cpp
clang -cc1 version 17.0.2 based upon LLVM 17.0.2 default target x86_64-unknown-linux-gnu
ignoring nonexistent directory "/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/local/google/home/appujee/g/binaries/clang-10059077/lib/clang/17/include
 /usr/local/include
 /usr/include
End of search list.

appujee, 2023-05-04:
At `-O0`, the interesting differences between aarch64-unknown-linux-android10000 and riscv64-unknown-linux-android10000 are:

For C programs,
Command line used:
`clang -O0 -target aarch64-linux-android10000 test.c -c -v `
`clang -O0 -target riscv64-linux-android10000 test.c -c -v `

| aarch64      | riscv64 |
| ----------- | ----------- |
| -mframe-pointer=non-leaf      | -mframe-pointer=all       |
|  -funwind-tables=2   |        |
|  -target-abi aapcs   | -target-abi lp64d        |
|    | -msmall-data-limit 8        |
||-target-feature -save-restore|
||-target-feature +relax|
| -target-feature +outline-atomics |  |
|-D__GCC_HAVE_DWARF2_CFI_ASM=1| |
 
For C++ programs,
Command line used:
`clang -O0 -target aarch64-linux-android10000 test.cpp -c -v `
`clang -O0 -target riscv64-linux-android10000 test.cpp -c -v `

appujee, 2023-05-04:
At `-O2`, the interesting differences between aarch64-unknown-linux-android10000 and riscv64-unknown-linux-android10000 are:

For C programs,
Command line used:
`clang -O2 -target aarch64-linux-android10000 test.c -c -v `
`clang -O2 -target riscv64-linux-android10000 test.c -c -v `

**Note:** The `target-feature` flags that specify the abi are ignored for this exercise.

| aarch64      | riscv64 |
| ----------- | ----------- |
| -mframe-pointer=non-leaf      | -mframe-pointer=none       |
|  -funwind-tables=2   |        |
|  -target-abi aapcs   | -target-abi lp64d        |
|    | -msmall-data-limit 8        |
||-target-feature -save-restore|
||-target-feature +relax|
| -target-feature +outline-atomics |  |

 
For C++ programs,
Command line used:
`clang -O2 -target aarch64-linux-android10000 test.cpp -c -v `
`clang -O2 -target riscv64-linux-android10000 test.cpp -c -v `

| aarch64      | riscv64 |
| ----------- | ----------- |
| -mframe-pointer=non-leaf      | -mframe-pointer=none       |
|  -funwind-tables=2   |        |
|  -target-abi aapcs   | -target-abi lp64d        |
|    | -msmall-data-limit 8        |
||-target-feature -save-restore|
||-target-feature +relax|
| -target-feature +outline-atomics |  |

appujee, 2023-06-15:
Now that https://reviews.llvm.org/D151512 has landed, the remaining differences in compiler flags w.r.t. AArch64 seem inconsequential.

closed 2023-06-15 by appujee