google/android-riscv64 #22: llvm: make LTO work

enh-google, 2023-02-01:
(internal bug http://b/254713216)

currently worked around by disabling LTO, but we should fix that.

enh-google, 2023-02-01:
@appujee

pvchupin, 2023-02-14:
@appujee, @pirama-arumuga-nainar, can we have bug description added here?

topperc, 2023-02-14:
ABI and mcpu/march for LTO is currently broken

Related
https://reviews.llvm.org/D132843
https://reviews.llvm.org/D71387
https://reviews.llvm.org/D72245
https://reviews.llvm.org/D102582
https://reviews.llvm.org/D106347

appujee, 2023-02-15:
we were able to build with lto when emulated tls is disabled.

appujee, 2023-02-15:
> ABI and mcpu/march for LTO is currently broken
> 
> Related https://reviews.llvm.org/D132843 https://reviews.llvm.org/D71387 https://reviews.llvm.org/D72245 https://reviews.llvm.org/D102582 https://reviews.llvm.org/D106347

thanks for sharing this. probably one of these patches fixes the warning  (`Hard-float 'd' ABI can't be used for a target that doesn't support the D instruction set extension (ignoring target-abi)`) we are getting when building with thinlto

appujee, 2023-02-15:
created task for mpcu/march issues: https://github.com/google/android-riscv64/issues/61

enh-google, 2023-06-28:
okay, we've re-enabled LTO for the platform (https://android-review.googlesource.com/2591987) and there's #61 for the remaining work here, so closing this bug...

closed 2023-06-28 by enh-google