google/android-riscv64 #26: goldfish: do we need goldfish too?

enh-google, 2023-02-01:
obviously we need cuttlefish for CI (https://github.com/google/android-riscv64/issues/25), but will we need goldfish for Studio use?

alibaba has their goldfish patch up (which we're ignoring for now as we work on cuttlefish): https://android-review.git.corp.google.com/c/platform/external/qemu/+/2142497

at some point we'll have to talk to Studio about this.

MaoHan001, 2023-03-22:
The version of QEMU may also need to be upgraded to support RISC-V V and Zb extensions.

gongwuchun, 2023-12-07:
It would be nice if we have risc-v goldfish which is more friendly to build target and start target.
As @MaoHan001 said, QEMU version in Android Emulator is too old to support latest risc-v extension features.

enh-google, 2023-12-07:
realistically, risc64-on-x86-64 qemu just isn't going to be fast enough to be useful for app developers. (it's pretty slow for OS developers, who already need a super fast machine!)

the current plan for app developers is https://android.googlesource.com/platform/frameworks/libs/binary_translation/

closed 2023-12-07 by enh-google

gongwuchun, 2023-12-07:
Understand, for app developers, yes, with second abi riscv64 on x86_64 emulator + berberis should be fast, I just wonder goldfish more lite than cuttlefish(for OS developers before HW ready) since we don't need to prepare host environment and start host daemons.

enh-google, 2023-12-07:
> I just wonder goldfish more lite than cuttlefish(for OS developers before HW ready) since we don't need to prepare host environment and start host daemons.

no, that's not true. and it would be a _ton_ of extra work because (as mentioned above) goldfish is very old. we're having enough trouble with bugs in qemu 8.1!