google/android-riscv64 #27: kernel: HAVE_EFFICIENT_UNALIGNED_ACCESS

enh-google, 2023-02-01:
bpf will fail when this feature is not enabled, and obviously Android won't support architectures where this isn't true, but the upstream situation seems unclear?

earlier lkml thread: https://lore.kernel.org/all/mhng-3469a0a8-a337-4b56-8e4f-b5a8bc47c4e9@palmerdabbelt-glaptop/

enh-google, 2023-09-12:
we've added a CTS test for this, and (in 6.6) the kernel actually checks at runtime whether unaligned access is fast or not, so no need to get kconfig right any more.

closed 2023-09-12 by enh-google

ebiggers, 2023-11-16:
Just to double check, Android will require that unaligned loads and stores be fast for both scalar and vector?  The vector crypto code being added to OpenSSL and Linux relies on unaligned vector loads and stores for the input and output data.

enh-google, 2023-11-16:
> Just to double check, Android will require that unaligned loads and stores be fast for both scalar and vector? The vector crypto code being added to OpenSSL and Linux relies on unaligned vector loads and stores for the input and output data.

yes. (though __riscv_hwprobe() doesn't distinguish between the two cases.)

SiFiveHolland, 2023-11-16:
The kernel does not currently probe/know vector unaligned access behavior, so we should probably check that along with adding a separate hwprobe key.

enh-google, 2023-11-16:
since i'm contractually obliged to read all the uapi header changes each release anyway, if/when i see that go in, i'll add it to our CTS test :-)