google/android-riscv64 #28: Studio: gradle support

enh-google, 2023-02-01:
something along the lines of https://android-review.git.corp.google.com/c/platform/tools/base/+/2280530

unicornx, 2023-02-06:
> something along the lines of https://android-review.git.corp.google.com/c/platform/tools/base/+/2280530

I can not access this URL, can you please provide some short description here?

enh-google, 2023-02-06:
i've fixed the url so it should work for everyone, since it's an AOSP change.

enh-google, 2023-08-14:
internal change https://googleplex-android-review.git.corp.google.com/c/platform/tools/base/+/23372540 (that link will only work for google employees :-( ) was submitted 2023-06-06 followed by https://googleplex-android-review.git.corp.google.com/c/platform/tools/base/+/23732318 on 2023-06-20.

even though (afaict) these _changes_ aren't public yet, they are in the current Android Studio Hedgehog Canary.

you'll need an NDK with riscv64 support, and you'll need to edit your app build.gradle.kts file so ndkPath points to it. the one bug i hit was that i had to manually add an ndkVersion lines as well:
```
    ndkPath = "/Applications/AndroidNDK10647936.app/Contents/NDK"
    ndkVersion = "27.0.10647936"
```
because otherwise i get
```
[CXX1100] android.ndkVersion is [25.1.8937393] but android.ndkPath /Applications/AndroidNDK10647936.app/Contents/NDK refers to a different version [27.0.10647936].
```
(where 25.1.25.1.8937393 is the current default NDK version.)

but this should be enough to play with for now.