google/android-riscv64 #31: external/libopus/: do we need optimization? (probably not?)

enh-google, 2023-02-01:
external/libopus/ has optimization for arm/x86/x86-64, but _not_ for arm64, so i'm assuming this isn't interesting any more?

(check with essick, since he's in OWNERS.)

rbosetti, 2023-06-06:
I was able to get some time yesterday to run cycle-comparison benchmarks. For the test I used the QEMU 8.0.2 on an x64 workstation emulating ARMv8 and RV64GCV, confined to four physical cores.

Starting from a sample in the Opus test vectors, I copied the contents to a ramfs test file and ran a tight loop of `opus_demo -d`  1000x to a second ramfs /tmp output. Mean decode cost is within 2%, 95% CI [1.6, 2.4], with the biggest outlier run being an ARMv8 iteration that took ~5% longer than the rest of its cohort. Measurements used the `perf` tool and normalized userspace cycle counts to emulator reported clock rate on a bare Debian image with no `thermald` or other policy engines active.

I think we can safely leave this library for later work. The Android use cases I'm aware of are all about decoding a single stream, rather than game-like sound design scenarios with 16+ active decodes.

closed 2023-06-06 by enh-google

nick-knight, 2023-06-06:
QEMU is a functional simulator, not a cycle-accurate (or even -approximate) one. And the RISC-V ISA has pretty minimal requirements on `rdcycle`, mainly just that it is increasing. IIRC, QEMU implements `rdcycle` on X86 as `rtsc`. Thus, the cycle counts you observe may say more about QEMU's JIT ("TCG"), and the resulting X86 code, than about the original RISC-V source.

rbosetti, 2023-06-06:
A fair point, but I think we can still be fairly conservative here given the relatively mature state of AArch64 in QEMU. I'll work through the same examples with spike but will need a bit more walltime :-)

nick-knight, 2023-06-06:
I'm not sure Spike will give you any better information: it is also just a functional simulator. 

I don't know what AArch64 QEMU does. Perhaps it has more sophisticated uarch models than RISC-V.

What is commonly done, especially at HW companies in exploring new uarch designs, is use a functional simulator like Spike/QEMU/etc. to dump an instruction trace, and then feed that into a model of their target uarch. Of course, someone has to develop that model.

Anyway, I'm happy with your conclusion, I'm not suggesting reopening this.

rbosetti, 2023-06-06:
ack. I appreciate the feedback Nick.

re: use of spike, it gives me a lot less noise in the measurements in exchange for needing a bit more setup. We don't currently have a uarch to target, so these optimization efforts are rough cuts. I expect to reopen some of them when our work makes contact with silicon :-)