google/android-riscv64 #35: external/libjpeg-turbo: optimization

enh-google, 2023-02-02:
missing optimized code using V extension, using portable C fallback for now.

rayehandes, 2023-05-31:
We (Andes Technology) are working on this item now.
If we done the imlementation, we will contribute the source.

enh-google, 2023-06-05:
> We (Andes Technology) are working on this item now. If we done the imlementation, we will contribute the source.

great to hear! it looks like we (Android) haven't been doing a good job of updating from chrome (https://chromium.googlesource.com/chromium/deps/libjpeg_turbo is our upstream), but Chrome _has_ been doing a good job of keeping in sync with the "true" upstream, so it shouldn't be a problem to pull your fixes when they're in!