google/android-riscv64 #38: external/libyuv: optimization

enh-google, 2023-02-02:
there's lots of inline assembler for arm/intel; we'll want similar for risc-v.

bhbruce, 2023-06-08:
Hi, I am Bruce from SiFive. 
We've enabled the RISC-V vector in Libyuv and upstreamed several RVV kernels.

enh-google, 2023-06-08:
nice! thanks for letting us know. it looks like chromium already updated (they certainly have your "[RVV] Use LMUL=2 for I4{44,22}To{ARGB,RGB24,RGBA} conversion" change), so i'll sync AOSP with chromium: https://android-review.googlesource.com/c/platform/external/libyuv/+/2620612

enh-google, 2023-06-08:
ah, the good news is that this has already been updated internally. the bad news is that we'll have to cherrypick those changes out to avoid merge conflicts, which will probably take a bit longer...

enh-google, 2023-06-12:
someone who works on libyuv did the cherrypick: https://android-review.googlesource.com/c/platform/external/libyuv/+/2622518

enh-google, 2023-07-26:
this doesn't build for me if i enable V in the build system:
```
external/libyuv/files/source/row_rvv.cc:277:36: error: variable 'vl' is uninitialized when used here [-Werror,-Wuninitialized]
  v_a = __riscv_vmv_v_x_u8m2(255u, vl);
                                   ^~
external/libyuv/files/source/row_rvv.cc:269:12: note: initialize the variable 'vl' to silence this warning
  size_t vl;
           ^
            = 0
external/libyuv/files/source/row_rvv.cc:365:36: error: variable 'vl' is uninitialized when used here [-Werror,-Wuninitialized]
  v_a = __riscv_vmv_v_x_u8m2(255u, vl);
                                   ^~
external/libyuv/files/source/row_rvv.cc:357:12: note: initialize the variable 'vl' to silence this warning
  size_t vl;
           ^
            = 0
external/libyuv/files/source/row_rvv.cc:425:36: error: variable 'vl' is uninitialized when used here [-Werror,-Wuninitialized]
  v_a = __riscv_vmv_v_x_u8m2(255u, vl);
                                   ^~
external/libyuv/files/source/row_rvv.cc:417:12: note: initialize the variable 'vl' to silence this warning
  size_t vl;
           ^
            = 0
```

other routines have something like `size_t vl = __riscv_vsetvl_e8m2(w);` but the functions with errors seem to rely on `YUVTORGB_SETUP()` to initialize vl, but it doesn't?

enh-google, 2023-07-26:
i'll get this patch into AOSP until we get this fixed properly, but for anyone who hits this between now and me getting this merged...
```
diff --git a/files/Android.bp b/files/Android.bp
index 9811a7b..96bc7cb 100644
--- a/files/Android.bp
+++ b/files/Android.bp
@@ -82,6 +82,7 @@ cc_library {
         "-Wno-unused-parameter",
         "-fexceptions",
         "-DHAVE_JPEG",
+        "-DLIBYUV_DISABLE_RVV",
     ],
 
     arch: {
```

enh-google, 2023-07-27:
the good news is that it looks like this was fixed by https://chromium.googlesource.com/libyuv/libyuv/+/7939e039e76072bc687128fc5c1f10fe04f7858c which (as you can tell from the link) is already in AOSP's upstream.

i'll go and pester Vignesh Venkatasubramanian again for another AOSP update...

bhbruce, 2023-07-28:
Hi @enh-google,
I have solved this bug as you mentioned. I am curious about which version of the clang you are using now.

enh-google, 2023-07-28:
right now, it's this:
```
Android (10386919, +pgo, +bolt, +lto, -mlgo, based on r498229) clang version 17.0.3 (https://android.googlesource.com/toolchain/llvm-project e34ed7d63863b45858e74126edaa738e75887800)
Target: x86_64-unknown-linux-gnu
Thread model: posix
InstalledDir: /usr/local/google/home/enh/aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r498229/bin
```

appujee, 2023-08-03:
Related discussion today: https://lists.riscv.org/g/sig-android/topic/100508949#270

bhbruce, 2023-08-08:
@enh-google I would suggest to disable RVV until this [ticket](https://bugs.chromium.org/p/libyuv/issues/detail?id=965) is solved.

enh-google, 2023-08-08:
yeah, although i built a fixed qemu locally to play with, until we have a working prebuilt qemu i don't think we can do anything useful with V anyway -- it would be unreasonable to expect everyone to build their own patched qemu! -- so i've put my V enablement patches to the side for now.

in parallel, we have someone working on fixing the way libyuv updates make it into AOSP so we should be able to track upstream much more quickly and easily.

prashanthswami, 2024-02-13:
From what I can tell, RVV support has safely landed in libyuv and has been there for a while - is there anything further we want to track in terms of optimization?

enh-google, 2024-02-14:
yeah, lgtm... @bhbruce knows where to find us any time he needs us to take another update :-)