google/android-riscv64 #39: external/skia: optimization

enh-google, 2023-02-02:
lots of neon/sse intrinsics; we'll want similar for risc-v.

rbosetti, 2023-04-14:
Sorry for the delay here. I'm working on finding a Google counterpart to review and take patches. That may end up being me :)

brianosman, 2023-10-05:
There are a couple folks on Skia (including myself) that are able to review patches here.

MaoHan001, 2023-10-07:
We have currently completed optimizations related to blit, and found that the SkNx class does not support variable-length SIMD optimization. We would like to know if Google has considered supporting variable-length SIMD optimization in the Skia framework for future iterations.

brianosman, 2023-10-09:
We have discussed variable-length SIMD, but we have no concrete plans right now. Note that SkNx was removed completely in 2022, and all usage was replaced with skvx. skvx has a simpler implementation in many ways, but still doesn't work well for architectures like RISC-V or SVE.

More importantly - the most complicated SIMD code in Skia is the Raster Pipeline (RP). That system is used for nearly all complex CPU drawing, and it's not clear to me how the design would work with something like RISC-V. RP is implemented like a fixed-width SIMT interpreter, shading spans of pixels. The logic is implemented in dozens of (small) functions with an identical ABI (8 vector registers, plus some scalar registers). The CPU backend examines a drawing operation, and constructs a sequence of these "stage" functions to implement the operations. Each stage function tail-calls into the next one in the sequence. Because of the tail-call optimizations and the identical ABI, there is no stack usage (except when a particular stage function requires it for local computation). The "state" of the miniature machine (those 8 vectors + scalars) just continue to pass from stage to stage in registers.