google/android-riscv64 #40: external/freetype: optimization (?)

enh-google, 2023-02-02:
src/smooth/ftgrays.c has one bit of arm64-specific configuration. check whether we need similar for rv64. (there's a lot more stuff for intel, but if arm64 hasn't needed it, does it matter?)

prashanthswami, 2023-11-10:
A quick cursory look at all references to arm/i386 and aarch64:

1. `include/freetype/internal/ftcalc.h`: The definition of `FTMulFix` (which computes `a*b/0x10000`), is optimized for arm/i386 but not aarch64. On 32-bit platforms, the signed multiply of two 32-bit numbers would need to be clever about storing the result in two 32-bit registers, doing the arithmetic right shift, etc. On 64-bit platforms, this problem is moot - you could simply multiply them normally and do the add-subtract-shift in place, so the compiler is likely to optimize this correctly. It seems like this is what aarch64 does in `src/base/ftcalc.c`.

2. `src/truetype/ttinterp.c`: Contains the definition of `FTMulFix14`. This could possibly be optimized - the code supplies three implementations: a) for ARM, which does the 32-bit carry, b) for gnuc + x86[_64], which does the simple multiply and shift and c) the default, which does the 32-bit carry in plain C.

At a glance, (B) seems more appropriate, because the register widths are fine, but the concern is whether the right-shift operator performs an arithmetic-right-shift. The claim is made in this code that this is always true for x86/x86_64 on GNU. Clang doesn't seem so forgiving, but someone may want to take a closer look.

For the same reason, the equivalent function `TT_DotFix14_long_long` could maybe not be optimized on clang. One might argue that we could make this change on C++20, eventually.

3.  `src/smooth/ftgrays.c` contains an ARM specific fix, but this is to deal with a very specific GNU compiler bug. This can be skipped.

4. `src/smooth/ftgrays.c` also contains an optimization for Bezier arcs, requires re-implementing in RISC-V vectors.

5. `src/base/md5.c` there's a block of code that implies this library can have correctness issues on X86 platforms without re-aligning some structures, so we should analyze if the same is true for RISC-V.

6. `src/gzip/crc32.c.` At a glance, the word size here is incorrect for RISCV64 platforms and is non-optimal. There's a CRC32 processing function leverage in ARM builds - we could potentially accelerate this via `clmulr`.

nick-knight, 2023-11-10:
I'll take care of the CRC32. I have an implementation for Zlib(-ng) that I've been meaning to upstream, but keep getting distracted by other projects. Should be relatively easy to put here as well.

enh-google, 2023-11-10:
scudo can use an optimized CRC32 too --- https://github.com/google/android-riscv64/issues/2. (though strictly scudo doesn't care that it's _specifically_ CRC32, if there's something similar-but-cheaper that makes more sense for riscv64.)

nick-knight, 2023-11-10:
Thanks for the reminder. I think I promised to address Scudo too.

apodtele, 2024-01-09:
Thanks for the review. The SSE2 stuff in `src/smooth/ftgrays.c` does not have any benefits frankly. I consider removing git.

enh-google, 2024-01-09:
(for md5, if that's actually important to freetype, we should define HAVE_OPENSSL for _all_ the architectures rather than touch `md5.c`...)

apodtele, 2024-01-09:
The md5 stuff is only used to detect rendering differences for debugging purposes, not used in production builds at all. The crc32 stuff is borrowed together with zlib and used only when system library is missing.

enh-google, 2024-01-09:
ah, yes, you're right --- we're already not building `crc32.c` for Android, for any architecture.

prashanthswami, 2024-01-10:
If we don't use `crc32.c` and `md5.c` is only used for debugging, can we drop 5 and 6 in the list? Looking at the remaining, can someone confirm:

1. Nothing to do here, likely auto-optimized.
2. Nothing to do here, if AARCH64 has no optimization we probably don't need one either.
3. Nothing to do here, GNU-ARM specific bug.
4. Bezier arcs . . .do we need this? @apodtele Is this the 'SSE2' stuff you're suggesting we can just remove?

apodtele, 2024-01-10:
- `calc.h` and `ttinterp.c`: make sure that 64-bit types can be and are used. In other words trust compiler before assembly.
- `ftgrays.c`: my benchmarking (ftbench of freetype-demos) shows that removing SSE2 code does not hurt at all. Please confirm.

The biggest CPU hogs are rendering (ftgrays.c)  and bytecode (ttinterp.c), otherwise FreeType is and should stay C89.

prashanthswami, 2024-01-11:
Just to make sure I understand:

1. "calc.h and ttinterp.c: make sure that 64-bit types can be and are used. In other words trust compiler before assembly." - are you confirming that there's nothing to do here and the compiler will probably deal with it properly? (I agree, but want to make sure we're on the same page).
2. "ftgrays.c: my benchmarking (ftbench of freetype-demos) shows that removing SSE2 code does not hurt at all. Please confirm." Arguably the only question we have here is "are we doing as much optimization as AARCH64 requires, and yeah, SSE2 is irrelevant in the regard.

Perhaps then in order to be a parity with existing AARCH64 implementations, we just need BEZIER_USE_DDA and that's it.

apodtele, 2024-01-11:
I put [a merge request](https://gitlab.freedesktop.org/freetype/freetype/-/merge_requests/314) on the table this morning. It seems that modern compilers outperform SSE2. The difference is so small that even if a counterexample found, it is not worth maintaining the complexity. Please hold on with ftgrays.c or help testing. This is rather straightforward code with a very limited paths for curves and for straight lines, i.e., it is easy to find and solve bottlenecks.

As for ftcalc.h, I suggest looking at the generated assembly for FT_MulFix etc. If you are happy, I am happy. Some of the architecture specific code is there to accommodate compilers without a 64-bit type. If 64-bit types are fast, we should not be doing assembly.

For ttinterp.c, which is a branched interpreter with tons of execution paths, I have no idea where to start. My profiling a few years back showed some bottlenecks. Or, maybe, I misinterpreted profiling. I was never to interested in optimizing it.

prashanthswami, 2024-02-11:
Ack - thanks for the change to ftgrays.c ! I'm going to assume that based on the benchmark results you posted to that bug that maybe we can forgo trying to optimize RISC-V and just trust the compiler there.

Looking at FT_MulFix, it looks reasonable to me:

```
00000000000322e0 <FT_MulFix>:
;     FT_Int64  ab = (FT_Int64)a_ * (FT_Int64)b_;
   322e0: 33 85 a5 02   mul     a0, a1, a0
;     return (FT_Long)( ( ab + 0x8000L - ( ab < 0 ) ) >> 16 );
   322e4: 93 55 f5 43   srai    a1, a0, 63
   322e8: 21 66         lui     a2, 8
;     return (FT_Long)( ( ab + 0x8000L - ( ab < 0 ) ) >> 16 );
   322ea: 32 95         add     a0, a0, a2
   322ec: 2e 95         add     a0, a0, a1
   322ee: 41 85         srai    a0, a0, 16
   322f0: 82 80         ret
```

It maps closely to the x86_64 compiler optimization and is pretty close to the GNU MulFix hand-written assembly in ftcalc.h, so I suspect that's good evidence we're safe here.

In regards to ttinterp.c, I say we revisit it if-and-when we find evidence that it's too slow. If we didn't find a reason to optimize it for aarch64, we should trust the compiler until we learn otherwise.

Unless there are strong reasons to keep pursuing this, can I suggest that there's no further optimization work to track here for RISC-V? Does anyone disagree / want to keep this issue open for further analysis?

enh-google, 2024-02-13:
> Unless there are strong reasons to keep pursuing this, can I suggest that there's no further optimization work to track here for RISC-V? Does anyone disagree / want to keep this issue open for further analysis?

yeah, closing sgtm. (in fact, i look forward to taking the updates that remove some of the historic optimizations for _other_ architectures :-) )

closed 2024-02-13 by enh-google