google/android-riscv64 #42: external/XNNPACK: optimization

enh-google, 2023-02-02:
this whole project is mostly optimized assembler :-)

Zvediv looks like it would be very useful for them.

rednoah91, 2023-03-15:
Looks like there are some RVV kernels are implemented: 
https://github.com/google/XNNPACK/commit/aac51aad45a24f1a1acec0a521ac63b6892028b6#diff-3f710269879983f53830fec90d0963ef9da0382cd4eed8bcaf699e8f4101560b

rednoah91, 2023-04-07:
@fbarchard Are you working on it?

fbarchard, 2023-04-07:
I am, but time frame is roughly by end of year.  I plan to focus on a full set of fp32 microkernels first.

sharvey-img, 2023-08-04:
@fbarchard I see you've already done some work here (e.g. https://github.com/google/XNNPACK/commit/ccfee7043a30babb9e3aec70b342fcc743a260d2 ). Can we pick this up? Are there any non-merged commits?

nick-knight, 2023-08-04:
We (SiFive) are also interested in contributing. My team is in the process of migrating (micro)kernels from our performance libraries into various open-source projects.

JaydeepIMG, 2023-08-08:
Hi @nick-knight  @fbarchard, we at Imagination Technologies are also planning to contribute. Could you please share list of kernels that you are planning to implement? We are planning to start with 8bit quantized operations for RVV.

ManojIMG, 2023-10-17:
Hi, We (Imagination) have created our first PR (QS8/QU8 RVV) here https://github.com/google/XNNPACK/pull/5638 but it seems waiting on the internal review. Is there anything missing from us to get it through? What is the expected turnaround time?