google/android-riscv64 #43: external/lzma: optimization (?)

enh-google, 2023-02-02:
there's arch-specific "custom filter chain" stuff in lzma, but it doesn't look like it was extended to arm64 (and _does_ support sparc!), so it's presumably of historical interest only?

rbosetti, 2023-03-29:
Confirmed that this is not of interest. It doesn't show up on sysui profiles at all and some basic CPI comparisons to an ARMv8 board show that performance on RISC-V is roughly the same.

closed 2023-03-29 by rbosetti

enh-google, 2023-03-29:
yeah, ironically upstream has added various armv8 optimizations recently, but in a way that's _prevented_ me from upgrading (because it would break armv7 builds)! so "not a regression".

nick-knight, 2024-03-29:
Also https://www.openwall.com/lists/oss-security/2024/03/29/4

enh-google, 2024-03-29:
thanks for the heads-up... luckily, that's not the xz we use --- we use xz-embedded. i am not a security person, and i expect lots of better-qualified people will take a look, but provisionally https://github.com/tukaani-project/xz-embedded/commit/48355f10e09c5075f3a56031fcbca18b2a708260 looks to be the only code change there from the same committer, and that seems harmless ((1 << 26) == (64 << 20)). (i'm assuming for the moment that it was a lone committer, but i'll hold off on taking an update to get https://github.com/tukaani-project/xz-embedded/commit/8b2382b5a99550e74eaf07684e026f60601d99c8 for now anyway!)

(also luckily, this seems to be very specifically targeted at distro maintainers, and requires gcc/binutils at build time and glibc at runtime. but obviously _in principle_ everyone's at risk of supply-chain attacks. we were just lucky this time. i dread the day someone attacks something the size and complexity of llvm ... assuming they haven't already, and it's just that no-one noticed yet. https://research.swtch.com/nih is a great read, even if you think you already know the "reflections on trusting trust" story. there are interesting devils in the details :-) )

enh-google, 2024-04-08:
(see https://github.com/google/android-riscv64/issues/139#issuecomment-2043670053 for a bit more detail on the xz BCJ "optimization". TL;DR: not useful, now disabled for other architectures too.)