google/android-riscv64 #46: llvm: function alignment

enh-google, 2023-02-02:
do we care? is the llvm default good enough? best to leave it to llvm to decide on a per-SoC basis rather than having anything in the build system?

appujee, 2023-02-08:
Investigate if current function alignment of 4 bytes is good for perf benchmarks that we care about: https://github.com/llvm/llvm-project/blob/main/llvm/lib/Target/RISCV/RISCVISelLowering.cpp#L1004

topperc, 2023-02-14:
Default function alignment is 2 bytes with C or Zca extension enabled. Those are the 16-bit instruction extensions.

enh-google, 2023-02-14:
yeah, the question was more "do/will we have SoCs whose performance significantly benefits from _over_-alignment?". (and even if so, do we want to increase the _global_ default alignment, or just annotate specific functions that particularly benefit?)

topperc, 2023-02-28:
I implemented a tiny bit of plumbing so that we could do alignment based on a particular -mtune https://reviews.llvm.org/D144048

appujee, 2023-03-01:
Nice! Should we add test case for this or consider this closed?

topperc, 2023-03-01:
It wasn't a functional change. I just wired the alignment to the file that contains CPU specific information, but I didn't chang the behavior for any CPU.

enh-google, 2023-03-01:
yeah, and if you look at my original questions, they're a lot more philosophical anyway. this provides a _mechanism_ for one of the possible policies (which is great, and useful in its own right, but is kind of orthogonal to the higher-level "what [if anything] should we do here?" question). really i think we need to wait until we have enough hardware variety to be able to actually make a decision.

(the obvious alternative if we really want to close this quickly would be "let's just copy the arm64/x86-64 16-_byte_ alignment and assume that's (a) good enough and (b) where we'll end up anyway", and change that if/when we have evidence to the contrary.)

enh-google, 2023-09-12:
bionic is currently going with the arm64/x86-64 16 _byte_ alignment for assembler functions. i'm assuming we'll just go with whatever llvm does for C/C++...

closed 2023-09-12 by enh-google