google/android-riscv64 #51: ship libomp.a and libomp.so

appujee, 2023-02-03:
both arm64 and x86_64 ship this in `prebuilts/clang/host/linux-x86/clang-r475365/runtimes_ndk_cxx/`

appujee, 2023-02-04:
cc: @pirama-arumuga-nainar do we need to have libomp in RISC-V?

pirama-arumuga-nainar, 2023-02-04:
Yes, I think it'd be good to have libomp to ensure compatibility.  This, and other libraries (fuzzer, lldb-server) are explicitly not built because we don't have a minimum NDK API level and a sysroot for that API level.  When we have some clarity, we should add it to https://android.googlesource.com/toolchain/llvm_android/+/205bdc6e08c820afa2c5380d67044db4b2d62825/configs.py#629

enh-google, 2023-10-03:
prebuilts/clang/host/linux-x86/clang-r498229b/runtimes_ndk_cxx/riscv64 lgtm?

closed 2023-10-03 by enh-google

appujee, 2023-10-03:
yes. https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/refs/heads/main/clang-r498229b/runtimes_ndk_cxx/riscv64/