google/android-riscv64 #53: kernel: crypto optimization

samitolvanen, 2023-02-07:
Linux doesn't have optimized crypto implementations for RISC-V. Storage encryption and dm/fs-verity are going to be unusably slow without optimized versions of AES and SHA-2 at least, preferably using [Scalar Cryptography extensions](https://github.com/riscv/riscv-crypto/releases/tag/v1.0.1-scalar). There's some potential overlap here with issue #36.

samitolvanen, 2023-02-08:
Coincidentally a few relevant patches were posted to linux-riscv a couple of days ago: https://lore.kernel.org/linux-riscv/20230206225846.1381789-1-heiko@sntech.de/

enh-google, 2023-02-08:
(interestingly, that also links to some openssl pull requests that might be useful for boringssl later...)

enh-google, 2023-03-16:
(TIL there's also a CTS test you can't pass without AES instructions: cts/tests/tests/security/native/encryption/FileBasedEncryptionPolicyTest.cpp)

samitolvanen, 2023-03-16:
AFAIK you _can_ pass the test without AES instructions, but you can't use the Adiantum encryption mode unless the test is able to confirm that AES instructions really aren't available.

samitolvanen, 2023-07-17:
The vector crypto patch series is now at v4, is based on the frozen vector crypto spec, and includes both SHA-256 and AES: https://lore.kernel.org/linux-riscv/20230711153743.1970625-1-heiko@sntech.de/

samitolvanen, 2023-10-02:
RVI ratified the vector crypto extension last week, but apparently the original author of the kernel patch series decided to take a break from development and SiFive is planning on taking over:
https://lore.kernel.org/linux-riscv/DD3113B1-AB9F-4D6D-BD6E-8F75A83DA45D@sifive.com/

ebiggers, 2023-11-16:
The new patch series from SiFive is [[PATCH 00/12] RISC-V: provide some accelerated cryptography implementations using vector extensions](https://lore.kernel.org/linux-crypto/20231025183644.8735-1-jerry.shih@sifive.com/T/#u), sent a few weeks ago.  It generally seems to be in good shape, and it adds acceleration for all important algorithms (more or less).  We are working through some details, such as how the use of the vector registers can be supported in softirq context, and whether the assembler can be used for the vector crypto instructions.

ebiggers, 2024-01-23:
The latest patchset is [[PATCH v3 00/10] RISC-V crypto with reworked asm files](https://lore.kernel.org/linux-crypto/20240122002024.27477-1-ebiggers@kernel.org/T/#u).  All prerequisites are upstream now, and we are aiming for the 6.9 kernel for the vector crypto patchset itself.

ebiggers, 2024-03-22:
RISC-V vector crypto optimizations were merged upstream for v6.9-rc1. See https://git.kernel.org/linus/67daf84203a02cf0 and https://git.kernel.org/linus/c150b809f7de2afd. The merged code covers all the basic use cases of kernel crypto. Future work will accelerate some remaining algorithms such as AES-GCM and AES-HCTR2. Fine-tuning may also be needed when this starts being tested on real RISC-V CPUs.

enh-google, 2024-03-22:
nice!

being a kernel n00b, am i interpreting
```
+# This symbol indicates that the toolchain supports all v1.0 vector crypto
+# extensions, including Zvk*, Zvbb, and Zvbc.  LLVM added all of these at once.
+# binutils added all except Zvkb, then added Zvkb.  So we just check for Zvkb.
+config TOOLCHAIN_HAS_VECTOR_CRYPTO
+	def_bool $(as-instr, .option arch$(comma) +zvkb)
+	depends on AS_HAS_OPTION_ARCH
```
and
```
+config CRYPTO_AES_RISCV64
+	tristate "Ciphers: AES, modes: ECB, CBC, CTR, XTS"
+	depends on 64BIT && RISCV_ISA_V && TOOLCHAIN_HAS_VECTOR_CRYPTO
```
correctly as "we don't even need to modify the kernel configuration --- just having a working toolchain will be enough"? or do you have to meet the dependencies _and_ manually configure this on?

SiFiveHolland, 2024-03-22:
Unless specified otherwise, Kconfig options default to being off, so `gki_defconfig` will need updates to enable these options.

ebiggers, 2024-03-22:
Android will need:

```
CONFIG_RISCV_ISA_V=y
CONFIG_CRYPTO_AES_RISCV64=y
CONFIG_CRYPTO_CHACHA_RISCV64=y
CONFIG_CRYPTO_GHASH_RISCV64=y
CONFIG_CRYPTO_SHA256_RISCV64=y
CONFIG_CRYPTO_SHA512_RISCV64=y
```