google/android-riscv64 #55: kernel: software shadow call stack

samitolvanen, 2023-02-07:
Linux doesn't have SCS support for RISC-V (`CONFIG_SHADOW_CALL_STACK`). Similarly to #52, we should support software SCS in the kernel until #14 is available and implemented.

enh-google, 2023-02-10:
https://android-review.git.corp.google.com/q/topic:%22rv64-scs%22 seems to work.

still need to look at making setjmp() not leak x18, but i think that's it?

samitolvanen, 2023-02-13:
Yes, it looks like that would do it for userspace. For the kernel, we'll first have to make sure the existing RISC-V assembly code doesn't use `x18`, and if it's not already done, we'll need to add code to save/restore `x18` during context switch, CPU suspend etc. We probably also need a separate shadow call stack to match the `shadow_stack` the kernel switches to when a stack overflow occurs.

enh-google, 2023-02-16:
i've updated setjmp()/longjmp() too (also on https://android-review.googlesource.com/q/topic:%22rv64-scs%22 ), but i'll wait until adelva's cuttlefish work is in CI before i submit this potentially disruptive topic...

samitolvanen, 2023-08-11:
SCS kernel patches v1: https://lore.kernel.org/linux-riscv/20230811233556.97161-7-samitolvanen@google.com/

samitolvanen, 2023-11-08:
`CONFIG_SHADOW_CALL_STACK` support was merged to v6.7: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit?id=24005d184aaa80984e0511c4ec6e6a0860fdddb8

The feature still needs to be enabled in android-mainline once -rc1 is merged there, and possibly backported to older kernels depending on which kernel versions RISC-V GKI will support.

samitolvanen, 2024-01-17:
Enabled in android-mainline: https://r.android.com/2912646

closed 2024-01-17 by samitolvanen