google/android-riscv64 #57: Fix RISC-V bugs in llvm upstream

appujee, 2023-02-08:
https://github.com/llvm/llvm-project/issues?q=is%3Aopen+label%3Abackend%3ARISC-V++label%3Abug

appujee, 2023-02-10:
- [ ] Introduce `-fno-emulated-tls` by default to clang for RISC-V
- [ ] Error on `-femulated-tls` passed for RISC-V targets

brad0, 2023-03-29:
This has been fixed upstream..

[https://github.com/llvm/llvm-project/commit/f5e63f8fc9f492944b8432d80c1c49051a7d600a](https://github.com/llvm/llvm-project/commit/f5e63f8fc9f492944b8432d80c1c49051a7d600a)

appujee, 2023-03-29:
Thanks for the update, marked both sub tasks as complete.

brad0, 2023-04-02:
> This has been fixed upstream..
> 
> [llvm/llvm-project@f5e63f8](https://github.com/llvm/llvm-project/commit/f5e63f8fc9f492944b8432d80c1c49051a7d600a)

To add to that I have had this back ported to 16.

closed 2023-04-11 by enh-google