google/android-riscv64 #64: crash with RISC-V scalable vectorization and kernel address sanitizer #61096

appujee, 2023-03-01:
Mirroring: https://github.com/llvm/llvm-project/issues/61096

nick-knight, 2023-03-01:
ASan doesn't yet support RVV, LLVM folks (including at SiFive) are working on it.

nick-knight, 2023-03-01:
While we're on the topic, ASan (upstream) makes some Sv39-specific assumptions and has issues on Sv48+. We're also looking at that too.

appujee, 2023-03-06:
> While we're on the topic, ASan (upstream) makes some Sv39-specific assumptions and has issues on Sv48+. We're also looking at that too.

do we have ticket for this in upstream?

appujee, 2023-03-06:
Is https://reviews.llvm.org/D137397 required for enabling asan?

kito-cheng, 2023-03-23:
https://reviews.llvm.org/D145259, https://reviews.llvm.org/D145198 and https://reviews.llvm.org/D145175 should fix this issue.

>> While we're on the topic, ASan (upstream) makes some Sv39-specific assumptions and has issues on Sv48+. We're also looking at that too.
>
> do we have ticket for this in upstream?

Oo ticket in upstream yet, I've made few fix on Sv48 (https://reviews.llvm.org/D139827 and https://reviews.llvm.org/D139823) that make Sv48 work, but not work well on Sv39, so we still not landed that yet.

We are looking for a better solution for adapt different Sv*, and has allocated dedicated resource on that, the goal is make it work well on both Sv39 and Sv48, the longer term goal is make it work well on Sv39, Sv48 and Sv57.