google/android-riscv64 #65: build: vNDK not packaged for RISC-V builds

prashanthswami, 2023-03-03:
When building RISC-V targets, the Cuttlefish images are missing com.android.vndk.* packages, which expose common libraries through the linkerconfig. As they are missing, other packages will hit errors like:

`linker  : CANNOT LINK EXECUTABLE "/apex/com.android ...": library "libcutils.so" not found: needed by main executable`

We can confirm that these libraries are really not present by observing that `/linkerconfig/ld.config.txt` doesn't reference them. These are intended to be provided by com.android.vndk.[some version]. For supported architectures, previous versions of the vNDK already exist, so this works properly. However, RISC-V, which is under development, is reporting a 'future' api level. [api_level.go](https://cs.android.com/android/platform/superproject/+/master:build/soong/cc/api_level.go;l=34).

This causes the vNDK version filter check to filter out all packages. [vndk.go](https://cs.android.com/android/platform/superproject/+/master:build/soong/apex/vndk.go;l=75).

In particular, when evaluating vNDK level 'current', the `vndkVersion` function maps the value to the latest platform version string (e.g. "UpsideDownCake") - [vndk.go](https://cs.android.com/android/platform/superproject/+/master:build/soong/apex/vndk.go;l=51). As the platform version string you build with is always included in the active preview list, `apiLevelFromUser` returns it as-is, and "the latest platform release" (a number in the 9000s) will always have a lower platform version than "current" (10000).

An initial attempt at patching this is [here](https://android-review.git.corp.google.com/c/platform/build/soong/+/2464669), which remaps the latest vNDK version back to current, as is done in other parts of the file. However, we're discussing if it is more appropriate to indicate that the minApiLevel of the target should simply report the latest platform version.

MaoHan001, 2023-03-22:
This link (https://android-review.git.corp.google.com/c/platform/build/soong/+/2464669) is not accessible externally.
 It appears that the issue has been resolved as seen in this link (https://android-review.googlesource.com/c/platform/build/soong/+/2464669) where it has already been merged？

enh-google, 2023-03-22:
> This link (https://android-review.git.corp.google.com/c/platform/build/soong/+/2464669) is not accessible externally. It appears that the issue has been resolved as seen in this link (https://android-review.googlesource.com/c/platform/build/soong/+/2464669) where it has already been merged？

yeah, sorry --- there's some stupid URL rewriting internally that breaks links for non-googlers. any time you see "git.corp.google", trying replacing it with "googlesource". in this specific case, yes, that's the right external link corresponding to the internal link.

i think this is done, but i'll let @prashanthswami close in case there's something still to be done...

closed 2023-03-22 by prashanthswami