google/android-riscv64 #66: external/libaom: optimization

prashanthswami, 2023-03-03:
The x86/ARM architectures make use of SSSE3 and NEON for the codec support in this library. We probably will need some optimization work done to meet parity.

mr-c, 2023-10-30:
FYI: optimized riscv64 vector implementations of NEON intrinsics are coming to SIMDe: 
https://doi.org/10.48550/arXiv.2309.16509
https://github.com/simd-everywhere/simde/issues/1087

(SIMDe already provides portable implementations of SSSE3 (and almost all of NEON) which compile and work on riscv64 with [GCC](https://github.com/simd-everywhere/simde/actions/runs/6673044915/job/18138021991) or [clang](https://github.com/simd-everywhere/simde/actions/runs/6673044915/job/18138020049))

Obviously hand-optimized RVV 1.0 intrinsics would be better, but this can be a stopgap.

prashanthswami, 2024-02-13:
We should probably reach out to the libaom team and kick-start the long-term plan. At the moment, I don't know of anyone who needs this urgently enough that we need to implement a stopgap.

At a glance, it looks like we'll have to do a few operations:

1. We'll have to update libyuv in the aom repository, it's version is old and doesn't contain the RVV intrinsics.
2. We'll have to create equivalent files for aom_dsp/arm by implementing intrinsics for RVV. This is about 60 files to rewrite.