google/android-riscv64 #69: llvm function multi-versioning

enh-google, 2023-03-08:
related to but distinct from https://github.com/google/android-riscv64/issues/8, we'll want function multi-versioning in llvm.

MaoHan001, 2023-03-22:
Is this issue related to the multi-versioning feature itself or to function optimization that supports different instruction set extensions?

nick-knight, 2023-03-22:
> ifuncs are too much trouble [...] fmv will help

To the best of my knowledge, function multiversioning in both LLVM and GCC is implemented using GNU indirect functions. So this ticket could be considered blocked on #8.

kito-cheng, 2023-03-23:
This is referring something like this? https://gcc.gnu.org/wiki/FunctionMultiVersioning

nick-knight, 2023-03-23:
Yes. As usual,  @maskray's blog has useful info: https://maskray.me/blog/2023-02-05-function-multi-versioning

topperc, 2023-04-06:
Kito filed an issue here https://github.com/riscv-non-isa/riscv-c-api-doc/pull/35