google/android-riscv64 #71: once V and Zb* are working in cuttlefish, add them to the clang driver's defaults for android [llvm]

enh-google, 2023-03-08:
subject says it all, really.

enh-google, 2023-10-10:
https://github.com/llvm/llvm-project/blob/main/clang/lib/Driver/ToolChains/Arch/RISCV.cpp
```
      if (Triple.isAndroid())
        return "rv64imafdc_zba_zbb_zbs";
```
so Zb* is done, and it's just V and vector crypto left. i'll update the title correspondingly.

ebiggers, 2023-11-16:
Which of the vector crypto extensions, exactly, are planned to be included in the baseline?  I believe it needs to be at least Zvkng ("NIST Algorithm Suite with GCM"), which is shorthand for Zvkned_Zvknhb_Zvkb_Zvkt_Zvkg.

It would also be helpful to include Zvbc, which adds support for generic 64-bit by 64-bit carryless multiplication.  That would be useful for various things such as CRC acceleration.  Without Zvbc, people would have to abuse the `vgmul.vv` instruction of Zvkg to get a generic carryless multiplication, which would be inefficient.

enh-google, 2023-11-16:
i think at the moment we're saying "all of them", and i haven't heard any pushback on that. (there's more background in https://github.com/google/android-riscv64/issues/36 from before vector crypto was ratified, where the question of "in an emergency, what would be most important?" came up.)

(but, yes, those specific features seem unavoidable even if an OEM did try to argue for a smaller subset. specifically for Zvbc, "all the compression libraries" plus "the native heap" already need that for the same reason.)

ebiggers, 2023-11-16:
"All of them" would be Zvkng_Zvbc_Zvbb_Zvks, right?  I.e.: NIST Algorithm Suite with GCM, Vector Carryless Multiplication, Vector Basic Bit-manipulation, and ShangMi Algorithm Suite.  I'd be quite happy with Zvkng_Zvbc_Zvbb, and I think all of those would be useful.  I don't think there's any reason for Android to require the ShangMi Algorithm Suite, unless we'd prefer not to get into that level of granularity and instead say "you must implement everything in [v1.0.0 of the vector crypto spec](https://github.com/riscv/riscv-crypto/releases/download/v1.0.0/riscv-crypto-spec-vector.pdf), no exceptions".

enh-google, 2023-11-16:
> I don't think there's any reason for Android to require the ShangMi Algorithm Suite, unless we'd prefer not to get into that level of granularity

that's my thinking, yes. we know that OEMs that also ship in China will need it and the corresponding software support, and the whole industry seems to prefer one SKU where possible (for obvious cost-saving reasons), and i'm assuming the SoC vendors will feel the same, so...

enh-google, 2024-11-12:
when we land qemu 9.2 we should be able to take a look at this...