google/android-riscv64 #72: reserve x18 in the compiler driver's defaults for android

enh-google, 2023-03-11:
i _think_ this already happened, but @appujee should sign off that this has been done, because it's one of the most important issues to resolve for any NDK...

appujee, 2023-03-11:
The feature to add reserved registers for RISC-V was implemented in: https://reviews.llvm.org/D67185, but the clang driver doesn't seem to reserve `x18` by default.

```
$ prebuilts/clang/host/linux-x86/clang-r487747/bin/clang -v -target riscv64-linux-android10000 -c hello.c

"aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r487747/bin/clang-17" -cc1 -triple riscv64-unknown-linux-android10000 -emit-obj -mrelax-all -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name hello.c -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=all -ffp-contract=on -fno-rounding-math -mconstructor-aliases -target-cpu generic-rv64 -target-feature +m -target-feature +a -target-feature +f -target-feature +d -target-feature +c -target-feature -e -target-feature -h -target-feature -zihintpause -target-feature -zfhmin -target-feature -zfh -target-feature -zfinx -target-feature -zdinx -target-feature -zhinxmin -target-feature -zhinx -target-feature -zba -target-feature -zbb -target-feature -zbc -target-feature -zbs -target-feature -zbkb -target-feature -zbkc -target-feature -zbkx -target-feature -zknd -target-feature -zkne -target-feature -zknh -target-feature -zksed -target-feature -zksh -target-feature -zkr -target-feature -zkn -target-feature -zks -target-feature -zkt -target-feature -zk -target-feature -zmmul -target-feature -v -target-feature -zvl32b -target-feature -zvl64b -target-feature -zvl128b -target-feature -zvl256b -target-feature -zvl512b -target-feature -zvl1024b -target-feature -zvl2048b -target-feature -zvl4096b -target-feature -zvl8192b -target-feature -zvl16384b -target-feature -zvl32768b -target-feature -zvl65536b -target-feature -zve32x -target-feature -zve32f -target-feature -zve64x -target-feature -zve64f -target-feature -zve64d -target-feature -zicbom -target-feature -zicboz -target-feature -zicbop -target-feature -svnapot -target-feature -svpbmt -target-feature -svinval -target-feature -xventanacondops -target-feature -xtheadvdot -target-feature -experimental-zihintntl -target-feature -experimental-zca -target-feature -experimental-zcd -target-feature -experimental-zcf -target-feature -experimental-zvfh -target-feature -experimental-zawrs -target-feature -experimental-ztso -target-feature +relax -target-feature -save-restore -target-abi lp64d -msmall-data-limit 8 -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v  -internal-isystem /usr/local/aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r487747/lib/clang/17/include -internal-isystem /usr/local/include -internal-externc-isystem /include -internal-externc-isystem /usr/include -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcolor-diagnostics -o hello.o -x c hello.c
```

Even for arm64 i can't find any reserved registers added by default:
```
$ prebuilts/clang/host/linux-x86/clang-r487747/bin/clang -v -target arm64-linux-android10000 -c hello.c

 "aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r487747/bin/clang-17" -cc1 -triple arm64-unknown-linux-android10000 -emit-obj -mrelax-all -disable-free -clear-ast-before-backend -disable-llvm-verifier -discard-value-names -main-file-name hello.c -mrelocation-model pic -pic-level 2 -pic-is-pie -mframe-pointer=non-leaf -ffp-contract=on -fno-rounding-math -mconstructor-aliases -funwind-tables=2 -target-cpu generic -target-feature +neon -target-feature +v8a -target-feature +fix-cortex-a53-835769 -target-abi aapcs -mllvm -treat-scalable-fixed-error-as-warning -debugger-tuning=gdb -v -resource-dir /usr/local/aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r487747/lib/clang/17 -internal-isystem /usr/local/aosp-master-with-phones/prebuilts/clang/host/linux-x86/clang-r487747/lib/clang/17/include -internal-isystem /usr/local/include -internal-externc-isystem /include -internal-externc-isystem /usr/include -ferror-limit 19 -fno-signed-char -fgnuc-version=4.2.1 -fcolor-diagnostics -target-feature +outline-atomics -D__GCC_HAVE_DWARF2_CFI_ASM=1 -o hello.o -x c hello.c
```

appujee, 2023-03-13:
As per: https://reviews.llvm.org/D84414, -ffixed-x18 needs to be passed explicitly to the frontend
e.g.,
```
$ prebuilts/clang/host/linux-x86/clang-r487747/bin/clang-17 -fsanitize=shadow-call-stack hello.c -target riscv64-linux-android10000 -c
clang-17: error: invalid argument '-fsanitize=shadow-call-stack' only allowed with '-ffixed-x18
```

appujee, 2023-03-13:
while arm64 targets do not require any explicit argument to reserve a register

```
$ ../../prebuilts/clang/host/linux-x86/clang-r487747/bin/clang -target arm64-linux-android10000 -c hello.c -fsanitize=shadow-call-stack && echo $?
0
```

samitolvanen, 2023-03-13:
> while arm64 targets do not require any explicit argument to reserve a register

That's because the AArch64 LLVM back-end reserves x18 by default for `-android` targets. If you want similar behavior for RISC-V, you can add `TT.isAndroid()` to `isX18ReservedByDefault` in `lib/TargetParser/RISCVTargetParser.cpp`.

appujee, 2023-03-13:
sounds good. i'm preparing a patch to reserve x18.

appujee, 2023-03-13:
https://reviews.llvm.org/D145999

appujee, 2023-03-15:
Merged. Based on today's discussion, it might be better to use a different register than X18. Compressing of instructions may not work optimally with X18.

enh-google, 2023-03-15:
anecdotally comparing binary sizes between all our architectures, C already didn't seem to be working very well for us... that's probably a separate investigation. (though -- anecdotally again -- from a few cases i looked at, it seemed like "rv64gc needs 10 instructions to do even trivial things" was having a negative effect on code size that was undoing much of the positive effect of C. so maybe it's not worth looking at that until we have at least Zb*.)

and the _real_ fix for this bug is to get the ss/lp/cfi extension so we can just hand back the general-purpose register :-)

samitolvanen, 2023-03-15:
I agree about Zisslpcfi, but if we don't get it for some reason and would like to ensure SCS instructions can be compressed, it looks like we would have to reserve a register between X8-X15 instead. As most of the registers in this range are used for other purposes, the only feasible option would be... X9?

appujee, 2023-03-15:
X8-X9 are the first two callee saved registers as per: https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-cc.adoc. We should probably chose the last one in the callee saved list e.g., X27

samitolvanen, 2023-03-15:
> We should probably chose the one of the last ones in the callee saved list e.g., X27

Perhaps I'm misinterpreting something, but looking at the C extension specification, it sounds like X27 wouldn't be an improvement over X18 because the three-bit register fields (_rs1′_, _rs2′_, and _rd′_) only have encodings for X8-X15?

Here's the current prologue/epilogue we emit for SCS:
```
sd ra, 0(s2)
addi s2, s2, 8
...
ld ra, -8(s2)
addi s2, s2, -8
ret
```

Since the CS/CL instruction formats specified in "16.2 Compressed Instruction Formats" _only_ support two three-bit register fields and the `sd`/`ld` instructions above must refer to `ra`, I wonder if they just can't be compressed at all no matter which register we reserve?

Note that the `addi` instructions can always be compressed to `c.addi` since both register arguments are the same.

enh-google, 2023-03-15:
see: i told you you want the ss/lp/cfi extension :-)

> I agree about Zisslpcfi, but if we don't get it for some reason and would like to ensure SCS instructions can be compressed...

if we're worried about not being able to transition from software -> hardware, maybe we should _outlaw_ scs in apps until we have hardware (and have the compiler driver only use the hardware if you say your minimum api level is high enough)? we could poison x18 in the JNI compiler to help with this.

samitolvanen, 2023-03-16:
> if we're worried about not being able to transition from software -> hardware, maybe we should _outlaw_ scs in apps until we have hardware (and have the compiler driver only use the hardware if you say your minimum api level is high enough)?

Sounds reasonable to me. The Zisslpcfi instructions use Zimop encoding, which means they could potentially be disabled on hardware that doesn't support the extension. However, I assume there's a good chance that if the hardware doesn't support Zisslpcfi, it also doesn't support Zimops, so enabling these at a later API level for apps is probably the way to go.

topperc, 2023-03-17:
My concern with using X18 is that is incompatible with -msave-restore. https://reviews.llvm.org/D134129 has proposed enabling -msave-restore by -default with -Oz. That would make -Oz and shadow call stack unusable together.

appujee, 2023-03-17:
Are there any registers safe from `-msave-restore` that can be used for SCS?

topperc, 2023-03-17:
> Are there any registers safe from `-msave-restore` that can be used for SCS?

-msave-restore wants to save the `ra` register plus contiguous range of `s` registers starting from `s0`. `s11`(`x27`) would be ok since it's the at the end of the range. `x18` is `s2` so it's in the middle.

enh-google, 2023-03-17:
should we care about `-msave-restore`? i'm personally a lot more worried about _performance_ than code size, so my default assumption is that `-msave-restore` would be a bad choice for us anyway, and won't get used? do you have data that suggests otherwise?

enh-google, 2023-04-03:
the psabi issue corresponding to this is https://github.com/riscv-non-isa/riscv-elf-psabi-doc/issues/370.

this topic came up in this morning's risc-v psabi meeting. seems like there's consensus for reusing the otherwise-unused gp rather than wasting another general-purpose register, which sgtm.

closed 2023-04-28 by appujee