google/android-riscv64 #73: clang compiler should emit atomic sequences that are compatible with hboehm's suggested new instructions [llvm]

enh-google, 2023-03-11:
hboehm has a plan for better atomic instructions, and that plan includes a forwards-compatible change to the way the compiler emits existing atomic instructions. we should make sure (a) clang implements that and (b) that it's on by default in the compiler driver (at least for android). needed for any NDK.

asb, 2023-04-04:
#992 is relevant, but I think on the LLVM side we need an agreed and updated mapping table (like table A.6 in the current ISA manual), in order to properly review any patches submitted in this area. It sounds like the preference on the riscv-isa-manual side is to move such an effort outside of the ISA manual and to something like the psABI doc.

appujee, 2023-04-06:
> like table A.6 in the current ISA manual

This one? Do we have something like this for ARM/X86 in llvm already?

[image]

asb, 2023-04-06:
That's the one - I didn't mean we need to incorporate that table into LLVM documentation, just that memory models are complicated enough as it is so it would be much better from a review perspective if any patches can be validated against a clear table like this rather than trying to ensure we have a common understanding of a prose description.

As discussed in the linked thread, it sounds like the plan remains to move more of this to the ABI document.

appujee, 2023-04-11:
After https://github.com/riscv/riscv-isa-manual/pull/992 lands, the table will change a little.

appujee, 2023-06-22:
The [patch](https://reviews.llvm.org/D149486) hasn't landed in upstream llvm but we are cherry-picking it: https://android-review.googlesource.com/c/toolchain/llvm_android/+/2631670

closed 2023-09-12 by enh-google