google/android-riscv64 #74: investigate if llvm requires any changes to make LR/SC sequence sequentially consistent

appujee, 2023-03-11:
See if the change proposed in the (patch](https://github.com/hboehm/riscv-isa-manual/pull/1/files) is already implemented in llvm or it needs to be done.

asb, 2023-03-13:
This definitely isn't implemented in LLVM so far (we very directly implement the mappings in table A.6). I understand there's a desire to move that table outside the spec, e.g. to the psABI repo. To implement this change in LLVM, I think we'd need these patches to make it into the upstream riscv-isa-manual, an updated mapping table to be agreed, and some discussion about the backwards compatibility of the change (i.e. if we changed the default lowering to these new forms, would you be able to mix with code compiled under the old lowerings without breaking the memory model properties).

appujee, 2023-03-21:
yeah looks like it.

closed 2023-03-21 by appujee