google/android-riscv64 #76: Enable -msave-restore for binary size

appujee, 2023-03-16:
RISCV has support for`-msave-restore`. __riscv_save and __riscv_restore are implemented in compiler-rt/lib/builtins/riscv. This reduces the prologue and epilogue size of most functions.

We can either enable it globally or only for cold parts of the code (if profile information is available)

asb, 2023-03-17:
On a related note, you'd think that the MachineOutliner could subsume the need for `-msave-restore` (i.e., it should synthesise similar functions if profitable - or indeed better ones based on the instructions in the binary) especially if you're doing LTO. It sounds like right now it is profitable to both use the MachineOutliner and enable -msave-restore. Perhaps there wouldn't be a lot of benefit, but it _might_ be worth investigating further.

Presumably you're motivated to reduce binary size due to a desire about itlb and i$ usage rather than just slightly smaller binaries? If so, the MachineFunctionSplitter might be of interest (and I'd be intrigued if you've had any positive experience with it on other targets).

Not to derail things too much - you probably want to keep on eye on the [zcb](https://github.com/riscv/riscv-code-size-reduction/blob/main/Zc-specification/Zc.adoc#17-zcb) extension too, which should provide minor incremental code size benefits.

Happy to file separate issues (or for you to do so) if you're interested in tracking any of these topics.

enh-google, 2023-03-17:
> Presumably you're motivated to reduce binary size due to a desire to optimise itlb and i$ usage rather than just slightly smaller binaries?

yeah, fwiw, i doubt we actually want `-msave-restore`... i think we have much bigger concerns around riscv64 _performance_ than around _size_ relative to the existing architectures in the market. i think this bug is really just "see what it costs in practice?" and -- as is mentioned at the end of the initial description -- "can we detect cold code and profitably use this there?".

> Not to derail things too much - you probably want to keep on eye on the [zcb](https://github.com/riscv/riscv-code-size-reduction/blob/main/Zc-specification/Zc.adoc#17-zcb) extension too (mandatory in the current [RVA23 profile draft](https://github.com/riscv/riscv-profiles/blob/main/rva23-profile.adoc), which should provide minor incremental code size benefits.

yeah, this seems more interesting to me too. (though last time i looked it seemed like there were still fairly fundamental arguments about the push/pop style versus the load-pair/store-pair style?)

appujee, 2023-03-21:
Thanks for providing the context. This is more like a P4 item, if at all. As asb@ mentioned, machine outliner subsumes most of `-msave-restore`, so if it comes to saving code size we are probably better off investigating that.

appujee, 2023-03-21:
>  i think this bug is really just "see what it costs in practice?" and -- as is mentioned at the end of the initial description -- "can we detect cold code and profitably use this there?".

+1, As of now bugs like these are more of P4 things.

appujee, 2023-03-29:
Deferring this as not an immediate priority.

closed 2023-03-29 by appujee

pirama-arumuga-nainar, 2023-05-19:
Agree with the punting on this but just posting for posterity that there's that [`-mllvm -save-restore-tailcall`](https://github.com/riscv-non-isa/riscv-toolchain-conventions/blob/master/README.mkd#:~:text=of%20ra.-,__riscv_restore_tailcall_%3CN%3E,-are%20additional%20entry) can speedup tail calls when using `-msave-restore`.