google/android-riscv64 #78: Fix default usage of reserved register

appujee, 2023-03-17:
- [ ] in llvm-project
```
e.g.,
lib/hwasan/hwasan_tag_mismatch_riscv64.S:  sd x27, 216(sp)
lib/builtins/riscv/restore.S:  lw      s11, 12(sp)
lib/builtins/riscv/restore.S:  ld      s11, 8(sp)
lib/builtins/riscv/save.S:  sw     s11, 12(sp)
lib/builtins/riscv/save.S:  sd     s11, 8(sp)
lib/hwasan/hwasan_setjmp_riscv64.S:  sd    s11,  12<<3(x10)
lib/hwasan/hwasan_setjmp_riscv64.S:  fsd   fs11, 25<<3(x10)
```

- [ ] in android: https://github.com/google/android-riscv64/issues/55

enh-google, 2023-03-17:
not this one :-)
```
lib/hwasan/hwasan_setjmp_riscv64.S:  fsd   fs11, 25<<3(x10)
```
remember that we'll need to update bionic and ART if we make this change too.

(is libffi affected?)

appujee, 2023-03-17:
> not this one :-) lib/hwasan/hwasan_setjmp_riscv64.S:  fsd   fs11, 25<<3(x10)

striked it out.

> (is libffi affected?)

doesn't seem like it: https://github.com/libffi/libffi/blob/master/src/riscv/sysv.S

enh-google, 2023-03-17:
> striked it out.

seems to have affected the wrong one?

MaoHan001, 2023-03-22:
What is S11/X27 reserved for? I haven't seen any related topics. I previously only saw that the software SCS requires the use of S2/X18.

enh-google, 2023-05-01:
@appujee --- given that this bug is marked as blocking #81 do we actually _need_ the kernel work here? that seems orthogonal?

appujee, 2023-05-02:
makes sense. I've removed the item tracking kernel patches.

closed 2023-05-02 by appujee