google/android-riscv64 #85: Android Cuttlefish fails to boot when I use the --gpu_mode=drm_virgl flag.

nylon7, 2023-04-19:
I try to refer to android-riscv64's [README.md](https://github.com/google/android-riscv64/blob/main/README.md)

The environment and build command of my host are as follows.
```
$ lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.6 LTS
Release:        18.04
Codename:       bionic

$ qemu-system-riscv64 --version
QEMU emulator version 7.2.0 (v5.1.0-20988-gf1d00f35c2)
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers

# build command
mkdir aosp && cd aosp
repo init -u https://android.googlesource.com/platform/manifest -b master
repo sync -f --force-sync -c --no-clone-bundle --no-tags -j16
source build/envsetup.sh
lunch aosp_cf_riscv64_phone-userdebug
make -j8
```

this command is workable 
`$ launch_cvd -cpus=4 --memory_mb=8192`

But when I add "--gpu_mode=drm_virgl", the boot fails.
`$ launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=drm_virgl`

The error log
```
===================================================================
NOTICE:

By using this Android Virtual Device, you agree to
Google Terms of Service (https://policies.google.com/terms).
The Google Privacy Policy (https://policies.google.com/privacy)
describes how Google handles information generated as you use
Google Services. You can adjust the permission for sending
diagnostic information to Google, such as crash reports and usage
data from this Android Virtual Device, at any time by running
"launch_cvd -report_anonymous_usage_stats=n"
===================================================================

launch_cvd I 04-19 02:31:36  8491  8491 main.cc:325] Host changed from last run: 0
assemble_cvd D 04-19 02:31:36  8930  8930 fetcher_config.cpp:224] Could not find file ending in kernel
assemble_cvd D 04-19 02:31:36  8930  8930 fetcher_config.cpp:224] Could not find file ending in initramfs.img
assemble_cvd I 04-19 02:31:36  8930  8930 config_flag.cpp:153] Launching CVD using --config='phone'.
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:333] Started (pid: 8932): /scratch2/nylonc/aosp/out/host/linux-x86/bin/extract-ikconfig
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:335] /scratch2/nylonc/aosp/out/target/product/vsoc_riscv64/boot.img
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:333] Started (pid: 8940): /scratch2/nylonc/aosp/out/host/linux-x86/bin/unpack_bootimg
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:335] --boot_img
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:335] /scratch2/nylonc/aosp/out/target/product/vsoc_riscv64/boot.img
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:335] --out
assemble_cvd D 04-19 02:31:36  8930  8930 subprocess.cpp:335] /tmp/0kBYsv
Failed to initialize display.
--gpu_mode=drm_virgl was requested but the prerequisites for accelerated rendering were not detected so the device may not function correctly. Please consider switching to --gpu_mode=auto or --
gpu_mode=guest_swiftshader.
The following files contain useful debugging information:
  Logcat output: /home/nylonc/cuttlefish/instances/cvd-1/logs/logcat
  Kernel log: /home/nylonc/cuttlefish/instances/cvd-1/kernel.log
  Launcher log: /home/nylonc/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/nylonc/cuttlefish/instances/cvd-1/cuttlefish_config.json
QEMU_AUDIO_DRV=none
/scratch2/nylonc/aosp/out/host/linux-x86/bin/adb_connector
/scratch2/nylonc/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/scratch2/nylonc/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/scratch2/nylonc/aosp/out/host/linux-x86/bin/tcp_connector
/scratch2/nylonc/aosp/out/host/linux-x86/bin/config_server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/echo_server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/gnss_grpc_proxy
/scratch2/nylonc/aosp/out/host/linux-x86/bin/logcat_receiver
/scratch2/nylonc/aosp/out/host/linux-x86/bin/kernel_log_monitor
/scratch2/nylonc/aosp/out/host/linux-x86/bin/metrics
/scratch2/nylonc/aosp/out/host/linux-x86/bin/openwrt_control_server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/log_tee
/scratch2/nylonc/aosp/out/host/linux-x86/bin/process_restarter
/scratch2/nylonc/aosp/out/host/linux-x86/bin/secure_env
/scratch2/nylonc/aosp/out/host/linux-x86/bin/tombstone_receiver
/scratch2/nylonc/aosp/out/host/linux-x86/bin/android.hardware.automotive.vehicle@2.0-virtualization-grpc-server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/log_tee
failed to read from socket, retry.
/scratch2/nylonc/aosp/out/host/linux-x86/bin/wmediumd
/scratch2/nylonc/aosp/out/host/linux-x86/bin/modem_simulator
/scratch2/nylonc/qemu-private/inst/bin/qemu-system-riscv64
Server listening on unix:/home/nylonc/cuttlefish/instances/cvd-1/grpc_socket/EchoServer.sock
Server listening on 0.0.0.0:7200
Server listening on unix:/home/nylonc/cuttlefish/instances/cvd-1/grpc_socket/OpenwrtControlServer.sock
Metrics started
Metrics host received: VMStart
Start modem simulator, server_fds: 66, Sim type: normal
android.hardware.automotive.vehicle@2.0-virtualization-grpc-server E 04-19 02:31:47  9442  9471 GarageModeServerSideHandler.cpp:131] operator(): marker file /home/nylonc/cuttlefish/instances/cv
d-1/power_state has not been created yet.
KeyMint TA running with fd_in=57, fd_out=56, security_level=TrustedEnvironment
qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'
Detected unexpected exit of monitored subprocess /scratch2/nylonc/qemu-private/inst/bin/qemu-system-riscv64
Subprocess /scratch2/nylonc/qemu-private/inst/bin/qemu-system-riscv64 (9448) has exited with exit code 1
Stopping all monitored processes due to unexpected exit of critical process
stop_cvd I 04-19 02:31:47  9485  9485 main.cc:120] Requesting stop
Exit request from parent process
```

Have you ever encountered a similar problem?

nylon7, 2023-04-19:
I noticed this warning message
```
--gpu_mode=drm_virgl was requested but the prerequisites for accelerated rendering were not detected so the device may not function correctly. Please consider switching to --gpu_mode=auto or --
gpu_mode=guest_swiftshader.
```
so I change the launch_cvd command

`launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=guest_swiftshader`

it's bootable, but the Surfaceflinger got a sigabrt(-6).

So when I open the emulator, it will get a black screen.

steven10628, 2023-04-19:
> Have anyone ever encountered a similar problem?
I have  encountered a similar problem

I also saw the same warning message.

So I change the launch_cvd command

`launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=guest_swiftshader  -qemu_binary_dir=<qemu>/build/riscv64-softmmu`

and the log as following:

(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
[2023/04/19 16:04:58:5439] N: __lws_lc_untag:  -- [wsisrv|0|adopted] (0) 5.146s
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (74.125.204.x:19302) failed with error 0 : [0x00000016] Invalid argument
(stun_port.cc:632): UDP send of 20 bytes to host stun.l.google.com:19302 (2404:6800:4008:x:x:x:x:x:19302) failed with error 0 : [0x00000065] Network is unreachable
(stun_port.cc:119): Binding request timed out from 192.168.98.x:15550 (cvd-ebr)
(stun_port.cc:119): Binding request timed out from 127.0.0.x:15550 (lo)
(stun_port.cc:119): Binding request timed out from 192.168.97.x:15550 (cvd-mtap-01)
(stun_port.cc:119): Binding request timed out from [0:0:0:x:x:x:x:x]:15550 (lo)
(stun_port.cc:119): Binding request timed out from 10.0.13.x:15550 (enp5s0)

I didn't get a sigabrt(-6) message

but also get a black screen on emulator
[image]

enh-google, 2023-04-19:
yeah, sadly we're still working on swiftshader for riscv64. for now you'll need a real gpu that you can virtualize (https://source.android.com/docs/setup/create/cuttlefish-ref-gpu#requirements).

nylon7, 2023-04-20:
Hi @enh-google  , thanks for your reply

For this issue I made another try

```
apt install libvirglrenderer-dev
apt install libepoxy-dev

# rebuild QEMU
./configure --prefix=$PWD/../inst --enable-kvm --enable-system --enable-sdl --enable-opengl --target-list="riscv64-softmmu,riscv64-linux-user"
The Meson build system
Version: 0.61.5
Source dir: /scratch2/nylonc/qemu-private
Build dir: /scratch2/nylonc/qemu-private/build
Build type: native build
Project name: qemu
Project version: 7.2.0
C compiler for the host machine: cc -m64 -mcx16 (gcc 7.5.0 "cc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0")
C linker for the host machine: cc -m64 -mcx16 ld.bfd 2.30
Host machine cpu family: x86_64
Host machine cpu: x86_64
...
  Dependencies
    SDL support                  : YES
    SDL image support            : NO
    GTK support                  : YES
    pixman                       : YES 0.34.0
    VTE support                  : YES 0.52.2
    slirp support                : YES 4.1.0
    libtasn1                     : YES 4.13
    PAM                          : NO
    iconv support                : YES
    curses support               : NO
    virgl support                : YES 0.6.0
    blkio support                : NO
...

```

But I encountered another issue that is egl: no drm render node available and egl: render node init failed

```
launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=drm_virgl -qemu_binary_dir=/scratch2/nylonc/qemu-private/inst/bin
....
failed to read from socket, retry.
failed to read from socket, retry.
Metrics started
Metrics host received: VMStart
failed to read from socket, retry.
Start modem simulator, server_fds: 66, Sim type: normal
Server listening on unix:/home/nylonc/cuttlefish/instances/cvd-1/grpc_socket/EchoServer.sock
Server listening on unix:/home/nylonc/cuttlefish/instances/cvd-1/grpc_socket/OpenwrtControlServer.sock
Server listening on 0.0.0.0:7200
2023-04-20T07:39:38.054936Z qemu-system-riscv64: egl: no drm render node available
2023-04-20T07:39:38.054950Z qemu-system-riscv64: egl: render node init failed
Detected unexpected exit of monitored subprocess /scratch2/nylonc/qemu-private/inst/bin/qemu-system-riscv64
Subprocess /scratch2/nylonc/qemu-private/inst/bin/qemu-system-riscv64 (28922) has exited with exit code 1
Stopping all monitored processes due to unexpected exit of critical process
failed to read from socket, retry.
android.hardware.automotive.vehicle@2.0-virtualization-grpc-server E 04-20 07:39:38 28918 29361 GarageModeServerSideHandler.cpp:131] operator(): marker file /home/nylonc/cuttlefish/instances/cvd-1/power_state has not been created yet.
failed to read from socket, retry.
transport message failed, response body:
Metrics posted to ClearCut
stop_cvd I 04-20 07:39:39 29358 29358 main.cc:120] Requesting stop
Exit request from parent process.
```

do you have any idea?

archanox, 2023-04-21:
I have a sneaking suspicion that this issue may be related https://gitlab.freedesktop.org/mesa/mesa/-/issues/8382#note_1876946

nylon7, 2023-04-21:
Hi @enh-google @archanox , 

Thanks for your sharing. I use this command to fix the error.

```
2023-04-20T07:39:38.054936Z qemu-system-riscv64: egl: no drm render node available
2023-04-20T07:39:38.054950Z qemu-system-riscv64: egl: render node init failed
```
```
sudo apt install qemu-kvm libvirt-daemon-system
sudo adduser $USER libvirt
```

And I try to run this command
```
 launch_cvd  -cpus=8 --memory_mb=8192 --start_webrtc=true --gpu_mode=drm_virgl
```

the GPU has been enabled: **gl_version 33 - core profile enabled**

```
launch_cvd I 04-21 08:39:34 20564 20564 main.cc:325] Host changed from last run: 1
assemble_cvd D 04-21 08:39:34 21003 21003 fetcher_config.cpp:224] Could not find file ending in kernel
assemble_cvd D 04-21 08:39:34 21003 21003 fetcher_config.cpp:224] Could not find file ending in initramfs.img
assemble_cvd I 04-21 08:39:34 21003 21003 config_flag.cpp:153] Launching CVD using --config='phone'.
assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:333] Started (pid: 21005): /scratch2/nylonc/aosp/out/host/linux-x86/bin/extract-ikconfig
assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:335] /scratch2/nylonc/aosp/out/target/product/vsoc_riscv64/boot.img
assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:333] Started (pid: 21013): /scratch2/nylonc/aosp/out/host/linux-x86/bin/unpack_bootimg
assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:335] --boot_img                                                                                         assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:335] /scratch2/nylonc/aosp/out/target/product/vsoc_riscv64/boot.img
assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:335] --out
assemble_cvd D 04-21 08:39:34 21003 21003 subprocess.cpp:335] /tmp/KwxY8X
Failed to initialize display.
--gpu_mode=drm_virgl was requested but the prerequisites for accelerated rendering were not detected so the device may not function correctly. Please consider s$
itching to --gpu_mode=auto or --gpu_mode=guest_swiftshader.
Requested resuming a previous session (the default behavior) but the base images have changed under the overlay, making the overlay incompatible. Wiping the ove$lay files.                                                                                                                                                       The following files contain useful debugging information:
  Logcat output: /home/nylonc/cuttlefish/instances/cvd-1/logs/logcat
  Kernel log: /home/nylonc/cuttlefish/instances/cvd-1/kernel.log
  Launcher log: /home/nylonc/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/nylonc/cuttlefish/instances/cvd-1/cuttlefish_config.json
QEMU_AUDIO_DRV=none
/scratch2/nylonc/aosp/out/host/linux-x86/bin/adb_connector
/scratch2/nylonc/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/scratch2/nylonc/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/scratch2/nylonc/aosp/out/host/linux-x86/bin/tcp_connector
/scratch2/nylonc/aosp/out/host/linux-x86/bin/config_server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/echo_server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/gnss_grpc_proxy
/scratch2/nylonc/aosp/out/host/linux-x86/bin/logcat_receiver
/scratch2/nylonc/aosp/out/host/linux-x86/bin/kernel_log_monitor
/scratch2/nylonc/aosp/out/host/linux-x86/bin/metrics
/scratch2/nylonc/aosp/out/host/linux-x86/bin/openwrt_control_server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/log_tee
/scratch2/nylonc/aosp/out/host/linux-x86/bin/process_restarter
/scratch2/nylonc/aosp/out/host/linux-x86/bin/secure_env
/scratch2/nylonc/aosp/out/host/linux-x86/bin/tombstone_receiver
/scratch2/nylonc/aosp/out/host/linux-x86/bin/android.hardware.automotive.vehicle@2.0-virtualization-grpc-server
/scratch2/nylonc/aosp/out/host/linux-x86/bin/log_tee
/scratch2/nylonc/aosp/out/host/linux-x86/bin/wmediumd
/scratch2/nylonc/aosp/out/host/linux-x86/bin/modem_simulator
/usr/bin/qemu-system-riscv64
4/bootloader.qemu
failed to read from socket, retry.
Server listening on unix:/home/nylonc/cuttlefish/instances/cvd-1/grpc_socket/OpenwrtControlServer.sock
Server listening on unix:/home/nylonc/cuttlefish/instances/cvd-1/grpc_socket/EchoServer.sock
Start modem simulator, server_fds: 66, Sim type: normal
android.hardware.automotive.vehicle@2.0-virtualization-grpc-server E 04-21 08:39:42 21513 21532 GarageModeServerSideHandler.cpp:131] operator(): marker file /hom
e/nylonc/cuttlefish/instances/cvd-1/power_state has not been created yet.
Server listening on 0.0.0.0:7200
Metrics started
Metrics host received: VMStart
[... 112 lines cut ...]
ok 2ms and failed: unlink() failed: No such file or directory
[  125.551006] init: processing action (drm.service.enabled=true) from (/system/etc/init/drmserver.rc:8)
[  125.556350] init: starting service 'drm'...
[  125.716146] init: ... started service 'drm' has pid 485
[  125.717712] init: Command 'start drm' action=drm.service.enabled=true (/system/etc/init/drmserver.rc:9) took 165ms and succeeded
   ...
[  140.138375] type=1400 audit(140.356:99): avc:  denied  { getattr } for  comm="BootAnimation" path="/dev/dri/card0" dev="tmpfs" ino=418 scontext=u:r:bootanim:s
0 tcontext=u:object_r:graphics_device:s0 tclass=chr_file permissive=0
[  140.144040] type=1300 audit(140.356:99): arch=c00000f3 syscall=79 success=no exit=-13 a0=ffffffffffffff9c a1=fffffb69c907df a2=fffffb69c906d8 a3=0 items=0 ppi
[... cut ...] =1003 fsuid=1003 egid=1003 sgid=1003 fsgid=1003 tty=(none) ses=4294967295 comm="BootAnimation" exe="/system/bin/bootanimation" subj=u:r:bootanim:s0 key=(null)
[  140.508493] type=1327 audit(140.736:103): proctitle="/system/bin/bootanimation"
[  140.514061] type=1320 audit(140.736:103):
[  141.029337] servicemanager: Found android.hardware.graphics.allocator.IAllocator/default in device VINTF manifest.
[  152.839850] virtio-pci 0000:00:01.0: [drm] drm_plane_enable_fb_damage_clips() not called
```

enh-google, 2023-04-21:
did you wait long enough? like i said:

> (Note that even on a fast Xeon workstation it takes several minutes to get to the boot animation and tens of minutes to get to the home screen!)

in logcat you should see it dex2oat'ing _everything_. (because we can't yet compile at build time, like on arm64/x86-64.)

nylon7, 2023-04-23:
Hi @enh-google , You are right, thanks for the tip.
[image]
[image]

nylon7, 2023-04-23:
Hi @steven10628 , 
I can successfully boot into the home screen now. 

Can you try the following command and related steps and try again?

```
sudo apt install qemu-kvm libvirt-daemon-system
sudo adduser $USER libvirt
launch_cvd  -cpus=8 --memory_mb=8192 --start_webrtc=true --gpu_mode=drm_virgl -guest_enforce_security=false
```

steven10628, 2023-04-24:
Hi @nylon7 
I tried using your steps above but still have some issues.

qemu config:
./configure  --enable-kvm --enable-system --enable-sdl --enable-opengl --target-list="riscv64-softmmu,riscv64-linux-user"

qemu version:
QEMU emulator version 7.2.0 (v7.2.0)
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers

command line:
launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=drm_virgl -qemu_binary_dir=<qemu>/build/riscv64-softmmu

error log:
...
Server listening on 0.0.0.0:7200
Metrics started
Metrics host received: VMStart
Start modem simulator, server_fds: 66, Sim type: normal
android.hardware.automotive.vehicle@2.0-virtualization-grpc-server E 04-24 11:47:23 378578 378609 GarageModeServerSideHandler.cpp:131] operator(): marker file /home/locus84/cuttlefish/instances/cvd-1/power_state has not been created yet.
**qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'**
KeyMint TA running with fd_in=57, fd_out=56, security_level=TrustedEnvironment
Detected unexpected exit of monitored subprocess /home/locus84/upstream_qemu/qemu/build/riscv64-softmmu/qemu-system-riscv64
Subprocess /home/locus84/upstream_qemu/qemu/build/riscv64-softmmu/qemu-system-riscv64 (378582) has exited with exit code 1
Stopping all monitored processes due to unexpected exit of critical process
stop_cvd I 04-24 11:47:23 378623 378623 main.cc:120] Requesting stop
Exit request from parent process

Have anyone ever encountered a similar problem?
`qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'`

nylon7, 2023-04-24:
Hi @steven10628 ,

Sorry I missed some details

```
sudo apt-get install qemu-kvm libvirt-daemon-system 
sudo adduser $USER libvirt

vim node.xml
    <graphics type='spice'>
      <listen type='none'/>
    </graphics>
    <graphics type='egl-headless'>
      <gl rendernode='/dev/dri/renderD128'/>
    </graphics>
    <video>
      <model type='virtio' heads='1' primary='yes'>
        <acceleration accel3d='yes'/>
      </model>
      <address type='pci' domain='0x0000' bus='0x00' slot='0x01' function='0x0'/>
    </video>
sudo virsh create node.xml 

launch_cvd  -cpus=8 --memory_mb=8192 --start_webrtc=true --gpu_mode=drm_virgl -guest_enforce_security=false
```

steven10628, 2023-04-24:
Hi @nylon7 

Thank you for your response and assistance.

Sorry, I'm not very familiar with this part.

Regarding node.xml, should I create a new file for it or should I append it to an existing file.

I tried to create node.xml as instructed, but I received the following error message. It seems that the error is due to the lack of a root domain, so I added it separately. However, I encountered other error messages and I'm not sure if I need to provide more information in order to use it.

 $ cat node.xml

[image]

command:
$  sudo virsh create node.xml

error log:
error: Failed to create domain from node.xml
error: XML error: an os <type> must be specified

nylon7, 2023-04-24:
Hi @steven10628 , 

I missed your log sorry
`qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'`

You can upgrade the version of qemu to v7.2(git checkout v7.2.0), and do the following simple test

`./qemu-system-riscv64 -display egl-headless`

If you still get does not accept value message 'egl-headless', you must check your qemu config result.

unicornx, 2023-04-24:
> I noticed this warning message
>
> ```
> --gpu_mode=drm_virgl was requested but the prerequisites for accelerated rendering were not detected so the device may not function correctly. Please consider [...]

how do you open your enulator, by web-browser(such as chrome) or vncviewer?

nylon7, 2023-04-25:
Hi @unicornx ,

> how do you open your emulator, by web-browser(such as chrome) or vncviewer?

option --gpu_mode=guest_swiftshader use web-browser

--gpu_mode=drm_virgl use vncviewer

nylon7, 2023-04-25:
Hi @steven10628 

you can use `-display ?` for check

```
./qemu-system-riscv64 -display ?
Available display backend types:
none
gtk
sdl
egl-headless
```

steven10628, 2023-04-25:
Hi @nylon7 

After I installed the **libgbme-dev** and rebuilded qemu again.
I can find  egl-headless

command:
$  launch_cvd  -cpus=8 --memory_mb=8192 --start_webrtc=true --gpu_mode=drm_virgl -guest_enforce_security=false -qemu_binary_dir=<qemu_dir>/build/riscv64-softmmu

log:
...
proxy_adb] From: fd: 35
proxy_adb] To: vsock: 3:5555
proxy_adb] Unable to connect to vsock server: Connection reset by peer
proxy_adb] Cannot connect to the target to setup proxying: Connection reset by peer
proxy_adb] Cannot connect to the target to setup proxying: Connection reset by peer
proxy_adb] Cannot connect to the target to setup proxying: Connection reset by peer
proxy_adb] Cannot connect to the target to setup proxying: Connection reset by peer
proxy_adb] Cannot connect to the target to setup proxying: Connection reset by peer
VIRTUAL_DEVICE_DISPLAY_POWER_MODE_CHANGED
transport message failed, response body:
VIRTUAL_DEVICE_DISPLAY_POWER_MODE_CHANGED
virtio_input_hid_handle_status: unknown type 20
virtio_input_hid_handle_status: unknown type 20
virtio_input_hid_handle_status: unknown type 20
virtio_input_hid_handle_status: unknown type 20
^BwVIRTUAL_DEVICE_BOOT_STARTED
VIRTUAL_DEVICE_BOOT_COMPLETED
Virtual device booted successfully
VIRTUAL_DEVICE_NETWORK_MOBILE_CONNECTED
^C2023-04-25T02:41:32Z qemu-system-riscv64: terminating on signal 1 from pid 1844478 (<unknown process>)
Failed to read event buffer size: Success
Check failed: read_result.has_value() Success

It seems to look correct now

However when I using  VNC Viewer, it had some error

[image]

I'm not sure if it's an issue with my firewall or if there's something else I haven't done yet.

BTW:  Was this step necessary? 
vim node.xml
    
[image]

sudo virsh create node.xml 

So now I'm using the following command to try it out
command: 
$  launch_cvd  -cpus=8 --memory_mb=8192 --start_webrtc=true --gpu_mode=guest_swiftshader  -guest_enforce_security=false -qemu_binary_dir=<qemu_dir>/build/riscv64-softmmu

nylon7, 2023-04-25:
Hi @steven10628 ,

You can forget virsh. 

If your cuttlefish run on the server, you can try this.

`vim device/google/cuttlefish/host/libs/vm_manager/qemu_manager.cpp
`
```
--- a/host/libs/vm_manager/qemu_manager.cpp
+++ b/host/libs/vm_manager/qemu_manager.cpp
@@ -413,11 +413,11 @@ Result<std::vector<MonitorCommand>> QemuManager::StartCommands(
     qemu_cmd.AddParameter("egl-headless");

     qemu_cmd.AddParameter("-vnc");
-    qemu_cmd.AddParameter("127.0.0.1:", instance.qemu_vnc_server_port());
+    qemu_cmd.AddParameter("0.0.0.0:", instance.qemu_vnc_server_port());
   } else if (instance.gpu_mode() == kGpuModeGfxstream ||
              instance.gpu_mode() == kGpuModeGfxstreamGuestAngle) {
     qemu_cmd.AddParameter("-vnc");
-    qemu_cmd.AddParameter("127.0.0.1:", instance.qemu_vnc_server_port());
+    qemu_cmd.AddParameter("0.0.0.0:", instance.qemu_vnc_server_port());
   } else {
     qemu_cmd.AddParameter("-display");
```

unicornx, 2023-04-25:
@enh-google hi, enh, do we have to follow the guides from nylon to make GUI work for cuttlefish? I'm not sure if you do the similar extra steps as nylon mentioned, at least I don't see this in the guide from README.

steven10628, 2023-04-25:
Hi @nylon7 

Thanks for help

I use this command

$ sudo vncviewer   127.0.0.1:6444

[image]

enh-google, 2023-04-26:
> @enh-google hi, enh, do we have to follow the guides from nylon to make GUI work for cuttlefish? I'm not sure if you do the similar extra steps as nylon mentioned, at least I don't see this in the guide from README.

no, it "just works" for me (apart from the bit where i need to restart vncviewer).

i'm assuming these problems are because they have a much older qemu? i think i'm using this:
```
~$ qemu-system-riscv64 --version
QEMU emulator version 7.2.0 (Debian 1:7.2+dfsg-1+build6)
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers
~$ 
```

nylon7, 2023-04-26:
Hi @enh-google , @unicornx ,
In my opinion, this part needs a troubleshooting section. for example...

- QEMU support issue
**-display egl-headless: Parameter 'type' does not accept value 'egl-headless'**
`sudo apt-get install  libgbm-dev`

- cuttlefish issue on your Ubuntu version(18.04/20.04)
**qemu-system-riscv64: egl: no drm render node available
qemu-system-riscv64: egl: render node init failed**

ubuntu 20.04:  `sudo usermod -a -G render $USER`
ubuntu 18.04:  `sudo usermod -a -G video $USER`

steven10628, 2023-04-26:
Hi @enh-google @nylon7 @unicornx ,

Perhaps we can create an article on a known issue to prevent others from encountering similar problems and to further improve AOSP

unicornx, 2023-04-26:
> Hi @enh-google @nylon7 @unicornx ,
> 
> Perhaps we can create an article on a known issue to prevent others from encountering similar problems and to further improve AOSP

Actually, I doubt if this is an known issue for aosp because the behavior for cf_riscv64 is different from that for cf_x86_64. For cf_x86_64, following https://source.android.com/docs/setup/create/cuttlefish-use should work w/o extra steps.

I have not tried cf_aarch64, I understand when the host is x86_64, cf_x86_64 will use crosvm instead of qemu, so how about the operations for cf_aarch64? I don't see document about this and say any extra steps required for qemu. So I would think current status for cf_riscv64 is unexpected and maybe the development work on cuttlefish for riscv64 has not finished, so what we see is an issue?

Anyway, if that's really expected behaviour, I agree a document is definitely required and where should we post it to?

unicornx, 2023-04-26:
> no, it "just works" for me (apart from the bit where i need to restart vncviewer).
>
> i'm assuming these problems are because they have a much older qemu? i think i'm using this:
>
> ```
> ~$ [...]

from the descriptions in  this thread, I think we are all using qemu 7.2.0, the difference is only on configurations when we build it by ourselves.

enh-google, 2023-04-26:
sounds like the easy option is for me to just say "please use a current Ubuntu/Debian, not one from 2018 (!)" :-)

has anyone had problems on 22.04 or later?

enh-google, 2023-04-26:
a better fix would be to have a new enough prebuilt qemu that it "just works" and isn't dependent on the user's system. we'll look into that...

steven10628, 2023-04-27:
Hi @enh-google 

In my opinion, I think the problems encountered above seem to be caused by the lack of installation of some packages, so users can be reminded to install the relevant packages before installing AOSP.

BTW
The system I use is ubuntu20.04

enh-google, 2023-04-27:
> In my opinion, I think the problems encountered above seem to be caused by the lack of installation of some packages, so users can be reminded to install the relevant packages before installing AOSP.

those were dependencies for _building_ qemu, right?

if they're for _running_ qemu, we should just fix the existing source.android.com setup docs that we already link to.

for _building_ qemu, does anyone have actual copy & paste instructions that work, even on old versions of ubuntu? (like i said, i'm hoping we can just ship a modern qemu prebuilt soon, but, yeah, anyone whose employer won't let them use a current OS is going to have trouble until then...)

fanghuaqi, 2023-07-06:
I  tried to boot aosp master in a docker image ubuntu 22.04, with cuttlefish installed.

But I faced issue below

~~~
root@941142fabfc5:/work# launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=drm_virgl --start_webrtc=true
===================================================================
NOTICE:

By using this Android Virtual Device, you agree to
Google Terms of Service (https://policies.google.com/terms).
The Google Privacy Policy (https://policies.google.com/privacy)
describes how Google handles information generated as you use
Google Services. You can adjust the permission for sending
diagnostic information to Google, such as crash reports and usage
data from this Android Virtual Device, at any time by running
"launch_cvd -report_anonymous_usage_stats=n"
===================================================================

07-06 06:50:59.160  3755  3755 I assemble_cvd: fetcher_config.cpp:142 Adjusting cvd_file paths to directory: /work/out/host/linux-x86
07-06 06:50:59.160  3755  3755 D assemble_cvd: fetcher_config.cpp:236 Could not find file ending in kernel
07-06 06:50:59.160  3755  3755 D assemble_cvd: fetcher_config.cpp:236 Could not find file ending in initramfs.img
07-06 06:50:59.160  3755  3755 I assemble_cvd: config_flag.cpp:143 Launching CVD using --config='phone'.
07-06 06:50:59.161  3755  3755 D assemble_cvd: subprocess.cpp:433 Started (pid: 3758): /work/out/host/linux-x86/bin/extract-ikconfig
07-06 06:50:59.161  3755  3755 D assemble_cvd: subprocess.cpp:435 /work/out/target/product/vsoc_riscv64/boot.img
07-06 06:50:59.292  3755  3755 D assemble_cvd: subprocess.cpp:433 Started (pid: 3766): /work/out/host/linux-x86/bin/unpack_bootimg
07-06 06:50:59.292  3755  3755 D assemble_cvd: subprocess.cpp:435 --boot_img
07-06 06:50:59.292  3755  3755 D assemble_cvd: subprocess.cpp:435 /work/out/target/product/vsoc_riscv64/boot.img
07-06 06:50:59.292  3755  3755 D assemble_cvd: subprocess.cpp:435 --out
07-06 06:50:59.292  3755  3755 D assemble_cvd: subprocess.cpp:435 /tmp/yvb7sP
07-06 06:50:59.379  3755  3755 E assemble_cvd: assemble_cvd.cc:180 Could not open O_TMPFILE precursor to assemble_cvd.log: Operation not supported
07-06 06:50:59.398  3755  3755 D assemble_cvd: flags.cc:1101 rootcanal_instance_num: 0
07-06 06:50:59.398  3755  3755 D assemble_cvd: flags.cc:1102 launch rootcanal: 1
07-06 06:50:59.398  3755  3755 D assemble_cvd: flags.cc:1112 pica_instance_num: 0
07-06 06:50:59.398  3755  3755 D assemble_cvd: flags.cc:1113 launch pica: 1
07-06 06:50:59.398  3755  3755 V assemble_cvd: subprocess.cpp:146 Running PopulateEglAndGlesAvailability in subprocess...
07-06 06:50:59.398  3755  3755 V assemble_cvd: subprocess.cpp:154 Waiting for subprocess 3769 running PopulateEglAndGlesAvailability...
07-06 06:50:59.401  3769  3769 V assemble_cvd: lib.cpp:39 Loaded library: libEGL.so
07-06 06:50:59.401  3769  3769 V assemble_cvd: egl.cpp:35 Loaded libEGL.so
07-06 06:50:59.401  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetProcAddress
07-06 06:50:59.401  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglQueryString
07-06 06:50:59.401  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetPlatformDisplay
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetPlatformDisplayEXT
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglBindAPI
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglChooseConfig
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateContext
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreatePbufferSurface
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroyContext
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroySurface
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetConfigAttrib
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetDisplay
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetError
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglInitialize
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglTerminate
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglMakeCurrent
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglSwapBuffers
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateWindowSurface
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglSwapInterval
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglSetBlobCacheFuncsANDROID
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateImageKHR
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroyImageKHR
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateImage
07-06 06:50:59.404  3769  3769 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroyImage
07-06 06:50:59.404  3769  3769 F assemble_cvd: egl.cpp:84 Failed to initialize display.
[... cut ...]
Setup failed for ValidateHostConfiguration
07-06 06:51:00.318  3734  3734 E launch_cvd: subprocess.cpp:196 Subprocess 3792 was interrupted by a signal: 6
07-06 06:51:00.318  3734  3734 E launch_cvd: main.cc:456 run_cvd returned -1
root@941142fabfc5:/work# cat /etc/lsb-release
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"
~~~

Any steps are missing, or I can only run it in a real machine or virtual machine, docker machine is not supported?

Thanks

enh-google, 2023-07-06:
did you try to fix the things it told you are wrong?
```
User must be a member of cvdnetwork
User must be a member of kvm
```
looks like you didn't actually follow the setup instructions we link to on the front page?

nylon7, 2023-07-07:
Hi @fanghuaqi , maybe you can try this
```
sudo usermod -aG kvm,cvdnetwork,render $USER
```

fanghuaqi, 2023-07-07:
> did you try to fix the things it told you are wrong?
> 
> ```
> User must be a member of cvdnetwork
> User must be a member of kvm
> ```
> 
> looks like you didn't actually follow the setup instructions we link to on the front page?

Hi @enh-google , I am trying it in a docker image, it seems this is not working. So I am asking whether it can only work in a real machine or virtual machine, thanks.

[image]

unicornx, 2023-07-07:
> > did you try to fix the things it told you are wrong?
> > ```
> > User must be a member of cvdnetwork
> > User must be a member of kvm
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > looks like you didn't actually follow the setup instructions we link to on the front page?
> 
> Hi @enh-google and @nylon7 , thanks a lot for your suggestions, I am trying it in a docker image, it seems this is not working. So I am asking whether it can only work in a real machine or virtual machine, thanks.
> 
> [image]

I would suggest you use a real linux machine when testing with cuttlefish, FYI: https://lists.riscv.org/g/sig-android/topic/95667663#147

fanghuaqi, 2023-07-07:
Thanks @unicornx for your experiment experience, so virtual machine also not work, only real physic machine works.

fanghuaqi, 2023-07-07:
Hello, I tried to build android using a real hardware machine, but still get error when bring up.

~~~shell
$ launch_cvd -cpus=4 --memory_mb=8192 --gpu_mode=drm_virgl
===================================================================
NOTICE:

By using this Android Virtual Device, you agree to
Google Terms of Service (https://policies.google.com/terms).
The Google Privacy Policy (https://policies.google.com/privacy)
describes how Google handles information generated as you use
Google Services.
===================================================================
Automatically send diagnostic information to Google, such as crash
reports and usage data from this Android Virtual Device. You can
adjust this permission at any time by running
"launch_cvd -report_anonymous_usage_stats=n". (Y/n)?:n
07-07 18:30:03.651 3655484 3655484 I assemble_cvd: fetcher_config.cpp:142 Adjusting cvd_file paths to directory: /Local/hqfang/workspace/software/android/aosp/out/host/linux-x86
07-07 18:30:03.651 3655484 3655484 D assemble_cvd: fetcher_config.cpp:236 Could not find file ending in kernel
07-07 18:30:03.651 3655484 3655484 D assemble_cvd: fetcher_config.cpp:236 Could not find file ending in initramfs.img
07-07 18:30:03.651 3655484 3655484 I assemble_cvd: config_flag.cpp:143 Launching CVD using --config='phone'.
07-07 18:30:03.651 3655484 3655484 D assemble_cvd: subprocess.cpp:433 Started (pid: 3655487): /Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/extract-ikconfig
07-07 18:30:03.651 3655484 3655484 D assemble_cvd: subprocess.cpp:435 /Local/hqfang/workspace/software/android/aosp/out/target/product/vsoc_riscv64/boot.img
07-07 18:30:03.786 3655484 3655484 D assemble_cvd: subprocess.cpp:433 Started (pid: 3655495): /Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/unpack_bootimg
07-07 18:30:03.786 3655484 3655484 D assemble_cvd: subprocess.cpp:435 --boot_img
07-07 18:30:03.786 3655484 3655484 D assemble_cvd: subprocess.cpp:435 /Local/hqfang/workspace/software/android/aosp/out/target/product/vsoc_riscv64/boot.img
07-07 18:30:03.786 3655484 3655484 D assemble_cvd: subprocess.cpp:435 --out
07-07 18:30:03.786 3655484 3655484 D assemble_cvd: subprocess.cpp:435 /tmp/qSDBpK
07-07 18:30:03.872 3655484 3655484 E assemble_cvd: assemble_cvd.cc:180 Could not open O_TMPFILE precursor to assemble_cvd.log: Operation not supported
07-07 18:30:03.934 3655484 3655484 D assemble_cvd: flags.cc:1101 rootcanal_instance_num: 0
07-07 18:30:03.934 3655484 3655484 D assemble_cvd: flags.cc:1102 launch rootcanal: 1
07-07 18:30:03.934 3655484 3655484 D assemble_cvd: flags.cc:1112 pica_instance_num: 0
07-07 18:30:03.934 3655484 3655484 D assemble_cvd: flags.cc:1113 launch pica: 1
07-07 18:30:03.936 3655484 3655484 V assemble_cvd: subprocess.cpp:146 Running PopulateEglAndGlesAvailability in subprocess...
07-07 18:30:03.936 3655484 3655484 V assemble_cvd: subprocess.cpp:154 Waiting for subprocess 3655498 running PopulateEglAndGlesAvailability...
07-07 18:30:03.938 3655498 3655498 V assemble_cvd: lib.cpp:39 Loaded library: libEGL.so
07-07 18:30:03.938 3655498 3655498 V assemble_cvd: egl.cpp:35 Loaded libEGL.so
07-07 18:30:03.938 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetProcAddress
07-07 18:30:03.938 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglQueryString
07-07 18:30:03.938 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetPlatformDisplay
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetPlatformDisplayEXT
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglBindAPI
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglChooseConfig
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateContext
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreatePbufferSurface
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroyContext
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroySurface
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetConfigAttrib
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetDisplay
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglGetError
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglInitialize
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglTerminate
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglMakeCurrent
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglSwapBuffers
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateWindowSurface
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglSwapInterval
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglSetBlobCacheFuncsANDROID
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateImageKHR
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglDestroyImageKHR
07-07 18:30:03.940 3655498 3655498 V assemble_cvd: egl.cpp:67 Loaded EGL function: eglCreateImage
[... cut ...]
07-07 18:30:04.309 3655484 3655484 I assemble_cvd: assemble_cvd.cc:225 Requested resuming a previous session (the default behavior) but the base images have changed under the overlay, making the overlay incompatible. Wiping the overlay files.
07-07 18:30:04.309 3655484 3655484 D assemble_cvd: clean.cc:93 Assuming prior files of /home/hqfang/cuttlefish/assembly /home/hqfang/.cuttlefish.sh /home/hqfang/.cuttlefish_config.json /home/hqfang/cuttlefish/instances/cvd-1 /tmp/cf_avd_400000018/cvd-1
~~~

Thanks

fanghuaqi, 2023-07-10:
I tried to reboot my real machine, I can see I am in the groups required, as below:

[image]

But I still can't boot android and get error message as below:

~~~
launch_cvd -cpus=4 --memory_mb=8192 --gpu_mode=auto -qemu_binary_dir=/Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin
... ...
The following files contain useful debugging information:
  Logcat output: /home/hqfang/cuttlefish/instances/cvd-1/logs/logcat
  Kernel log: /home/hqfang/cuttlefish/instances/cvd-1/kernel.log
  Launcher log: /home/hqfang/cuttlefish/instances/cvd-1/logs/launcher.log
  Instance configuration: /home/hqfang/cuttlefish/instances/cvd-1/cuttlefish_config.json
QEMU_AUDIO_DRV=none
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/adb_connector
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/config_server
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/echo_server
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/gnss_grpc_proxy
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/logcat_receiver
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/kernel_log_monitor
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/openwrt_control_server
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/netsimd
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/socket_vsock_proxy
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/secure_env
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/tombstone_receiver
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/android.hardware.automotive.vehicle@2.0-virtualization-grpc-server
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/log_tee
Server listening on unix:/tmp/cf_avd_400000018/cvd-1/grpc_socket/EchoServer.sock
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/wmediumd
/Local/hqfang/workspace/software/android/aosp/out/host/linux-x86/bin/modem_simulator
/Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin/qemu-system-riscv64
Server listening on 0.0.0.0:7200
root-canal I 07-10 11:33:32.094 device_boutique.cc:39                    Registering beacon
root-canal I 07-10 11:33:32.094 device_boutique.cc:39                    Registering beacon_swarm
root-canal I 07-10 11:33:32.094 device_boutique.cc:39                    Registering scripted_beacon
root-canal I 07-10 11:33:32.094 device_boutique.cc:39                    Registering sniffer
root-canal I 07-10 11:33:32.094 controller_properties.cc:1385            DELETE_STORED_LINK_KEY command validation failed (c121,excluded)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1435            LE_ISO_READ_TEST_COUNTERS command validation failed (excluded,c46)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1436            LE_ISO_RECEIVE_TEST command validation failed (excluded,c46)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1437            LE_ISO_TEST_END command validation failed (excluded,c47)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1439            LE_ISO_TRANSMIT_TEST command validation failed (excluded,c45)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1457            LE_READ_ISO_LINK_QUALITY command validation failed (excluded,c50)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1458            LE_READ_ISO_TX_SYNC command validation failed (excluded,c45)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1477            LE_RECEIVER_TEST_V2 command validation failed (excluded,c13)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1491            LE_REQUEST_PEER_SCA command validation failed (excluded,c44)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1507            LE_SET_DATA_RELATED_ADDRESS_CHANGES command validation failed (excluded,c10)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1545            LE_TRANSMITTER_TEST_V2 command validation failed (excluded,c12)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1637            REMOTE_NAME_REQUEST_CANCEL command validation failed (c106,excluded)
root-canal I 07-10 11:33:32.094 controller_properties.cc:1692            WRITE_DEFAULT_ERRONEOUS_DATA_REPORTING command validation failed (c135,excluded)
Server listening on unix:/tmp/cf_avd_400000018/cvd-1/grpc_socket/OpenwrtControlServer.sock
07-10 11:33:32.099 112531 112567 E android.hardware.automotive.vehicle@2.0-virtualization-grpc-server: GarageModeServerSideHandler.cpp:131 operator(): marker file /home/hqfang/cuttlefish/instances/cvd-1/power_state has not been created yet.
Start modem simulator, server_fds: 68, Sim type: normal
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=55, out=54
KeyMint Rust TA running with fd_in=61, fd_out=60, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
2023-07-10T03:33:32Z qemu-system-riscv64: egl: no drm render node available
2023-07-10T03:33:32Z qemu-system-riscv64: egl: render node init failed
[... cut ...]
$ /Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin/qemu-system-riscv64 -display ?
Available display backend types:
none
gtk
sdl
egl-headless
curses
dbus
$ /Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin/qemu-system-riscv64 -display egl-headless
qemu-system-riscv64: egl: no drm render node available
qemu-system-riscv64: egl: render node init failed
# but I already installed qemu-kvm libvirt-daemon-system, the above error still appeared
~~~

What else is missing?

Thanks

nylon7, 2023-07-17:
Hi @unicornx , In my opinion, we can use Swiftshader instead of vendor driver on Docker.
Hi @fanghuaqi could you try these steps?
```
# first step. //update swiftshader
cd external/swiftshader
git pull aosp main

# second setp
vim vsoc_riscv64/phone/aosp_cf.mk

diff --git a/vsoc_riscv64/phone/aosp_cf.mk b/vsoc_riscv64/phone/aosp_cf.mk
index 784c95962..70a397b6b 100644
--- a/vsoc_riscv64/phone/aosp_cf.mk
+++ b/vsoc_riscv64/phone/aosp_cf.mk
@@ -56,7 +56,7 @@ $(call inherit-product, device/google/cuttlefish/shared/bluetooth/device_vendor.
 $(call inherit-product, device/google/cuttlefish/shared/camera/device_vendor.mk)
 $(call inherit-product, device/google/cuttlefish/shared/graphics/device_vendor.mk)
 # TODO: FIXME: Enable swiftshader for graphics, and remove angle below.
-#$(call inherit-product, device/google/cuttlefish/shared/swiftshader/device_vendor.mk)
+$(call inherit-product, device/google/cuttlefish/shared/swiftshader/device_vendor.mk)

 # Use angle until swiftshader is unabled so that `gfxstream_guest_angle_host_swiftshader`
 # works.
@@ -123,7 +123,10 @@ PRODUCT_MODEL := Cuttlefish riscv64 phone
 PRODUCT_PACKAGES += \
     androidx.window.extensions \
     androidx.window.sidecar \
+    vulkan.pastel \

 PRODUCT_VENDOR_PROPERTIES += \
     ro.soc.manufacturer=$(PRODUCT_MANUFACTURER) \
-    ro.soc.model=$(PRODUCT_DEVICE)
+    ro.soc.model=$(PRODUCT_DEVICE) \
+    ro.hardware.egl=angle \
+    ro.hardware.vulkan=pastel
```

enh-google, 2023-07-18:
see also https://android-review.googlesource.com/c/device/google/cuttlefish/+/2657022 :-)

fanghuaqi, 2023-07-19:
Hi @enh-google ,

I tried to update source code in `device/google/cuttlefish` and `external/swiftshader` in aosp to main branch, which will include this https://android-review.googlesource.com/c/device/google/cuttlefish/+/2657022  patch, and build aosp again using `m -j`, and tried to boot android using command

~~~shell
launch_cvd -cpus=4 --memory_mb=8192 --gpu_mode=guest_swiftshader -qemu_binary_dir=/Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin
~~~

Get following output:

~~~
Server listening on unix:/tmp/cf_avd_400000018/cvd-1/grpc_socket/ControlEnvProxyServer.sock
/Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin/qemu-system-riscv64
Server listening on 0.0.0.0:7200
root-canal I 07-19 17:49:57.081 device_boutique.cc:39                    Registering beacon
root-canal I 07-19 17:49:57.081 device_boutique.cc:39                    Registering beacon_swarm
root-canal I 07-19 17:49:57.081 device_boutique.cc:39                    Registering scripted_beacon
root-canal I 07-19 17:49:57.081 device_boutique.cc:39                    Registering sniffer
root-canal I 07-19 17:49:57.081 controller_properties.cc:1385            DELETE_STORED_LINK_KEY command validation failed (c121,excluded)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1435            LE_ISO_READ_TEST_COUNTERS command validation failed (excluded,c46)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1436            LE_ISO_RECEIVE_TEST command validation failed (excluded,c46)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1437            LE_ISO_TEST_END command validation failed (excluded,c47)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1439            LE_ISO_TRANSMIT_TEST command validation failed (excluded,c45)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1457            LE_READ_ISO_LINK_QUALITY command validation failed (excluded,c50)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1458            LE_READ_ISO_TX_SYNC command validation failed (excluded,c45)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1477            LE_RECEIVER_TEST_V2 command validation failed (excluded,c13)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1491            LE_REQUEST_PEER_SCA command validation failed (excluded,c44)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1507            LE_SET_DATA_RELATED_ADDRESS_CHANGES command validation failed (excluded,c10)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1545            LE_TRANSMITTER_TEST_V2 command validation failed (excluded,c12)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1637            REMOTE_NAME_REQUEST_CANCEL command validation failed (c106,excluded)
root-canal I 07-19 17:49:57.081 controller_properties.cc:1692            WRITE_DEFAULT_ERRONEOUS_DATA_REPORTING command validation failed (c135,excluded)
Server listening on unix:/tmp/cf_avd_400000018/cvd-1/grpc_socket/OpenwrtControlServer.sock
07-19 17:49:57.084 1722577 1722599 E android.hardware.automotive.vehicle@2.0-virtualization-grpc-server: GarageModeServerSideHandler.cpp:131 operator(): marker file /home/hqfang/cuttlefish/instances/cvd-1/power_state has not been created yet.
WaylandServer running on socket 15
[2023/07/19 17:49:57:1050] N: lws_create_context: LWS: 4.3.0, NET CLI SRV H1 H2 WS ConMon IPv6-absent
[2023/07/19 17:49:57:1051] N: __lws_lc_tag:  ++ [wsi|0|pipe] (1)
[2023/07/19 17:49:57:1054] N: __lws_lc_tag:  ++ [vh|0|default|127.0.0.1|127.0.0.1|0] (1)
[2023/07/19 17:49:57:1069] N: [vh|0|default|127.0.0.1|127.0.0.1|0]: lws_socket_bind: source ads 127.0.0.1
[2023/07/19 17:49:57:1069] N: __lws_lc_tag:  ++ [wsi|1|listen|default|127.0.0.1|36039] (2)
Size of OBJECT = 1204
Size of components in TPMT_SENSITIVE = 744
    TPMI_ALG_PUBLIC                 2
    TPM2B_AUTH                      50
    TPM2B_DIGEST                    50
    TPMU_SENSITIVE_COMPOSITE        642
MAX_CONTEXT_SIZE can be reduced to 1264 (1344)
(adm_helpers.cc:48): Unable to access speaker.
size : 360
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=80, out=79
KeyMint Rust TA running with fd_in=86, fd_out=85, security_level=TrustedEnvironment
No secure deletion data file found. Creating one.
vector version is not specified, use the default value v1.0
vector version is not specified, use the default value v1.0
vector version is not specified, use the default value v1.0
vector version is not specified, use the default value v1.0
GUEST_UBOOT_VERSION: 2023.01-maybe-dirty (Jan 01 1970 - 00:00:00 +0000)
U-Boot 20
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=80, out=79
KeyMint Rust TA running with fd_in=86, fd_out=85, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
netsimd INFO 2023-07-19 09:49:58.86429313 | netsim generated pcap files in temp directory has been removed.
root-canal I 07-19 17:49:58.086 posix_async_socket_server.cc:79          Listening on: 7500 (3)
root-canal I 07-19 17:49:58.087 test_command_handler.cc:113              Add beacon@be:ac:01:55:00:01
root-canal I 07-19 17:49:58.087 test_command_handler.cc:113              Add beacon@be:ac:01:55:00:02
root-canal I 07-19 17:49:58.087 test_model.cc:76                         StartTimer()
netsim D 07-19 17:49:58.087 Set up test channel.
[... cut ...]
add 64 bytes of entropy
early boot ended
transport message failed, response body: device offline
transport message failed, response body: device offline
Generating new secret with slot ID: 3
Fetching secret with slot ID: 3
Generating new secret with slot ID: 4
Fetching secret with slot ID: 4
Fetching secret with slot ID: 4
transport message failed, response body: device '0.0.0.0:6520' not found
transport message failed, response body: device '0.0.0.0:6520' not found
~~~

vnc no response
[image]

nylon7, 2023-07-20:
Hi @fanghuaqi 
You need to wait long enough and it should show `virtual device booted successfully`.

fanghuaqi, 2023-07-20:
> Hi @fanghuaqi You need to wait long enough and it should show `virtual device booted successfully`.

Yes, it finally boot with the info you present, but it is really slow, mouse click almost no response.

enh-google, 2023-07-25:
> Yes, it finally boot with the info you present, but it is really slow, mouse click almost no response.

fwiw, i just tried using guest_swiftshader instead of drm_virgl, and i also found that it's _way_ slower. drm_virgl isn't _fast_, but it's pretty usable. guest_swiftshader was pretty unusable for me too, even on a fast 128 core machine.

matthewgui, 2023-08-17:
hi @enh-google ,

I follow https://github.com/google/android-riscv64/ README.md  to build aosp and run launch_cvd ，
```shell
launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=drm_virgl  -qemu_binary_dir=/Local/hqfang/workspace/software/android/aosp/rvprebuilt/qemu/bin
```
but it cannot enter into android main UI，it seems that android bootanimation is displayed，main ui cannot display,
and bootanimation task is complete，launcher and UI task is running。the startup time is enough to show main UI which is not the reason。
[image]
[image]

There is an error message in the cvd log file. I do not know if this is the reason why the main UI cannot be displayed。
If so, what should i do about it，thanks for your help！
[image]
[image]
[launch_cvd startup log.txt](https://github.com/google/android-riscv64/files/12367388/launch_cvd.startup.log.txt)

thank you！

matthewgui, 2023-09-11:
hi @enh-google @nylon7 @steven10628 @unicornx ,

 Can cuttlefish work successfully with the AOSP latest version?

enh-google, 2023-09-11:
i personally have never reproduced the issue mentioned here.

i built ToT AOSP today (with V enabled!) and the current debian testing qemu ran that just fine for me with `--gpu_mode=drm_virgl`.

matthewgui, 2023-09-12:
hi @enh-google 
Can you paste your cuttlefish log and android logcat output？ 
thank you for help！

nylon7, 2023-09-13:
Hi @matthewgui ,

Now, we use our internal stable branch for AOSP; the last re-base date is 2023/09.01.

matthewgui, 2023-09-13:
hi @nylon7 
thank you for your info，your qemu version is 8.0 ？

nylon7, 2023-09-13:
Hi @matthewgui , we also use internal QEMU, but the code base is 7.2.

matthewgui, 2023-09-13:
Hi @nylon7 ,
Is it convenient to provide your cuttlefish log? 
thank you!

nylon7, 2023-09-13:
Hi @matthewgui ,
I need some time to confirm, if possible, I will be sent it here.

matthewgui, 2023-09-13:
Hi @nylon7 ,
ok, thank you!

gongwuchun, 2023-10-19:
Is there anyone tried the latest risc-v cuttlefish? I cannot boot to home screen

> Hi @matthewgui ,
> 
> Now, we use our internal stable branch for AOSP; the last re-base date is 2023/09.01.

Would you mind share the local manifest of 2023.09.01? I mean the revision of all the android repos on 2023.09.01, thanks!

enh-google, 2023-10-19:
(we are making some progress with CI --- https://android-review.googlesource.com/c/platform/bionic/+/2796852?checksRunsSelected=aosp_cf_riscv64_phone-trunk_staging-userdebug:aosp-main is the first change i know of that at least _built_ a riscv64 build before being merged. we're still working on qemu and getting to where we can be sure it _boots_ with every change...)

i synced, built, and booted today... i had a cuttlefish crash that was fixed by `rm -rf ~/cuttlefish*` but i've been stuck at the "android" logo for 18 minutes now... (normally i only need a shell anyway, so i don't bother with the ui, so i can't say when the ui last worked for me.)

gongwuchun, 2023-10-20:
> (we are making some progress with CI --- https://android-review.googlesource.com/c/platform/bionic/+/2796852?checksRunsSelected=aosp_cf_riscv64_phone-trunk_staging-userdebug:aosp-main is the first [...]

thank you for your clarification, I'm exactly encounter the same result as yours.

gongwuchun, 2023-10-23:
> Is there anyone tried the latest(HEAD of (2023.10.19)) risc-v cuttlefish? I cannot boot to home screen either.
> 
> > Hi @matthewgui ,
> > Now, we use our internal stable branch for AOSP; the last re-base date is 2023/09.01.
> 
> Would you mind share the local manifest of 2023.09.01? I mean the revision of all the android repos on 2023.09.01, thanks!

Never mind, I got the manifest from ci.android.com

enh-google, 2023-11-02:
(i've update the instructions on the front page to show how to use the prebuilt qemu.)

prashanthswami, 2024-02-15:
As this bug hasn't gotten any feedback since November, I'm going to assume things are okay - the instructions have also been updated today to remove some of the extra flags (and you'll now get our prebuilt QEMU by default, so that everyone's QEMU version and features are aligned). I'll tentatively close this, but feel free to open a separate issue if things have changed or the latest ToT doesn't work for you.

closed 2024-02-15 by prashanthswami