google/android-riscv64 #86: android cuttlefish infinite reboot when I use -kernel_path and -initramfs_path

nylon7, 2023-05-02:
Android cuttlefish provides two options for user -kernel_path and -initramfs_path.

When we use these options, that will be infinite reboot.

- build kernel
```
mkdir kernel && cd kernel
repo init -u https://android.googlesource.com/kernel/manifest -b common-android-mainline
repo sync -j8
tools/bazel run --jobs 8 //common:kernel_riscv64_dist
tools/bazel run --jobs 8 //common-modules/virtual-device:virtual_device_riscv64_dist

```

```
GUEST_UBOOT_VERSION: 2023.01-maybe-dirty (Jan 01 1970 - 00:00:00 +0000)
U-Boot 20
KeyMint TA running with fd_in=59, fd_out=58, security_level=TrustedEnvironment
GUEST_KERNEL_VERSION: 6.2.0-mainline-maybe-dirty (build-user@build-host) (Android (9796371, based on r487747) clang version 17.0.0 (https://a
ndroid.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.0) #1 SMP PREEMPT Thu Jan  1 00:00:00 UTC
1970
] Linux version
GUEST_UBOOT_VERSION: 2023.01-maybe-dirty (Jan 01 1970 - 00:00:00 +0000)
U-Boot 20
KeyMint TA running with fd_in=59, fd_out=58, security_level=TrustedEnvironment
GUEST_KERNEL_VERSION: 6.2.0-mainline-maybe-dirty (build-user@build-host) (Android (9796371, based on r487747) clang version 17.0.0 (https://a
ndroid.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.0) #1 SMP PREEMPT Thu Jan  1 00:00:00 UTC
1970
] Linux version
GUEST_UBOOT_VERSION: 2023.01-maybe-dirty (Jan 01 1970 - 00:00:00 +0000)
U-Boot 20
KeyMint TA running with fd_in=59, fd_out=58, security_level=TrustedEnvironment
GUEST_KERNEL_VERSION: 6.2.0-mainline-maybe-dirty (build-user@build-host) (Android (9796371, based on r487747) clang version 17.0.0 (https://a
ndroid.googlesource.com/toolchain/llvm-project d9f89f4d16663d5012e5c09495f3b30ece3d2362), LLD 17.0.0) #1 SMP PREEMPT Thu Jan  1 00:00:00 UTC
1970
] Linux version
GUEST_UBOOT_VERSION: 2023.01-maybe-dirty (Jan 01 1970 - 00:00:00 +0000)
....
```
- kernel.log
```
OpenSBI v1.1
   ____                    _____ ____ _____
  / __ \                  / ____|  _ \_   _|
 | |  | |_ __   ___ _ __ | (___ | |_) || |
 | |  | | '_ \ / _ \ '_ \ \___ \|  _ < | |
 | |__| | |_) |  __/ | | |____) | |_) || |_
  \____/| .__/ \___|_| |_|_____/|____/_____|
        | |
        |_|

Platform Name             : riscv-virtio,qemu
Platform Features         : medeleg
...
avb_slot_verify.c:768: ERROR: boot_a: Error verifying vbmeta image: OK_NOT_SIGNED
...
Starting kernel ...
…
[    0.040470][    T1] cacheinfo: Unable to detect cache hierarchy for CPU 0
[    0.062627][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 1
[    0.066258][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 2
[    0.067671][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 3
[    1.069058][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 4
[    0.072855][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 5
[    0.074195][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 6
[    0.075382][    T0] cacheinfo: Unable to detect cache hierarchy for CPU 7
...
[    0.896771][    T1] init: init first stage started!
[    0.930008][    T1] init: Loading module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/block/virtio_blk.ko with args ''
[    0.932636][    T1] virtio_blk: module verification failed: signature and/or required key missing - tainting kernel
[    0.937463][    T1] BPF: [117048] Invalid name_offset:2034556
[    0.937642][    T1] Kernel module BTF mismatch detected, BTF debug info may be unavailable for some modules
[    0.940264][    T1] init: Loaded kernel module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/block/virtio_blk.ko
[    0.941714][    T1] init: Loading module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/char/hw_random/virtio-rng.ko with args ''
[    0.943371][    T1] BPF: [117045] Invalid name_offset:2034556
[    0.944662][    T1] init: Loaded kernel module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/char/hw_random/virtio-rng.ko
[    0.945885][    T1] init: Loading module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/char/virtio_console.ko with args ''
[    0.949424][    T1] init: Loaded kernel module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/char/virtio_console.ko
[    0.951017][    T1] init: Loading module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/virtio/virtio_dma_buf.ko with args ''
[    0.952494][    T1] BPF: [117046] Invalid name_offset:2034556
[    0.953262][    T1] init: Loaded kernel module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/virtio/virtio_dma_buf.ko
[    0.953739][    T1] init: Loading module /lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/gpu/drm/virtio/virtio-gpu.ko with args ''
[    0.957737][    T1] virtio_gpu: Unknown relocation type 57
[    0.972661][    T1] init: Failed to insmod '/lib/modules/6.2.0-mainline-maybe-dirty/kernel/drivers/gpu/drm/virtio/virtio-gpu.ko' with args '': Invalid argument
[    0.973157][    T1] init: LoadWithAliases was unable to load virtio_gpu
[    0.975715][    T1] init: Failed to load kernel modules
[    0.981640][    T1] init: InitFatalReboot: signal 6
[    1.024478][    T1] init:   #00 pc 00000000001241bc  /init (unwindstack::AndroidLocalUnwinder::InternalUnwind(std::__1::optional<int>, unwindstack::AndroidUnwinderData&)+82) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.024966][    T1] init:   #01 pc 00000000001168a2  /init (android::init::InitFatalReboot(int)+230) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.025452][    T1] init:   #02 pc 0000000000116e68  /init (android::init::InstallRebootSignalHandlers()::$_0::__invoke(int)+24) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.025941][    T1] init:   #03 pc 0000000000000800  [vdso] (__vdso_rt_sigreturn)
[    1.026205][    T1] init:   #04 pc 000000000027a49e  /init (syscall+14) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.026502][    T1] init:   #05 pc 000000000027a62e  /init (abort+98) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.027040][    T1] init:   #06 pc 000000000011bac0  /init (android::init::InitAborter(char const*)+42) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.027814][    T1] init:   #07 pc 000000000023df9c  /init (android::base::SetAborter(std::__1::function<void (char const*)>&&)::$_0::__invoke(char const*)+56) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.028367][    T1] init:   #08 pc 000000000023d900  /init (android::base::LogMessage::~LogMessage()+292) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.028729][    T1] init:   #09 pc 000000000010e66a  /init (android::init::FirstStageMain(int, char**)+7664) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.029181][    T1] init:   #10 pc 0000000000279096  /init (__real_libc_init(void*, void (*)(), int (*)(int, char**, char**), structors_array_t const*, bionic_tcb*)+504) (BuildId: 8bb54dc890bb5235e474840c7fe2e6a9)
[    1.029625][    T1] init: Reboot ending, jumping to kernel
[    1.030381][    T1] kvm: exiting hardware virtualization
[    1.032882][    T1] reboot: Restarting system with command 'bootloader'
```

But if we copy these images to AOSP and rebuild it, that can be bootable.

Have anyone ever encountered a similar problem?

nylon7, 2023-05-04:
the virtio_gpu: Unknown relocation type 57 is because virtio_gpu.ko has R_RISCV_32_PCREL but kernel can't handle it.

appujee, 2023-05-10:
The following symbols have R_RISCV_32_PCREL

$ readelf -r virtual_device_riscv64/dist/virtio-gpu.ko  | grep R_RISCV_32_PCREL
```
000000000004  065400000039 R_RISCV_32_PCREL  0000000000001934 .Ltmp2 + 0
000000000014  065600000039 R_RISCV_32_PCREL  000000000000193c .Ltmp4 + 0
000000000024  065e00000039 R_RISCV_32_PCREL  0000000000001944 .Ltmp8 + 0
000000000034  066000000039 R_RISCV_32_PCREL  000000000000194c .Ltmp10 + 0
000000000044  066600000039 R_RISCV_32_PCREL  0000000000001954 .Ltmp14 + 0
000000000054  066800000039 R_RISCV_32_PCREL  000000000000195c .Ltmp16 + 0
000000000064  067000000039 R_RISCV_32_PCREL  0000000000001964 .Ltmp20 + 0
000000000074  067200000039 R_RISCV_32_PCREL  000000000000196c .Ltmp22 + 0
```

For example `.Ltmp2` is called by

```
1507 0000000000000ec4 <virtio_gpu_array_add_obj>:
 1508      ec4: 01 11         addi  sp, sp, -32
 1509      ec6: 06 ec         sd  ra, 24(sp)
 1510      ec8: 22 e8         sd  s0, 16(sp)
 1511      eca: 26 e4         sd  s1, 8(sp)
 1512      ecc: 4a e0         sd  s2, 0(sp) 
 1513      ece: 00 10         addi  s0, sp, 32
 1514      ed0: aa 84         mv  s1, a0
 1515      ed2: 08 55         lw  a0, 40(a0)
 1516      ed4: d0 54         lw  a2, 44(s1)
 1517      ed6: 63 0e c5 02   beq a0, a2, 0xf12 <.Ltmp2>
 1518      eda: 2e 89         mv  s2, a1
 1519      edc: 85 45         li  a1, 1
 1520      ede: 2f 25 b9 00   amoadd.w  a0, a1, (s2)
 1521      ee2: 1b 06 05 00   sext.w  a2, a0
 1522      ee6: 05 ca         beqz  a2, 0xf16 <.Ltmp2+0x4>

...
1540 0000000000000f12 <.Ltmp2>:
 1541      f12: 02 90         ebreak
 1542      f14: cd bf         j 0xf06 <virtio_gpu_array_add_obj+0x42>
 1543      f16: 89 45         li  a1, 2
 1544      f18: 4a 85         mv  a0, s2
 1545      f1a: 97 00 00 00   auipc ra, 0
 1546      f1e: e7 80 00 00   jalr  ra
 1547      f22: c9 bf         j 0xef4 <virtio_gpu_array_add_obj+0x30>

```

From what i read, R_RISCV_32_PCREL is used for exception handling related code. https://github.com/riscv-non-isa/riscv-elf-psabi-doc/pull/110/files

enh-google, 2023-05-10:
looks like this is actually self-modifying code for the WARN_ONCE() stuff. @ndesaulniers might know more about how that's supposed to work/whether anything's changed there recently...

kito-cheng, 2023-05-11:
@nylon7 @appujee I suspect it because we enabled `-fasynchronous-unwind-tables` by default now https://reviews.llvm.org/D145164, that enabled by default on both GCC and LLVM recently.

kito-cheng, 2023-05-11:
Did some experimental to compile a simple hello world: `R_RISCV_32_PCREL` will gone if `-fno-asynchronous-unwind-tables` is given, but I think the right way is to adding support on linux kernel land.

appujee, 2023-05-11:
> ... but I think the right way is to adding support on linux kernel land.

Thanks for suggesting the fix @kito-cheng, where should we create a ticket for this?

kito-cheng, 2023-05-14:
Notified our kernel folks,  they will reply later :)

nylon7, 2023-05-14:
> > ... but I think the right way is to adding support on linux kernel land.
> 
> Thanks for suggesting the fix @kito-cheng, where should we create a ticket for this?

Hi @appujee , @kito-cheng , Thank you for your opinions and feedback.

Because there doesn't seem to be a specific place to open tickets related to the Linux Kernel, I will be responsible for implementing and submitting the handler for R_RISCV_32_PCREL and sending it upstream.

appujee, 2023-05-15:
Thanks @nylon7 for offering to help with this.

steven10628, 2023-05-17:
Hi @nylon7 
FYI, After modifying the following code, the kernel no longer encounters errors.
I'm not sure if this is the correct approach. Below is the reference material I used, which may help save you some time.

[image]

ref:
https://reviews.llvm.org/D70204
https://github.com/riscv-non-isa/riscv-elf-psabi-doc/blob/master/riscv-elf.adoc#:~:text=57-,32_PCREL,-Static

nylon7, 2023-05-18:
hi, @steven10628  thanks for the information :D

nylon7, 2023-05-26:
Hi @steven10628 @enh-google @appujee 

This is the version we tested initially, because we found that there are still many relocation types that are not supported, so it may be upstream together with other supported parts in the future. Please use this patch before that :D.

[0001-RISC-V-Support-32_PCREL-relocation-type-in-kernel-mo.patch](https://github.com/google/android-riscv64/files/11572488/0001-RISC-V-Support-32_PCREL-relocation-type-in-kernel-mo.patch)

adelva1984, 2023-06-09:
The workaround was submitted to ACK for building Cuttlefish kernels in aosp/2618134.

closed 2023-06-09 by enh-google

nickdesaulniers, 2023-06-09:
Can that patch be sent to LKML please? It LGTM.

nylon7, 2023-06-11:
Hi @nickdesaulniers 
I will do it later, thanks for your code review.