google/android-riscv64 #87: add a cts test for RISCV_HWPROBE_MISALIGNED_FAST

enh-google, 2023-05-04:
when we have kernels with https://lore.kernel.org/lkml/20230221190858.3159617-6-evan@rivosinc.com/ we should add a test to make it clear that this is a requirement.

enh-google, 2023-07-12:
> Should we test for all supported cpu features like zba, zbb etc?

we won't need to because CTS _itself_ will be built using those instructions. this is about checking the _kernel configuration_ (which you could potentially screw up even if you _do_ have good hardware, for example).

enh-google, 2023-07-12:
(note that if you're interested in implementing this, cferris did merge the 6.4 uapi headers earlier this week, so the necessary constants etc are now available, even if we don't have a 6.4 _kernel_ just yet.)

appujee, 2023-07-12:
i see https://android.googlesource.com/platform/bionic/+/37c3f3c67ea5bb750708d325bb2140bfc59f98a1%5E%21/#F3

enh-google, 2023-07-13:
https://android-review.googlesource.com/c/platform/bionic/+/2657071 pokes a hole in the seccomp filter to allow access to riscv_hwprobe.

enh-google, 2023-07-29:
https://android-review.googlesource.com/c/platform/bionic/+/2679530 adds <sys/hwprobe.h> and the test for that tests RISCV_HWPROBE_MISALIGNED_FAST (and all bionic tests are in CTS), so a separate test isn't _strictly_ necessary any more, but the more i think about it, the more i think it's probably a good idea to have a separate explicit CTS test for RISCV_HWPROBE_MISALIGNED_FAST anyway? (i'll let @appujee decide by either sending me a CTS change to +2 or closing this bug :-) )

appujee, 2023-07-31:
i think it makes sense to have cts test as arm64 tests are also there. i'll create a patch for that.

appujee, 2023-08-11:
test has been added: https://android-review.googlesource.com/c/platform/cts/+/2675633

closed 2023-08-11 by appujee