google/android-riscv64 #89: The executable formar of cvd host package of aosp_cf_riscv64_phone maybe wrong

rayeh5, 2023-05-15:
$file cvd_internal_status
cvd_internal_status: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[md5/uuid]=a20ede1ced5e28bc91cd9e3d211c2140, with debug_info, not stripped
-------------------------------
Should the cvd be x86 format? The cvd will run on x86 Linux, right?

enh-google, 2023-05-15:
where did you get that binary from? mine is in out/host/linux-x86/bin/ and -- as you'd expect from that directory name -- it's an x86(-64) binary.

closed 2023-05-15 by enh-google

rayeh5, 2023-05-15:
I got the cvd (cvd-host_package.tar.gz) from: 
https://ci.android.com/builds/submitted/10139363/aosp_cf_riscv64_phone-userdebug/latest/cvd-host_package.tar.gz 
----------
rayeh/cvd/bin$ file cvd_internal_start
cvd_internal_start: ELF 64-bit LSB shared object, ARM aarch64, version 1 (SYSV), dynamically linked, BuildID[md5/uuid]=c6bbb8a66abeb68ed4393bf639e82b75, with debug_info, not stripped
rayeh/cvd/bin$ file launch_cvd
launch_cvd: symbolic link to cvd_internal_start

enh-google, 2023-05-16:
ah, yeah, our build bots only support one host arch at a time, and for aosp_cf_riscv64_phone we're currently using arm64 for that.

you can either build your own host binaries (that's what i do, which is why i hadn't seen this), or you can just use the cvd-host_package.tar.gz from aosp_cf_x86_64_phone-userdebug at the same build id as the aosp_cf_riscv64_phone-userdebug build you're trying to use.