google/android-riscv64 #93: really slow boot time on cuttlefish

nylon7, 2023-06-09:
I would like to ask how long the average boot time is for everyone on Cuttlefish, and if there are any minimum hardware requirements for it.

The server I am using is an AMD Ryzen 9 3950X 16-Core, but it takes me about 12 hours to see the Android home screen when I boot it up for the first time.

I suspect that this long boot time may be due to the `dex2oat` processing during the initial boot. The boot time improves significantly on subsequent boots, but it still takes around 1 to 2 hours.

I have made the following efforts, but the results have not been significant. Could you please provide some directions or suggestions?

- dex2oat
```
compiler flag use "verify"
set some option
WITH_DEXPREOPT := true
WITH_DEXPREOPT_PIC := true
DONT_DEXPREOPT_PREBUILTS := false
LOCAL_DEX_PREOPT := false
```
- other
```
Kernel configs -> 
removing any debug related -> SLAB_DEBUG, SPINLOCK_DBG
UART -> disable uart
Remove packages included in system partition
```
- some workaround
```
e.g. system/update_engine/aosp/cleanup_previous_update_action.cc
- constexpr auto kCheckBootCompletedInterval = base::TimeDelta::FromSeconds(2);
+ constexpr auto kCheckBootCompletedInterval = base::TimeDelta::FromSeconds(60);
// Interval to check IBootControl::isSlotMarkedSuccessful
- constexpr auto kCheckSlotMarkedSuccessfulInterval = base::TimeDelta::FromSeconds(2);
+ constexpr auto kCheckSlotMarkedSuccessfulInterval = base::TimeDelta::FromSeconds(60);
// Interval to call SnapshotManager::ProcessUpdateState
- constexpr auto kWaitForMergeInterval = base::TimeDelta::FromSeconds(2);
+ - constexpr auto kWaitForMergeInterval = base::TimeDelta::FromSeconds(60);
```

enh-google, 2023-06-09:
are you using the exact command lines on this project's main page? i boot to shell in less than a minute. for the graphical run, i see the "android" screen in about 1 minute, and i see the home screen after pretty much exactly 10 minutes.

your mention of dex2oat makes me think you might be on an old build? sync your tree, rebuild, and try again. this improved significantly over the last month or so.

nylon7, 2023-06-11:
Hi @enh-google ,

Thanks for your suggestion; I will try.

closed 2023-09-12 by enh-google