google/android-riscv64 #96: Question about video play

rayeh5, 2023-06-20:
When I play video, the output color seems wrong?
Please refer to the attachment.
Thanks for your time.

[image]

enh-google, 2023-06-20:
what format of video? played with what?

rayeh5, 2023-06-21:
the video file is as attachment.
I push into the cuttlefish, then use mouse to click it to play this file.
https://github.com/google/android-riscv64/assets/5005947/9f10b7c7-7966-4f3e-8742-48bac769d5ff

negge, 2023-06-22:
Hi @rayeh5, I see in https://github.com/google/android-riscv64/issues/38#issuecomment-1581824786 that a fix for libyuv was cherrypicked into AOSP. Can you rebuild and check if this fixes the issue for you?

rayeh5, 2023-08-11:
Any comment for this issue? It seems the "Blue" color is missing when play video file.

rayeh5, 2023-08-15:
I build aosp_cf_x86_64_phone-userdebug, aosp_cf_riscv64_phone-userdebug and aosp_cf_arm64_phone-userdebug by using the same source, the output color of video is correct for x86_64 version, the arm64 and riscv64 are the same with wrong color.
Any comment?

jmacnak, 2023-08-15:
@rayeh5: Could you share more information about how you are launching the Cuttlefish instances, about the enviroment the Cuttlefish instances are running in (mostly just interested in whether or not you are using the default `--gpu_mode=auto` and whether or not it is detecting acceleration support), and about how you are interacting with the Cuttlefish instances (which VNC client)?

[image]

^ This is `aosp_cf_arm64_phone-userdebug` using `cvd start --gpu_mode=guest_swiftshader --vm_manager=qemu_cli` and interacting with the instance via TightVNC.

rayeh5, 2023-08-16:
The problem should be caued by "--gpu_mode=drm_virgl".
By using "--gpu_mode=guest_swiftshader" and TightVNC, the "black and white" video is normal.
Use "scrcpy" program to connect to Cuttlefish, the video is normal.
But with color video, the TightVNC has problem to show correct color.
The TightVNC shows the following message:
------
Connected to RFB server, using protocol version 3.8
No authentication needed
Authentication successful
Desktop name "QEMU (cvd-1)"
VNC server default format:
  32 bits per pixel.
  Least significant byte first in each pixel.
  True colour: max red 255 green 255 blue 255, shift red 16 green 8 blue 0
Using default colormap which is TrueColor.  Pixel format:
  32 bits per pixel.
  Least significant byte first in each pixel.
  True colour: max red 255 green 255 blue 255, shift red 16 green 8 blue 0
Same machine: preferring raw encoding
-------------------------

jmacnak, 2023-08-17:
Ack, thanks for the info. I might not get to looking at this until next week.

rayeh5, 2023-08-24:
got it and thanks!

closed 2023-08-24 by rayeh5