google/android-riscv64 #99: Question about audio output

rayeh5, 2023-07-05:
The latest image seems support "pulseaudio", when I don't run "pulseaudio", the log show message as below:
----------
pulseaudio: pa_context_connect() failed
pulseaudio: Reason: Connection refused
pulseaudio: Failed to initialize PA contextMetrics posted to ClearCut
-----------
But after enable the pulseaudio, when play audio file in Cuttlefish, the sound can't be heard.
By the way, the "Media volume" can't be adjusted.
Does the Cuttlefish support audio output now?

Thanks for your time.

[image]

enh-google, 2023-07-07:
i think it should work... are you running directly on your linux box, or is this docker or something?

rayeh5, 2023-07-10:
The Cuttlefish runs on Linux (20.04.1-Ubuntu SMP Thu Apr 20 22:12:07 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux) box directly.
QEMU emulator version 7.1.0; 
VNC(R) Viewer 7.1.0 (r49525) x64 (Apr 3 2023 17:17:18)
Launch command:
launch_cvd -cpus=8 --memory_mb=8192 --gpu_mode=drm_virgl --enable_audio=true -qemu_binary_dir=~/local1/qemu/build

Thanks for your time.

enh-google, 2023-07-10:
does pulseaudio work for you otherwise?

rayeh5, 2023-07-11:
Yes, pulseaudio works.
Maybe it's cause by VNC server? The "Session information" shows:
"Audio format: VNC Server does not support audio"
[image]

adelva1984, 2023-07-11:
Can you try removing this line and see if it fixes it?
https://android.googlesource.com/device/google/cuttlefish/+/refs/heads/main/host/libs/vm_manager/qemu_manager.cpp#775

rayeh5, 2023-07-12:
Thanks for the information.
I modify the soruce of "./device/google/cuttlefish/host/libs/vm_manager/qemu_manager.cpp" as below,
but the audio still can't be heard.
Any suggestion? Thanks!

#if 0
void LogAndSetEnv(const char* key, const std::string& value) {
  setenv(key, value.c_str(), 1);
  LOG(INFO) << key << "=" << value;
}
#endif
...
//  LogAndSetEnv("QEMU_AUDIO_DRV", "none");

rayeh5, 2023-08-11:
The audio is OK via scrcpy to connect to Cuttlefish.

closed 2023-08-11 by rayeh5