google/android-riscv64 #100: Add RISCV64 ABI and cpu features to the ndk/guides

appujee, 2023-07-12:
Android support ABIs are documented in https://developer.android.com/ndk/guides/abis. It needs to mention RISC-V too.
- [ ] ABIs
- [ ] How to get cpu features
- [ ] Compatibility Test Suite to check for supported cpu features.

enh-google, 2023-07-12:
(we won't be adding support for risc-v to the long-deprecated ndk cpufeatures library. i've filed https://github.com/google/android-riscv64/issues/101 for the updates to the text on https://developer.android.com/ndk/guides/cpu-features though.)

enh-google, 2023-07-12:
(i've unlinked #87 because i don't think that's really related, nor is there any dependency in either direction, but i've created new bugs for the specific CDD and d.a.c changes we need. arguably CDD isn't 100% related, but since CDD _references_ d.a.c, there's a dependency here.)