google/android-riscv64 #104: TSAN: Generate TSAN clang rt libs for riscv64

prashanthswami, 2023-07-14:
To be at parity with other architectures in the Android NDK, we would like to have the libclang_rt.tsan-riscv64-android.so library included as part of the imported sysroot from Android (Clang Prebuilts). This is distinct from it actually _working_, which is tracked here: https://github.com/android/ndk/issues/1041 .

pirama-arumuga-nainar, 2023-07-18:
https://reviews.llvm.org/D145214 enables tsan runtime for riscv64.

enh-google, 2023-09-12:
seems like no-one's reviewed that change yet. @appujee --- is that something we can help with?

appujee, 2023-09-13:
let me reach out to the author if he is still working on it.

appujee, 2023-09-14:
nobody responded to my query on discord. I can commandeer the patch and put for review.

appujee, 2023-09-18:
Update: The author is waiting on review as the patch is feature complete. I've pinged the reviewers

appujee, 2023-09-28:
Update: I pinged reviewers to take another look.
Except for minor nits, the reviewers are okay with the patch but would like dvyukov to take a look. I've pinged him.

closed 2023-10-13 by appujee