google/android-riscv64 #105: Compilation error occurred for riscv64 configuration on the main branch

MaoHan001, 2023-07-17:
20:25:14 Build sandboxing disabled due to nsjail error.
============================================
PLATFORM_VERSION_CODENAME=VanillaIceCream
PLATFORM_VERSION=VanillaIceCream
PRODUCT_INCLUDE_TAGS=com.android.mainline
TARGET_PRODUCT=aosp_cf_riscv64_phone
TARGET_BUILD_VARIANT=userdebug
TARGET_ARCH=riscv64
TARGET_ARCH_VARIANT=riscv64
TARGET_CPU_VARIANT=generic
HOST_OS=linux
HOST_OS_EXTRA=Linux-3.10.0-1127.el7.x86_64-x86_64-Ubuntu-18.04.6-LTS
HOST_CROSS_OS=linux_musl
BUILD_ID=AOSP.MAIN
OUT_DIR=out
============================================
wildcard(out/target/product/vsoc_riscv64/clean_steps.mk) was changed, regenerating...
[ 99% 2223/2224] finishing legacy Make module parsing ...
build/make/core/tasks/tools/compatibility.mk:66: warning: test/catbox/tools/catbox-tradefed/README has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/app_compat/csuite/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/cts-root/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/mts/README.md has no license metadata
build/make/core/tasks/tools/compatibility.mk:66: warning: test/vts/tools/vts-core-tradefed/README has no license metadata
[100% 622/622] writing legacy Make module rules ...
FAILED: 
out/soong/installs-aosp_cf_riscv64_phone.mk:62318: error: overriding commands for target `out/target/product/vsoc_riscv64/system/lib64/libcronet.114.0.5735.84.so', previously defined at out/soong/installs-aosp_cf_riscv64_phone.mk:62314

enh-google, 2023-07-17:
"try again"? i believe the change in question affected other builds too and was reverted here: https://android-review.googlesource.com/c/platform/external/cronet/+/2657748

(i'm rebuilding now to confirm.)

enh-google, 2023-07-17:
(yeah, built for me with ToT from this morning. that revert seems to have been the active ingredient.)

closed 2023-07-17 by enh-google