google/android-riscv64 #109: ART abort when TARGET_FLATTEN_APEX is removed

peyo-hd, 2023-09-05:
The issue is observed on JH7110 build https://github.com/android-risc-v/device_arv_jh7110 - which is based on AOSP main branch.

Before TARGET_FLATTEN_APEX removal -   It can successfully boot-up to home screen .
(full-logcat https://drive.google.com/file/d/172VWMzCj1viU9Eg286Vu_zBrphwsxZG1 )

After TARGET_FLATTEN_APEX removal - https://android.googlesource.com/platform/build/+/0d500c58aaee61bd8827ed74088d89ec966c0d16 -
ART Init failed with following log : 
>  01-01 00:05:41.382   714   714 F system_server: art_method.cc:661] Check failed: IsNative() java.lang.Class java.lang.Class.forName(java.lang.String, boolean, java.lang.ClassLoader)

(full-logcat https://drive.google.com/file/d/1c7LdlGGwUHYrST-7kn_4nNzU0nRyoWu5 ) 

When it is compared against successful logcat, ART aborted before reaching following point : 
>  01-01 00:08:13.406   851   851 D SystemServerTiming: InitBeforeStartServices

When the TARGET_FLATTEN_APEX removal commit is reverted, the build output is working fine.
And with same AOSP source snapshot,  ARM64 build can work even with TARGET_FLATTEN_APEX removal.

Do you observe this problem on cuttlefish emulator build of riscv64?

peyo-hd, 2023-09-06:
Found that DEXPREOPT should be enabled when TARGET_FLATTEN_APEX is removed.
ART Abort is fixed with https://github.com/android-risc-v/device_arv_jh7110/commit/f5a6befb27d27d479632fa65170288a0071dbb28.

peyo-hd, 2023-09-08:
Previous comment regarding DEXPREOPT is false information.
The issue is still there,  When TARGET_FLATTEN_APEX removed ART Abort occurs.

peyo-hd, 2023-09-08:
For successful boot, I can see log from dex2oatd64 like following:
> 01-01 00:00:21.284   358   358 I dex2oatd64: /apex/com.android.art/bin/dex2oatd64 --runtime-arg -

But for ART abort case (NO FLATTEN APEX),  there is no log from dex2oatd.

Then, if this different could be explaned by someone from ART team, It's helpful.

enh-google, 2023-09-08:
"works for me" with qemu/cuttlefish at AOSP ToT. (_yesterday's_ build was broken in a different way, where qemu wouldn't even start, but i did `repo sync` this morning and rebuilt, and everything's fine for me now, booting to the UI.)

closed 2023-11-02 by enh-google