google/android-riscv64 #111: Enable fast unaligned access for aosp

appujee, 2023-10-04:
Maybe we can add this to clang driver once we have tested it?

appujee, 2023-10-04:
aosp testing in progress: https://android-review.googlesource.com/c/platform/build/soong/+/2734373

enh-google, 2023-10-04:
> Maybe we can add this to clang driver once we have tested it?

oh, definitely --- any time we land something in AOSP (and we feel like it's _really_ landed and isn't going to get reverted), we should start moving it upstream. any time i _don't_ suggest that to you is a bug on my part :-)

(the alternative is that every single build system has to add these flags, which obviously doesn't scale.)

appujee, 2023-10-13:
merged.

appujee, 2023-10-13:
i think it is okay to keep it open as the currently used (backend) flags will be updated (to -munaligned-access) with the new toolchain.

SiFiveHolland, 2023-10-13:
Any reason not to use `-mno-strict-align`, which is [common with GCC](https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Options.html#index-mstrict-align-3)?

prashanthswami, 2024-02-26:
Bumping this thread because the last comment mentions we'll have 'unaligned-access' in the new toolchain, and I believe we've completed a round of updates since Oct.

@appujee - should we be switching our flags to "-munaligned-access" now? WDYT about @SiFiveHolland 's comments on 'no-strict-align'?

appujee, 2024-02-26:
yes, `-munaligned-access` flag works with the latest clang update.

enh-google, 2024-03-18:
> I'll replace the previous workaround with -munaligned-access

is there a reason we haven't moved this into the driver yet? this flag has been in AOSP for a long time now with no problems...

appujee, 2024-03-18:
The change landed few weeks back, https://android-review.googlesource.com/c/platform/build/soong/+/2977952 I forgot to close this bug.

appujee, 2024-03-18:
Interestingly `-munaligned-access`  was replaced by `-mno-strict-align` in llvm upstream a few days back. Both flags were aliases. Pirama has change to that effect https://android-review.googlesource.com/c/platform/build/soong/+/3002035

appujee, 2024-03-18:
I'll  enable the flag by default for Android. https://github.com/llvm/llvm-project/pull/85704 WIP.

enh-google, 2024-04-04:
the llvm change was merged ... is that cherrypicked for our llvm? (and is that cherrypick in the toolchain that we're working on currently for the forthcoming ndk release, or is it for the _next_ one?)

appujee, 2024-04-05:
it was cherry-picked to our llvm in https://android-review.googlesource.com/c/toolchain/llvm_android/+/3009313 it is not in the toolchain (clang-r522817) we are working on.

enh-google, 2024-11-12:
@appujee --- is this in our current clang?

appujee, 2024-11-12:
yes https://android.googlesource.com/platform/prebuilts/clang/host/linux-x86/+/refs/heads/main/clang-r530567/clang_source_info.md

closed 2024-11-12 by appujee