google/android-riscv64 #113: Does the cuttlefish work on current main?

alpencolt, 2023-10-13:
I've tried to build RISC-V cuttlefish from latest AOSP sources, but emulator failed with error:
```
2023-10-13T17:10:52.161892Z qemu-system-riscv64: -chardev pipe,id=hvc9,path=uwb_fifo_vm: Could not open 'uwb_fifo_vm': No such file or directory
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=62, out=61
KeyMint Rust TA running with fd_in=68, fd_out=67, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
Detected unexpected exit of monitored subprocess /home/alexander/src/qemu/build/qemu-system-riscv64
Subprocess /home/alexander/src/qemu/build/qemu-system-riscv64 (1918420) has exited with exit code 1
Stopping all monitored processes due to unexpected exit of critical process
```

Passing `-enable_host_uwb=false` to `launch_cvd` helps to launch image, but it loads in the recovery mode:
```
$ adb devices
List of devices attached
0.0.0.0:6520	recovery
```

Used QEMU 8.1.
Everything work few weeks ago. What was changed?

enh-google, 2023-10-13:
i'm not seeing any qemu crashes with
```
QEMU emulator version 8.0.4 (Debian 1:8.0.4+dfsg-2)
```
try adding `--qemu_binary_dir=device/google/cuttlefish_vmm/qemu/x86_64-linux-gnu/usr/bin/` to your command line? that's the qemu prebuilt we just checked in (but haven't made the default yet).

note though that i'm seeing a different crash (with either qemu) if i try to start a graphical session:
```
Subprocess /home/enh/aosp-main-with-phones/out/host/linux-x86/bin/webRTC (1859842) was interrupted by a signal 'Killed' (9)
```
just booting to a shell is working fine for me though.

enh-google, 2023-10-13:
yeah, i see this too with the debian qemu:
```
2023-10-13T22:25:39.958992Z qemu-system-riscv64: -chardev pipe,id=hvc9,path=uwb_fifo_vm: Could not open 'uwb_fifo_vm': No such file or directory
Detected unexpected exit of monitored subprocess /usr/bin/qemu-system-riscv64
```
and got a different failure with the new qemu prebuilts:
```
qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'
netsimd I 10-13 22:24:21.506 socket.rs:64 - Hci socket server is listening on: 7300
Server listening on unix:/tmp/cf_avd_73769/cvd-1/grpc_socket/OpenwrtControlServer.sock
netsimd I 10-13 22:24:21.509 server.rs:38 - Frontend http server is listening on http://localhost:7681
Detected unexpected exit of monitored subprocess /home/enh/aosp-main-with-phones/device/google/cuttlefish_vmm/qemu/x86_64-linux-gnu/usr/bin//qemu-system-riscv64
Subprocess /home/enh/aosp-main-with-phones/device/google/cuttlefish_vmm/qemu/x86_64-linux-gnu/usr/bin//qemu-system-riscv64 (2350409) has exited with exit code 1
```

prashanthswami, 2023-10-14:
Thanks for flagging! A recent unrelated-to-RISCV change turned on UWB by default, which exposed a problem in the QEMU manager for Cuttlefish that wasn't present in the CrosVM manager. I've put up a fix for this here: https://android-review.git.corp.google.com/c/device/google/cuttlefish/+/2786107 .

alpencolt, 2023-10-14:
@enh-google @prashanthswami thank you!
Unfortunately I don't have access to https://android-review.git.corp.google.com/c/device/google/cuttlefish/+/2786107. When it will be synced with public AOSP?

prashanthswami, 2023-10-14:
Ah, apologies on that, provided the wrong domain - here's the same AOSP link, with the right domain: https://android-review.googlesource.com/c/device/google/cuttlefish/+/2786107 . It'll be checked in directly to AOSP, so I'm hoping it'll get in by end-of-day Monday at the latest, when the relevant folks have had a chance to review.

prashanthswami, 2023-10-15:
Got this reviewed and merged, things seem to work fine locally after syncing the latest AOSP, so I'll consider this closed for now.

closed 2023-10-15 by prashanthswami

alpencolt, 2023-10-16:
@prashanthswami thank you! It works with qemu build from sources, but failed with prebuild one:
```
netsimd I 10-16 11:51:15.638 rust_main.rs:89 - netsim artifacts path: /tmp/android-alexander/netsimd
netsimd E 10-16 11:51:15.638 os_utils.cc:94 - Unable to find netsim ini file: /run/user/1000/netsim_1.ini
root-canal I 10-16 14:51:15.638 posix_async_socket_server.cc:79          Listening on: 7500 (3)
Server listening on 0.0.0.0:7200
root-canal I 10-16 14:51:15.639 test_model.cc:77                         StartTimer()
netsimd I 10-16 11:51:15.639 bluetooth_facade.cc:179 - Set up test channel.
qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'
```

Probably it also should be updated