google/android-riscv64 #115: AOSP mainline fails to boot (ActivityManager fails to start any applications)

SiFiveHolland, 2023-10-20:
I am currently hitting an issue where AOSP fails to boot because all application processes forked from Zygote fail to call `IActivityManager::attachApplication`. I'm synced to AOSP mainline from earlier today. My last known good build is from October 3.

```
10-20 07:27:24.967  5955  5955 D SystemServerTiming: OnBootPhase_550_com.android.server.policy.PermissionPolicyService
10-20 07:27:24.968  5955  5955 D SystemServerTiming: Permission_grant_default_permissions-0
10-20 07:27:24.971  5955  5955 D SystemServerTiming: Permission_callback_waiting-0
10-20 07:27:25.106  5955  5983 W ActivityManager: Slow operation: 100ms so far, now at startProcess: asking zygote to start proc
10-20 07:27:25.135  5955  6134 W ActivityManager: Slow operation: 124ms so far, now at startProcess: done starting proc!
10-20 07:27:25.145  5902  5902 D Zygote  : Forked child process 6180
10-20 07:27:25.150  5955  5983 W ActivityManager: Slow operation: 144ms so far, now at startProcess: returned from zygote!
10-20 07:27:25.161  5955  6134 W Looper  : Slow dispatch took 189ms android.perm h=android.os.Handler c=com.android.internal.infra.ServiceConnector$Impl$$ExternalSyntheticLambda2@9519f8d m=0
10-20 07:27:25.187  5955  5983 W ActivityManager: Slow operation: 181ms so far, now at startProcess: done updating battery stats
10-20 07:27:25.188  5955  5983 W ActivityManager: Slow operation: 182ms so far, now at startProcess: building log message
10-20 07:27:25.189  5955  5983 I ActivityManager: Start proc 6180:com.android.permissioncontroller/u0a81 for service {com.android.permissioncontroller/com.android.permissioncontroller.permission.service.PermissionControllerServiceImpl}
10-20 07:27:25.190  5955  5983 W ActivityManager: Slow operation: 184ms so far, now at startProcess: starting to update pids map
10-20 07:27:25.191  5955  5983 W ActivityManager: Slow operation: 185ms so far, now at startProcess: done updating pids map
10-20 07:27:25.211  6180  6180 I ssioncontroller: Using CollectorTypeCMC GC.
10-20 07:27:25.237  6180  6180 E ssioncontroller: Not starting debugger since process cannot load the jdwp agent.
10-20 07:29:04.854  5955  5982 W ActivityManager: Process ProcessRecord{31d635d 6147:com.android.systemui/u0a62} failed to attach
10-20 07:29:04.857  5955  5982 W ActivityManager: Forcing bringing down service: ServiceRecord{e275ca2 u0 com.android.systemui/.wallpapers.ImageWallpaper}
10-20 07:29:04.861  5955  5982 I ActivityManager: Killing 6147:com.android.systemui/u0a62 (adj -10000): start timeout
10-20 07:29:04.866  5955  5973 D DisplayManagerService: Drop pending events for gone uid 10062
10-20 07:29:04.871  5955  5984 I libprocessgroup: Successfully killed process cgroup uid 10062 pid 6147 in 0ms
10-20 07:29:04.895  5902  5902 I Zygote  : Process 6147 exited due to signal 9 (Killed)
10-20 07:29:05.193  5955  5982 W ActivityManager: Process ProcessRecord{24bd38c 6180:com.android.permissioncontroller/u0a81} failed to attach
10-20 07:29:05.195  5955  5982 W ActivityManager: Forcing bringing down service: ServiceRecord{79f3d9 u0 com.android.permissioncontroller/.permission.service.PermissionControllerServiceImpl}
10-20 07:29:05.198  5955  5982 I ActivityManager: Killing 6180:com.android.permissioncontroller/u0a81 (adj -10000): start timeout
10-20 07:29:05.202  5955  5973 D DisplayManagerService: Drop pending events for gone uid 10081
10-20 07:29:05.207  5955  5984 I libprocessgroup: Successfully killed process cgroup uid 10081 pid 6180 in 5ms
10-20 07:29:05.213  5955  6134 E PermissionControllerManager: Error updating user_sensitive flags for uid -1
10-20 07:29:05.213  5955  6134 E PermissionControllerManager: java.util.concurrent.CancellationException
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at java.util.concurrent.CompletableFuture.cancel(CompletableFuture.java:2466)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.AndroidFuture.cancel(AndroidFuture.java:153)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl$CompletionAwareJob.cancel(ServiceConnector.java:779)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl.cancelPendingJobs(ServiceConnector.java:608)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl.unbindJobThread(ServiceConnector.java:593)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl$$ExternalSyntheticLambda0.run(Unknown Source:2)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at android.os.Handler.handleCallback(Handler.java:958)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at android.os.Handler.dispatchMessage(Handler.java:99)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at android.os.Looper.loopOnce(Looper.java:205)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at android.os.Looper.loop(Looper.java:294)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at android.os.HandlerThread.run(HandlerThread.java:67)
10-20 07:29:05.213  5955  6134 E PermissionControllerManager:   at com.android.server.ServiceThread.run(ServiceThread.java:46)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager: Error granting or upgrading runtime permissions
10-20 07:29:05.220  5955  6134 E PermissionControllerManager: java.util.concurrent.CancellationException
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at java.util.concurrent.CompletableFuture.cancel(CompletableFuture.java:2466)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.AndroidFuture.cancel(AndroidFuture.java:153)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl$CompletionAwareJob.cancel(ServiceConnector.java:779)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl.cancelPendingJobs(ServiceConnector.java:608)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl.unbindJobThread(ServiceConnector.java:593)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at com.android.internal.infra.ServiceConnector$Impl$$ExternalSyntheticLambda0.run(Unknown Source:2)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at android.os.Handler.handleCallback(Handler.java:958)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at android.os.Handler.dispatchMessage(Handler.java:99)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at android.os.Looper.loopOnce(Looper.java:205)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at android.os.Looper.loop(Looper.java:294)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at android.os.HandlerThread.run(HandlerThread.java:67)
10-20 07:29:05.220  5955  6134 E PermissionControllerManager:   at com.android.server.ServiceThread.run(ServiceThread.java:46)
10-20 07:29:05.221  5955  6134 E PermissionPolicyService: Error granting/upgrading runtime permissions for user 0
10-20 07:29:05.229  5902  5902 I Zygote  : Process 6180 exited due to signal 9 (Killed)
10-20 07:29:05.239  5955  5955 V SystemServerTiming: Permission_callback_waiting-0 took to complete: 100268ms
10-20 07:29:05.240  5955  5955 V SystemServerTiming: OnBootPhase_550_com.android.server.policy.PermissionPolicyService took to complete: 100272ms
10-20 07:29:05.240  5955  5955 E System  : ******************************************
10-20 07:29:05.283  5955  5955 E System  : ************ Failure starting system services
10-20 07:29:05.283  5955  5955 E System  : java.lang.RuntimeException: Failed to boot service com.android.server.policy.PermissionPolicyService: onBootPhase threw an exception during phase 550
10-20 07:29:05.283  5955  5955 E System  :      at com.android.server.SystemServiceManager.startBootPhase(SystemServiceManager.java:294)
10-20 07:29:05.283  5955  5955 E System  :      at com.android.server.SystemServer.lambda$startOtherServices$6(SystemServer.java:2894)
10-20 07:29:05.283  5955  5955 E System  :      at com.android.server.SystemServer.$r8$lambda$R8_YVQM1rsXoSgswlNNq9SsFhyw(SystemServer.java:0)
[... cut ...]
10-20 22:11:40.348   943   943 W Role    : Cannot get ApplicationInfo for default holder: com.android.messaging
10-20 22:11:40.403   517   564 I system_server: Using smaps_rollup for pss collection
10-20 22:11:40.821   517   550 I ActivityManager: Flag disabled. Ignoring finishAttachApplication from uid: 10069. pid: 943
```

If I instead attach LLDB to a stuck application process, I can see that ART is interpreting something, but I don't know how to reconstruct a Java backtrace from the native backtrace. Attaching and detaching LLDB is not sufficient to get un-stick the process.

enh-google, 2023-10-20:
@prashanthswami has been looking into the "boot phase 550" problem...

SiFiveHolland, 2023-10-20:
Note that this issue is different from the one we've seen earlier where `com.android.permissioncontroller` doesn't get started at all because of the `isAllowedWhileBooting` check. In this case, we pass the check, and Zygote forks a process (see all the "Slow operation" log lines), but something prevents that process from handshaking with ActivityManager.

prashanthswami, 2023-10-20:
Staring at this - sorry about the slowness here, I'm still paging in a lot of Android and ActivityManager is it's own beast to understand.

To follow up on where I'm at, agree with @SiFiveHolland here - it's pretty clear that on the latest build, it's _trying_ to launching permissioncontroller but not getting there. I'm tracking down something that looks incredibly concerning:

You'll note below that we're starting two processes with PIDs 3206 and 3224.

```
10-20 23:33:18.815  3028  3058 I ActivityManager: Start proc 3206:com.android.systemui/u0a82 for service {com.android.systemui/com.android.systemui.wallpapers.ImageWallpaper}
10-20 23:33:19.337  3028  3058 I ActivityManager: Start proc 3224:com.android.permissioncontroller/u0a101 for service {com.android.permissioncontroller/com.android.permissioncontroller.permission.service.PermissionControllerServiceImpl}
```

But . . .

```
prashanthsw:/media/android-ssd/aosp-master-with-phones$ adb shell ps | grep systemui
u0_a82        3206  2967   16894412  74348 0                   0 R com.android.systemui
u0_a101       3224  2967   16894412  74612 0                   0 R com.android.systemui
```

I just noticed the edit posted above, but yes, agree - this is affecting multiple processes. I'm not sure why my attempts to bisect aren't able to pinpoint when this got introduced though, but this isolates the problem down to somewhere in ActivityManager or ProcessList, I would think.

prashanthswami, 2023-10-27:
Sorry for the silence @SiFiveHolland - I'm close to resolving this. There's a bit more work to do to properly fix this, this is my highest priority. However, to unblock yourself:

1. You can patch this in or just wait for it to merge: https://android-review.googlesource.com/c/platform/frameworks/base/+/2807216 - this will fix the duplicate process names.
2. Patch this in: https://android-review.googlesource.com/c/platform/external/conscrypt/+/2808553

I wouldn't read too much into the commit message for the second patch, I'm still mulling over it because the explanation doesn't really track with why this is working for all the other targets. There's possibly some Dalvik investigation to do here to solve this properly. However, with both these patches on ToT, I'm able to boot AOSP.

SiFiveHolland, 2023-10-27:
I can confirm your patch https://android-review.googlesource.com/c/platform/external/conscrypt/+/2808553 does allow the boot process to complete.

prashanthswami, 2023-10-30:
Thanks for confirming! We figured out the actual fix here: https://android-review.git.corp.google.com/c/platform/art/+/2809833 . You can drop my patch now and patch this in (or sync the tree) and you should be good to go. I've confirmed locally that a clean tip-of-tree build boots properly.

Thanks for your support on this @SiFiveHolland , closing this out!

closed 2023-10-30 by prashanthswami