google/android-riscv64 #116: Fail to boot riscv64 cuttlefish

Bruce0316, 2023-11-01:
I am currently attempting to build RISC-V Cuttlefish following the instructions from [README.md](https://github.com/google/android-riscv64/blob/main/README.md.)

Here is my environment and build command:
```
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.3 LTS
Release:	22.04
Codename:	jammy

$ qemu-system-riscv64 --version
QEMU emulator version 8.1.0
Copyright (c) 2003-2023 Fabrice Bellard and the QEMU Project developers

#build command 
$ cd aosp
$ source build/envsetup.sh
$ lunch aosp_cf_riscv64_phone-trunk_staging-userdebug
============================================
PLATFORM_VERSION_CODENAME=VanillaIceCream
PLATFORM_VERSION=VanillaIceCream
PRODUCT_INCLUDE_TAGS=com.android.mainline mainline_module_prebuilt_nightly
TARGET_PRODUCT=aosp_cf_riscv64_phone
TARGET_BUILD_VARIANT=userdebug
TARGET_ARCH=riscv64
TARGET_ARCH_VARIANT=riscv64
TARGET_CPU_VARIANT=generic
HOST_OS=linux
HOST_OS_EXTRA=Linux-6.2.0-35-generic-x86_64-Ubuntu-22.04.3-LTS
HOST_CROSS_OS=linux_musl
BUILD_ID=MAIN
OUT_DIR=out
============================================
$ make -j8
```
When I run `launch_cvd -cpus=4 --memory_mb=8192`, I encounter the following issue:
```
/usr/bin/qemu-system-riscv64
qemu-system-riscv64: -display egl-headless: Parameter 'type' does not accept value 'egl-headless'
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=32, out=31
KeyMint Rust TA running with fd_in=38, fd_out=37, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
Detected unexpected exit of monitored subprocess /usr/bin/qemu-system-riscv64
Subprocess /usr/bin/qemu-system-riscv64 (3605200) has exited with exit code 1
Stopping all monitored processes due to unexpected exit of critical process
11-01 17:00:19.291 3605220 3605226 I stop_cvd: main.cc:124 Requesting stop
Metrics host received: VMStop
Metrics posted to ClearCut
Exit request from parent process
```
QEMU is showing an "-display egl-headless" error, indicating that this parameter value is not supported. I'd like to know how to configure QEMU correctly to avoid the "-display egl-headless" error or if there are other factors affecting a successful boot.

I also attempted to use **qemu-system-riscv64 8.1.2** and encountered an error with the following message:
```
/usr/bin/qemu-system-riscv64
2023-10-31T07:52:19.480170Z qemu-system-riscv64: can't apply global rv64-riscv-cpu.v=true: Property 'rv64-riscv-cpu.v' not found
Detected unexpected exit of monitored subprocess /usr/bin/qemu-system-riscv64
Subprocess /usr/bin/qemu-system-riscv64 (1712626) has exited with exit code 1
Stopping all monitored processes due to unexpected exit of critical process
starting Rust KeyMint TA implementation in a thread
starting C++ KeyMint implementation in a thread with FDs in=31, out=30
KeyMint Rust TA running with fd_in=37, fd_out=36, security_level=TrustedEnvironment
Secure deletion data file found. Parsing.
10-31 15:52:19.629 1712656 1712664 I stop_cvd: main.cc:124 Requesting stop
Exit request from parent process
```

Has anyone encountered a similar situation? I would like to understand what is causing these issues and how to resolve them, if possible. Thank you.

enh-google, 2023-11-02:
yeah, we've checked in new qemu prebuilts, but we're still working on having them actually used by default. i think it's only https://android-review.googlesource.com/c/device/google/cuttlefish_vmm/+/2810815 left to go? hopefully that should make things work "out of the box" for more people, regardless of what qemu your host OS has.

enh-google, 2023-11-02:
(i've update the instructions on the front page to show how to use the prebuilt qemu.)

Bruce0316, 2023-12-12:
Thank you for your previous response. 
I rebuilt RISC-V Cuttlefish from scratch following the new guide [README.md](https://github.com/google/android-riscv64) 
My environment remains the same, and this time I used the AOSP prebuilt QEMU. 

However, my command line has been consistently stuck at:
```
init: starting service 'adbd'...
proxy_fastboot] Stop event (11) received. Stopping proxy
proxy_adb] Start event (5) received. Starting proxy
proxy_adb] From: tcp: 6520
proxy_adb] To: vsock: 3:5555 vhost_user: false
```

But my launcher.log keeps updating with the following content:
```
kernel_log_monitor I 12-12 23:44:55 23186 23186 kernel_log_server.cc:140] init: starting service 'adbd'...
proxy_fastboot I 12-12 23:44:55 23178 23178 socket_vsock_proxy.cpp:182] Stop event (11) received. Stopping proxy
proxy_adb I 12-12 23:44:55 23177 23177 socket_vsock_proxy.cpp:175] Start event (5) received. Starting proxy
proxy_adb I 12-12 23:44:55 23177 23177 socket_vsock_proxy.cpp:146] From: tcp: 6520
proxy_adb I 12-12 23:44:55 23177 23177 socket_vsock_proxy.cpp:147] To: vsock: 3:5555 vhost_user: false
modem_simulator D 12-12 23:45:02 23173 23191 channel_monitor.cpp:126] added one RIL client
tombstone_receiver D 12-13 00:10:25 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-001025
tombstone_receiver D 12-13 00:10:25 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-001025_1
tombstone_receiver D 12-13 00:22:26 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002226
tombstone_receiver D 12-13 00:22:26 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002226_1
tombstone_receiver D 12-13 00:23:16 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002316
tombstone_receiver D 12-13 00:23:16 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002316_1
tombstone_receiver D 12-13 00:25:08 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002508
tombstone_receiver D 12-13 00:25:08 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002508_1
tombstone_receiver D 12-13 00:25:13 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002513
tombstone_receiver D 12-13 00:25:13 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002513_1
tombstone_receiver D 12-13 00:26:33 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002633
tombstone_receiver D 12-13 00:26:33 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-002633_1
tombstone_receiver D 12-13 00:34:37 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-003437
tombstone_receiver D 12-13 00:34:37 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-003438
tombstone_receiver D 12-13 00:39:15 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-003915
tombstone_receiver D 12-13 00:39:15 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-003915_1
tombstone_receiver D 12-13 00:40:10 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004010
tombstone_receiver D 12-13 00:40:10 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004010_1
tombstone_receiver D 12-13 00:43:37 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004337
tombstone_receiver D 12-13 00:43:37 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004337_1
tombstone_receiver D 12-13 00:43:47 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004347
tombstone_receiver D 12-13 00:43:47 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004347_1
tombstone_receiver D 12-13 00:45:08 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004508
tombstone_receiver D 12-13 00:45:08 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004508_1
tombstone_receiver D 12-13 00:45:18 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004518
tombstone_receiver D 12-13 00:45:18 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004518_1
tombstone_receiver D 12-13 00:49:31 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004931
tombstone_receiver D 12-13 00:49:31 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-004931_1
[... cut ...]
tombstone_receiver D 12-13 00:53:29 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-005329_1
tombstone_receiver D 12-13 00:55:25 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-005525
tombstone_receiver D 12-13 00:55:25 23174 23174 main.cpp:46] Creating /tmp/1005/cvd_1/cuttlefish/instances/cvd-1/tombstones/tombstone_2023-12-13-005525_1
...
```

The process has been running for over an hour, and I eventually paused it.
I would like to inquire if anyone has encountered this issue and if there are possible solutions. Thank you.

enh-google, 2023-12-12:
what are the crashes in the tombstones?

Bruce0316, 2023-12-13:
it's all empty
```
file tombstone_2023-12-13-003915
tombstone_2023-12-13-003915: empty
```

Bruce0316, 2023-12-28:
I rebuilt RISC-V Cuttlefish again, and launch by
```
$ launch_cvd -cpus=4 --memory_mb=8192 --gpu_mode=none \
      --vm_manager=qemu_cli \
      --qemu_binary_dir=$ANDROID_BUILD_TOP/device/google/cuttlefish_vmm/qemu/x86_64-linux-gnu/bin
```
This time my launcher.log keeps looping this:
```
secure_env::kmr_ta I 12-29 00:56:02 17164 17481 lib.rs:581] Setting boot_info to BootInfo { verified_boot_key: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], device_boot_locked: false, verified_boot_state: Unverified, verified_boot_hash: [38, 18, 193, 72, 255, 153, 50, 28, 104, 215, 223, 108, 233, 17, 174, 108, 255, 33, 0, 115, 103, 254, 82, 224, 138, 73, 18, 170, 44, 68, 81, 123], boot_patchlevel: 20231205 }
secure_env::kmr_ta I 12-29 00:56:02 17164 17481 lib.rs:606] Setting hal_info to HalInfo { os_version: 140000, os_patchlevel: 202312, vendor_patchlevel: 20231205 }
secure_env::kmr_ta I 12-29 00:56:02 17164 17481 lib.rs:624] Set aidl_version to V3
secure_env::kmr_ta W 12-29 00:56:02 17164 17481 lib.rs:635] Setting attestation IDs directly
secure_env::kmr_ta::secret I 12-29 00:56:02 17164 17481 secret.rs:37] Setting HMAC key from 1 shared secret parameters
secure_env::kmr_ta E 12-29 00:56:02 17164 17481 lib.rs:1060] secure deleting all keys -- device likely to need factory reset!
secure_env::kmr_cf::sdd I 12-29 00:56:02 17164 17481 sdd.rs:180] Deleting all secrets
secure_env::kmr_cf::sdd I 12-29 00:56:02 17164 17481 sdd.rs:114] Generating new secret with slot ID: 1
secure_env::kmr_cf::sdd I 12-29 00:56:02 17164 17481 sdd.rs:65] No secure deletion data file found. Creating one.
secure_env::kmr_cf::sdd I 12-29 00:56:02 17164 17481 sdd.rs:147] Fetching secret with slot ID: 1
kernel_log_monitor I 12-29 00:56:22 17153 17153 kernel_log_server.cc:133] GUEST_BUILD_FINGERPRINT: generic/aosp_cf_riscv64_phone/vsoc_riscv64:VanillaIceCream/MAIN.2612c148/eng.bruce.20231228.222236:userdebug/test-keys
secure_env::kmr_ta::secret I 12-29 00:56:25 17164 17481 secret.rs:37] Setting HMAC key from 1 shared secret parameters
kernel_log_monitor I 12-29 00:56:30 17153 17153 kernel_log_server.cc:133] GUEST_UBOOT_VERSION: 2023.04-gc2f63e06296a-ab11193194 (Dec 05 2023 - 03:08:51 +0000)
kernel_log_monitor I 12-29 00:56:30 17153 17153 kernel_log_server.cc:140] U-Boot 20
secure_env D 12-29 00:56:30 17164 17486 secure_env_not_windows_main.cpp:129] secure_env detected guest reboot, restarting.
secure_env D 12-29 00:56:30 17164 17164 tpm_storage.cpp:39] Initializing secure index file
secure_env D 12-29 00:56:30 17164 17164 tpm_storage.cpp:45] Restoring index from file
secure_env I 12-29 00:56:30 17164 17164 secure_env_not_windows_main.cpp:289] starting Rust KeyMint TA implementation in a thread
secure_env I 12-29 00:56:30 17164 17164 secure_env_not_windows_main.cpp:304] starting C++ KeyMint implementation in a thread with FDs in=63, out=62
secure_env::kmr_cf I 12-29 00:56:30 17164 17632 lib.rs:66] KeyMint Rust TA running with infile=69, outfile=68, security_level=TrustedEnvironment
secure_env::kmr_cf::sdd I 12-29 00:56:30 17164 17632 sdd.rs:60] Secure deletion data file found. Parsing.
kernel_log_monitor I 12-29 00:56:31 17153 17153 kernel_log_server.cc:133] GUEST_KERNEL_VERSION: 6.6.0-mainline-gc387c4e84ce5-ab11047873 (build-user@build-host) (Android (10742551, +pgo, +bolt, +lto, +mlgo, based on r498229b) clang version 17.0.4 (https://android.googlesource.com/toolchain/llvm-project e34ed7d63863b45858e74126edaa738e75887800), LLD 17.0.4) #1 SMP PREEMPT Wed Nov  1 22:15:43 UTC 2023
kernel_log_monitor I 12-29 00:56:31 17153 17153 kernel_log_server.cc:140] ] Linux version
kernel_log_monitor I 12-29 00:57:10 17153 17153 kernel_log_server.cc:140] init: starting service 'adbd'...
proxy_adb I 12-29 00:57:10 17145 17145 socket_vsock_proxy.cpp:175] Start event (5) received. Starting proxy
proxy_fastboot I 12-29 00:57:10 17146 17146 socket_vsock_proxy.cpp:182] Stop event (11) received. Stopping proxy
proxy_adb I 12-29 00:57:10 17145 17145 socket_vsock_proxy.cpp:146] From: tcp: 6520
proxy_adb I 12-29 00:57:10 17145 17145 socket_vsock_proxy.cpp:147] To: vsock: 3:5555 vhost_user: false
kernel_log_monitor I 12-29 00:57:10 17153 17153 kernel_log_server.cc:133] GUEST_UBOOT_VERSION: 2023.04-gc2f63e06296a-ab11193194 (Dec 05 2023 - 03:08:51 +0000)
kernel_log_monitor I 12-29 00:57:10 17153 17153 kernel_log_server.cc:140] U-Boot 20
secure_env D 12-29 00:57:10 17164 17637 secure_env_not_windows_main.cpp:129] secure_env detected guest reboot, restarting.
secure_env D 12-29 00:57:10 17164 17164 tpm_storage.cpp:39] Initializing secure index file
secure_env D 12-29 00:57:10 17164 17164 tpm_storage.cpp:45] Restoring index from file
secure_env I 12-29 00:57:10 17164 17164 secure_env_not_windows_main.cpp:289] starting Rust KeyMint TA implementation in a thread
secure_env I 12-29 00:57:10 17164 17164 secure_env_not_windows_main.cpp:304] starting C++ KeyMint implementation in a thread with FDs in=63, out=62
secure_env::kmr_cf I 12-29 00:57:10 17164 17725 lib.rs:66] KeyMint Rust TA running with infile=69, outfile=68, security_level=TrustedEnvironment
secure_env::kmr_cf::sdd I 12-29 00:57:10 17164 17725 sdd.rs:60] Secure deletion data file found. Parsing.
kernel_log_monitor I 12-29 00:57:11 17153 17153 kernel_log_server.cc:133] GUEST_KERNEL_VERSION: 6.6.0-mainline-gc387c4e84ce5-ab11047873 (build-user@build-host) (Android (10742551, +pgo, +bolt, +lto, +mlgo, based on r498229b) clang version 17.0.4 (https://android.googlesource.com/toolchain/llvm-project e34ed7d63863b45858e74126edaa738e75887800), LLD 17.0.4) #1 SMP PREEMPT Wed Nov  1 22:15:43 UTC 2023
kernel_log_monitor I 12-29 00:57:11 17153 17153 kernel_log_server.cc:140] ] Linux version
```
Has anyone encountered a similar situation? Thanks.

prashanthswami, 2024-02-15:
We've managed to clean some things up in regards to this - I haven't been able to reproduce your exact issue, but I've uploaded some new prebuilts that work for me and removed the additional flags required to start them.

Can you simply try `launch_cvd -cpus=4 --memory_mb=8192 --gpu_mode=none` after re-syncing your tree?

closed 2024-11-12 by enh-google